lvgl/lvgl #1096: lv_win strange behaviour

aenchevich, 2019-06-06:
I noticed the following weird things in lv_win behavour:

1. You have to set the window size in advance or else you will get a window as big as the screen. WHy?? This way you can not create a window which will resize itself to include/involve all the children that are added to it AND also setting layots on the window is kinda useless.

2. The above problem is because there is code in `lv_win_create` which sets the size to the size of the screen. So.. I removed it :) hoping that it will solve the problem. The window got smaller (default obj size but it DID NOT autofit to involve the children. What happend instead was that when scrolling the content the entire window body was scrolling, instead of just the area containing the scroll bar??

3. "Ok, no problem! I need to call set fit on the container of the content" - I thought. So I did `lv_win_get_content`, then set_fit on that object, then used that object as parent when adding components to the window. Still no auto fit.

4. What DID happen due to the above, however, was that the window's border around the content area disappeared?

= = =
So to summarize, IMO the expected behaviour should be:

1. If you create a window without setting it's size it should be the default object size. Adding objects to it should make thw window bigger, if necessary (to involve all children).

2. When scrolling the window the border should always be visible and only the area inside it (with the scrollbars) should scroll.

3. Perhaps 1. above should auto fit itself only when the inner page of the wndow is set to auto fit?

4. Adding objects to the content page of the window should not remove the border.

I illustrated these with screenshots (used a simplified lv_test_win_1() which only draws one window). Last img is expected behaviour.

[image]
[image]
[image]

aenchevich, 2019-06-06:
This is my modified lv_test_win_1 func

```
void lv_test_win_1(void)
{
    static lv_style_t header;
    lv_style_copy(&header, &lv_style_plain);
    header.body.main_color = LV_COLOR_RED;
    header.body.grad_color = LV_COLOR_MAROON;
    header.body.padding.inner = 0;
    header.text.color = LV_COLOR_WHITE;

    lv_obj_t * win3 = lv_win_create(lv_scr_act(), NULL);
    lv_obj_align(win3, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_win_set_style(win3, LV_WIN_STYLE_HEADER, &header);
    lv_win_set_style(win3, LV_WIN_STYLE_BTN_REL, &lv_style_transp);
    lv_win_set_style(win3, LV_WIN_STYLE_CONTENT_BG, &lv_style_plain_color);
    lv_win_set_style(win3, LV_WIN_STYLE_CONTENT_SCRL, &lv_style_plain);
    lv_win_set_style(win3, LV_WIN_STYLE_BG, &lv_style_plain_color);
    lv_win_set_btn_size(win3, LV_DPI / 3);

    lv_obj_t * bgnd = win3;

#if 0
    lv_obj_t * page = lv_win_get_content(win3);
    lv_cont_set_fit(page, true, true);
    bgnd = page;
#endif

    lv_obj_t * label = lv_label_create(bgnd, NULL);
    lv_obj_set_pos(label, 10, 10);
    lv_label_set_text(label, "Styled window\n\nThe content background has\ndifferent color");
}
```

kisvegabor, 2019-06-07:
It's the intended behavior. The window can't resize itself to involve al children. Instead, its content area will be resized and scrollbars will appear. That's how windows work on PCs too. You have a fix sized area with a header and scrollable content area. 

The default size is set to "screen size" because it's the most probable use case.

It seems the window is not the best object type for your needs. Please open a new topic in the [Forum's How-to category](https://forumtest.littlevgl.com/c/how-to) and continue there.

closed 2019-06-07 by kisvegabor