lvgl/lvgl #1097: lv_obj_align IN_BOTTOM_LEFT/RIGHT produce non-symetric results

aenchevich, 2019-06-07:
[image]

Produced with this code (resolution is 640x480):
```
void lv_test_win_1(void)
{
    lv_obj_t * win;

    static lv_style_t header;
    lv_style_copy(&header, &lv_style_plain);
    header.body.main_color = LV_COLOR_RED;
    header.body.grad_color = LV_COLOR_MAROON;
    header.body.padding.inner = 0;
    header.text.color = LV_COLOR_WHITE;

    win = lv_win_create (lv_scr_act(), NULL);

    lv_win_set_style(win, LV_WIN_STYLE_HEADER, &header);
    lv_win_set_style(win, LV_WIN_STYLE_BTN_REL, &lv_style_transp);
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_BG, &lv_style_plain_color);
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_SCRL, &lv_style_plain);
    lv_win_set_style(win, LV_WIN_STYLE_BG, &lv_style_plain_color);

    lv_win_set_title (win, "Format HDD Drive");
    lv_obj_set_size(win, LV_HOR_RES/3, LV_VER_RES/2);
    lv_obj_align (win, NULL, LV_ALIGN_CENTER, 0, 0);

    lv_obj_t * parent = win;
    lv_obj_t * label;

    /* progress indicatior */
    lv_obj_t * bar = lv_bar_create(parent, NULL);
    lv_obj_align(bar, NULL, LV_ALIGN_IN_TOP_MID, 0, +10);
    lv_bar_set_value(bar, 30);

    /* % */
    label = lv_label_create(bar, NULL);
    lv_label_set_text(label, "30%");
    lv_label_set_style(label, &lv_style_pretty);
    lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);

    /* 'OK' label */
    label = lv_label_create (parent, NULL);
    lv_label_set_style (label, &lv_style_pretty);
    lv_label_set_body_draw (label, true);
    lv_label_set_text (label, "OK");
    lv_obj_align (label, NULL, LV_ALIGN_IN_BOTTOM_LEFT, 0, 0);

    /* 'EXIT' label */
    label = lv_label_create (parent, NULL);
    lv_label_set_style (label, &lv_style_pretty);
    lv_label_set_body_draw (label, true);
    lv_label_set_text (label, "EXIT");
    lv_obj_align (label, NULL, LV_ALIGN_IN_BOTTOM_RIGHT, 0, 0);
}
```

kisvegabor, 2019-06-07:
It's really not so intuitive in `v5.3` but it was improved for `v6.0`. In `6.0` by default the "white area" will fill the whole "blue area". This (and other updates) help to make positioning on pages (the content area of the window) easier. 

If you are still at the beginning of your project I suggest changing to `dev-6.0`. `v6.0` will be released soon, see this post: https://forumtest.littlevgl.com/t/v6-0-is-near-the-merge-window-will-be-closed-this-week/39

If you want to solve it in `v5.3` I suggest disabling the vertical fit for the content area and set its height manually. 

In both cases, you should avoid aligning to the very edge of the parent because if the parent has horizontal or vertical fit, its size will be increased by the padding value and you might get unexpected scrollbars.

aenchevich, 2019-07-24:
Hello Gabor

I checked if these problems were fixed in v.6. They are not :)

I have a fix for the problems. With this fix applied, the output looks like this, which IMO is almost 100% as expected (except that the rounded borders of the buttons are outside the visible area):

[image]

Would you like to implement it and if yes - what do you want me to do: open this PR  again or create a new one?

kisvegabor, 2019-07-25:
Could you share the source code of the above example? 

A PR would be welcome to see the updates.

aenchevich, 2019-07-25:
Of course!

The code is in the last commit of branch  "ae-dev-5.2" in my lvgl forked repo:  https://github.com/aenchevich/lvgl.git

aenchevich, 2019-07-25:
This is what I have after I (manually) merged my fixes from v5.2 into dev6:
[image]

Is it acceptable? If yes I will provide a patch file or give you the code in some other suitable way (my forked repo is using v5.2 and I would like to keep it that way for now).

 - - -
**Question about a possible BUG:**
 - - -
While working on this I ran into SEGV errors and my debugging shows that there is an endless recursion in lv_cont code:
  lv_cont_signal(LV_SIGNAL_CORD_CHG)  calls lv_cont_refr_autofit()
  lv_cont_refr_autofit  calls lv_cont_signal(LV_SIGNAL_CORD_CHG)

embeddedt, 2019-07-25:
I ran into that issue back when 6.0 was still in development, but I thought @kisvegabor had fixed it. Are you using the latest 6.0 from `master`?

aenchevich, 2019-07-25:
Maybe we should re-open this? 

Oh and also - on Aug 2nd  I go away for about two weeks. I will do my best to answer here and do some work on the code before that but just to let you know... in case I don't respond for a long time... :)

aenchevich, 2019-07-25:
I used the version in pc_simulator to test if the issue is fixed in 6. So in there the lvgl is at a42154777414bd768e753c55e3f80248f613de22   (Committer: GitHub <noreply@github.com>  2019-07-10 16:05:01)

embeddedt, 2019-07-25:
More than new enough, the problem I was referring to might have been as far back as March.

aenchevich, 2019-07-25:
Actuallyh are you referring to the original problem (layout) or to the recursion seg fault?

embeddedt, 2019-07-25:
The segfault.

kisvegabor, 2019-07-26:
@aenchevich 
Please could share to the code you have used to create this UI?

aenchevich, 2019-07-26:
I thought I provided the code? The lv_test_win_1 function + the branch of lvgl from my repo which I quoted above should give the resulting code for v5.2. As far as v6.x - I asked if you want my fix branched off master or dev-6.1 but did not get an answer, that's why I have not pushed it yet.

Anyway - here's the archives of my pc_simulator dirs for both: version 5 and 6, everything is in there!    :)

http://www.linux-pages.com/pc_simulator.ae.v5.2.tar.gz
http://www.linux-pages.com/pc_simulator.v6.x.tar.gz

embeddedt, 2019-07-27:
> As far as v6.x - I asked if you want my fix branched off master or dev-6.1 but did not get an answer, that's why I have not pushed it yet.

Make it against `master`. If we change our mind later it should still merge with no conflicts (not much has happened on `dev-6.1` yet).

I have a few concerns about your patch in its current state, but it's easier to review as a pull request rather than manually.

kisvegabor, 2019-07-28:
> I asked if you want my fix branched off master or dev-6.1 but did not get an answer, that's why I have not pushed it yet.

I wanted to check the code first to see if if it's really a bug.

In v6.0 `auto-fit` has been improved a lot. See it int he [docs](https://docs.littlevgl.com/en/html/object-types/cont.html#auto-fit). The default fit policy on windows is `LV_FIT_FILL`, so if you place something outside window the content area will be larger with padding.
So you just need to align to e.g. 10 instead of 0:
```c 
    lv_obj_align(bar, NULL, LV_ALIGN_IN_TOP_MID, 0, 10);
    lv_obj_align (label_ok, NULL, LV_ALIGN_IN_BOTTOM_LEFT, 10, -10);
    lv_obj_align (label_exit, NULL, LV_ALIGN_IN_BOTTOM_RIGHT, -10, -10);
```
Or change the fit policy to `LV_FIT_FLOOD` which prevents addig scrollbars.
```c
    lv_page_set_scrl_fit(lv_win_get_content(win), LV_FIT_FLOOD);
```

Note that the background of the labels is drawn only virtually i.e. doesn't make the labels really larger. Therefore still the text will be aligned. I've updated the [docs](https://docs.littlevgl.com/en/html/object-types/label.html#draw-background) to describe this.

So it seems to it works as expected.

aenchevich, 2019-07-28:
When you say "it works as expected" are you saying that my fix works as expected or that the code as-is works as expected (and hence the fix is not needed)?

kisvegabor, 2019-07-29:
I meant by adding the above-mentioned functions it works with the latest `master` branch.

closed 2019-08-26 by stale[bot]