lvgl/lvgl #1098: Button release signal didn't create when you keep button down and move you finger away from button?

lianggongkai, 2019-06-08:
Hi , I meet this problem as below.
Button release signal didn't create when you keep button down and move you finger away from button. I set a button release callback function to do some label settings. When you press the button, the label is set. But when you keep press button, and move your finger away from the button, there's no button release signal callback function excuted. So the author or anyone who can help me , Please give some advise. Thank you .

embeddedt, 2019-06-08:
@lianggongkai This is how most GUIs work. Try it now on GitHub. Put your mouse on the "Close issue" button, press, and then drag away. It won't actually close the issue.

kisvegabor, 2019-06-08:
@lianggongkai 
Actually, there is a solution for it but please ask your question in the [Forum](https://forumtest.littlevgl.com/).

We use GitHub issues for development related discussions and the Forum to answer questions.
You can simply log in with your GitHub account.

closed 2019-06-08 by kisvegabor