lvgl/lvgl #1099: [v6.0] Update the docs

kisvegabor, 2019-06-11:
One important thing remained before releasing `v6.0`: the documentation needs to be updated.

I will update the general part of the documentation but looking for some help to review and update the documentation of the object types. 

Every object type's documentation should have these sections:
- Overview
- Styles
- Events: which events are sent by the objects (`LV_EVENT_...`)
- Keys: which keys are used by the object (`LV_KEY_...`)
- Example

I created the `dev-6.0` branch in the [docs repo]https://github.com/littlevgl/docs/tree/dev-6.0) and updated the slider as reference. (Just to see the format)

If you want to contribute, please comment here.

If you find some typos or obscure comments please fix them or comment here to discuss them.

The translation should be updated when the English version is ready.

Her we can keep track of the progress. 

- [ ] general part: @kisvegabor 

- [ ] lv_arc.c
- [ ] lv_bar.c
- [ ] lv_btn.c
- [ ] lv_btnm.c
- [ ] lv_calendar.c
- [ ] lv_canvas.c
- [ ] lv_cb.c
- [ ] lv_chart.c
- [ ] lv_cont.c
- [ ] lv_ddlist.c
- [ ] lv_gauge.c
- [ ] lv_imgbtn.c
- [ ] lv_img.c
- [ ] lv_kb.c
- [ ] lv_label.c
- [ ] lv_led.c
- [ ] lv_line.c
- [ ] lv_list.c
- [ ] lv_lmeter.c
- [ ] lv_mbox.c
- [ ] lv_objx_templ.c
- [ ] lv_page.c
- [ ] lv_preload.c
- [ ] lv_roller.c
- [ ] lv_slider.c: @kisvegabor 
- [ ] lv_spinbox.c
- [ ] lv_sw.c
- [ ] lv_table.c
- [ ] lv_tabview.c
- [ ] lv_ta.c
- [ ] lv_tileview.c
- [ ] lv_win.c

C47D, 2019-06-11:
I will try to get the Spanish translation of v5.3 fully translated while the English translation of v6.0 is done!

embeddedt, 2019-06-11:
@C47D I'm not sure if that's a good idea.

Once we release 6.0, the only way to access the older documentation would be to find it in the Wayback Machine or download the GitHub repository and go back in the history.

So it might be a better idea to wait for us to finish updating the documentation and then update your existing translations (and translate what you haven't done yet).

C47D, 2019-06-11:
@embeddedt Seems fair, i will wait.

amirgon, 2019-06-12:
> Every object type's documentation should have these sections:
> 
> ...
> * Example

@kisvegabor Did you consider making the examples interactive (using Emscripten), instead of putting an image next to the code, as it is today?

kisvegabor, 2019-06-12:
@amirgon 
I think in the documentation "static" images and code would be better to enable converting to PDF.
However, we can add links, where the example program could be automatically loaded from an URL. What do you think?

amirgon, 2019-06-12:
> However, we can add links, where the example program could be automatically loaded from an URL. What do you think?

@kisvegabor This would be easy with micropython, but harder for C since every example C program would need to be compiled by Emscripten.  
Did you think of a python or a C program automatically loaded?

kisvegabor, 2019-06-12:
I meant to load the Micropython version of the C example.

amirgon, 2019-06-12:
> I meant to load the Micropython version of the C example.

That could be very nice. Especially if we add an online editor so the user could change the code interactively on the web page and see how it affects the results (we are discussing this on #792).  

It would require converting these examples to Micropython. Also, these examples are just snippets that need to be converted to full working examples.

embeddedt, 2019-06-12:
Using MicroPython here doesn't make sense to me. Currently, I think a good portion (most?) of our users are **not** using the MicroPython binding. Thus, I don't think the ability to modify the code is going to be of much advantage to most of them. Maybe I'm missing something, but a better idea, in my opinion, is to take advantage of the VPS being used to host the forum, and modify the way we reference examples in the documentation. This is similar to what @kisvegabor was suggesting.

This is just a hypothetical workflow; we would probably have to modify it given the constraints of our tooling.

I'll use the button widget as an example.

1. Write documentation for the button as normal.
2. Instead of actually writing the example in the documentation itself - find some way of indirectly referencing that a parser should include the code for "lv_btn_example_1".
3. Have the generator (or a script that runs before it) detect the indirect reference to "lv_btn_example_1". Find `lv_btn_example_1.lv_example` in a new `examples` folder inside the `docs` repository. The contents of that file would be *just the example-specific lines*, without any `#include` directives or any functions.

```c
lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
lv_obj_t *label = lv_label_create(btn, NULL);
lv_label_set_text(label, "Button");
lv_obj_align(btn, NULL, LV_ALIGN_CENTER, 0, 0);
```

4. Insert the contents of that file into the documentation where the indirect reference was.
5. find `lv_btn_example_1.jpg` inside the `examples` folder, and insert that into the documentation as well.

The result is identical to the previous behavior, where the example (and a screenshot) are included inline inside the documentation file.

If we're generating static HTML, we can also insert an `iframe` that links to `https://examples.littlevgl.com/lv_btn_example_1.html`. This is a static copy of the compiled example that could be generated by our VPS or possibly even Travis. I could easily write a script that will generate it for us.

The result is an interactive example available on `docs.littlevgl.com`, while still making the code (and possibly a screenshot) available.

Another benefit of this method is that we can easily modify examples in a centralized location without having to find the individual documentation file and update that.

kisvegabor, 2019-06-12:
@embeddedt 
It sounds great! Just to be sure, the C example would be generated with Emscripten and displayed as an iframe. If it would work, besides the micropython simulator, we could add this to `littlevgl.com` with a simple editor too.

Having a C and a Micropython editor would be awesome!

---

Meanwhile, I'm working on the documentation and found [Sphinx](https://www.sphinx-doc.org/en/master/index.html) and I was able to make it work with the theme used by the [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html).

Here it is:
https://littlevgl.com/sphinx/index.html

And the PDF
https://littlevgl.com/sphinx/LittlevGL.pdf

What do you think?

embeddedt, 2019-06-12:
> Just to be sure, the C example would be generated with Emscripten and displayed as an iframe, right?

> Having a C and a Micropython editor would be awesome!

Not entirely accurate. Unlike MicroPython, I can't compile C code on-the-fly in the browser. So the editor would not be available for C. However, we could use Travis or the VPS to generate static versions of all the examples in the documentation.

kisvegabor, 2019-06-14:
> Unlike MicroPython, I can't compile C code on-the-fly in the browser. So the editor would not be available for C. 

However, we could compile the code on the VPS. But I suppose the compiling would be very slow and would give a very poor user experience.

So the original idea, to make examples interactive, is still great. Both approaches seem reasonable and have advantages and disadvantages.

**C examples**
- Create the examples only once

**Micropythn examples**
- No need to develop the VPS part
- The examples can be modified by the user

IMO the Micropython examples would work the best if we would have a simple editor. At first, it could be only a text area without any fancy syntax highlighting but it would be important to see the script at one piece. If a URL is provided. the content can be copied to text are first to see the script and then loaded to the terminal. To enable the reloading script can we store the script in a Session variable and reload it when the page has reloaded?

amirgon, 2019-06-14:
> At first, it could be only a text area without any fancy syntax highlighting

@kisvegabor There is no reason to use text area. With very little effort we can use a good online editor such as [CodeMirror](https://github.com/codemirror/CodeMirror) or [Ace](https://github.com/ajaxorg/ace), we are discussing this on #792.

> To enable the reloading the script, can we store the script in a Session variable and reload it when the page has reloaded?

I have a different idea.  
What if we allow encoding the entire script in the URL?  
The advantage is that it will become very easy to share scripts. Imagine you write some script (or change an existing one), then copy-paste a URL to the forum or to GitHub. Anyone clicking your URL would see (edit and run) your script, as it's self contained in the URL.
URLs can be very long (in modern browsers up to 64KB characters!) so I think it's reasonable to allow embedding the script itself in the URL (as an alternative to specifying the script location, as we have today).  

What do you think?

kisvegabor, 2019-06-14:
@amirgon 
> There is no reason to use text area. With very little effort we can use a good online editor such as CodeMirror or Ace, we are discussing this on #792.

If I remember well @embeddedt said it would be too much work now. Anyway, if we have something which already works it's easier to improve it later.

> I have a different idea.
What if we allow encoding the entire script in the URL?

I like the script in the URL idea but I don't know the web development pitfalls well enough. :)

embeddedt, 2019-06-14:
> If I remember well @embeddedt said it would be too much work now.

The ability to edit the script and re-run it was the problem; not the actual integration of Ace/CodeMirror itself. Putting the script in the URL itself could solve that.

> What if we allow encoding the entire script in the URL?

Something like base64?

-------

The reason I suggested the C idea is because it provides a better way of keeping the examples **in sync with what's in the documentation.** Unless we have a way of converting simple examples to MicroPython with a script?

kisvegabor, 2019-06-14:
> The ability to edit the script and re-run it was the problem; not the actual integration of Ace/CodeMirror itself. Putting the script in the URL itself could solve that.

Great!

> The reason I suggested the C idea is because it provides a better way of keeping the examples in sync with what's in the documentation. Unless we have a way of converting simple examples to MicroPython with a script?

I don't know how difficult it would be to convert the examples with scripts, but should be only a few hours by hand. :smile: As the examples don't change frequently I think it can be a solution as well.

amirgon, 2019-06-14:
> Something like base64?

@embeddedt Yes. I once wrote some web application that kept the long url param gziped, with [lz-string](https://github.com/pieroxy/lz-string/). It has a function called `compressToEncodedURIComponent` which compresses the string, encodes it to Base64 and makes it URL safe so you could use it on the URL directly. I recommend using this. (There's also [CDN](https://cdnjs.com/libraries/lz-string)).

> Unless we have a way of converting simple examples to MicroPython with a script?

The Micropython syntax is not that different from C, once you get used to it. We can probably convert the examples by a script, but even if we don't, if you look at a piece of micropython script it's very easy to guess how the C would look like.  
I agree with @kisvegabor that writing such a script to work reliably just for converting the examples is probably not worth the effort.

embeddedt, 2019-06-14:
> if you look at a piece of micropython script it's very easy to guess how the C would look like.

Are you suggesting that we switch to only showing Python examples?

amirgon, 2019-06-14:
> Are you suggesting that we switch to only showing Python examples?

Well, I suggest we show both C and python examples.  
The Python examples would also be runnable online, and the C just for reference.  
The question is whether we convert C to Python manually (and keep them in sync), or have a script do that. Both options are doable, the question is which one is less work.

kisvegabor, 2019-06-15:
> Well, I suggest we show both C and python examples.

Once we will have the examples in both languages we can add both them.

> The question is whether we convert C to Python manually (and keep them in sync), or have a script do that. Both options are doable, the question is which one is less work.

These examples are about very basic things. We should try to keep them short and easy to digest. So I think manual translation can work. But we will see.

kisvegabor, 2019-06-16:
I've updated the documentation with some new sections. Please take look and tell what you think.
https://littlevgl.com/sphinx/index.html

embeddedt, 2019-06-16:
@kisvegabor It looks great!

kisvegabor, 2019-06-18:
Some updates about the docs:
- Doxygen is adopted. Some tailoring is still required in the header files but seems working.
- Examples in a separate folder but I needed to use symlinks. There it can be compiled on Windows. 

See https://littlevgl.com/sphinx/object-types/btn.html

embeddedt, 2019-06-18:
Note to anyone reading @kisvegabor's comment: you may need to press Ctrl+F5 once the page loads to see the latest documentation.

seyyah, 2019-06-20:
> What do you think?

@kisvegabor It great works.

Links don't work,

- https://littlevgl.com/sphinx/
- https://littlevgl.com/sphinx

 Link to document can be added in the https://github.com/littlevgl/docs/tree/sphinx repo.

kisvegabor, 2019-06-21:
@seyyah thank you! It's just a temporal location. It will be moved to docs.littlevgl.com

kisvegabor, 2019-06-21:
The current state of the documentation is:
- All object types are added. The first half is reviewed and the examples are updated. I created very simple examples to make it easy to convert to MicroPython. 
See: https://littlevgl.com/sphinx/object-types/bar.html#example (Refresh with Ctrl + F5 might be required)
- Almost all the overview sections are ready.
- It'd be great to write something about MicroPython support and its simulator. @amirgon, @embeddedt What is the current state of the simulator? Can we publish it next week? If it still needs some polishing we can add this section later.

I'll be on holiday at the weekend but probably will finish the docs on Tuesday.

amirgon, 2019-06-22:
> * It'd be great to write something about MicroPython support and its simulator. @amirgon, @embeddedt What is the current state of the simulator? Can we publish it next week? If it still needs some polishing we can add this section later.

@kisvegabor I've committed some changes to the binding, it works well on the unix port.
[I wasn't able to run the Emscipten port](https://github.com/littlevgl/lvgl/issues/792#issuecomment-504595940), perhaps @embeddedt could help.

I suggest adding links to the online simulator in the docs, for each example, I think it can be very useful.  

I can help with Micropython documentation and examples. There is a lot of information in https://github.com/littlevgl/lv_binding_micropython/blob/dev-6.0/README.md, I don't think we need to duplicate that.  
In the docs themselves I suggest adding a link to this README and maybe add some basic Micropython explanations and examples.

embeddedt, 2019-06-23:
@kisvegabor 

> I'll be on holiday at the weekend but probably will finish the docs on Tuesday.

Hope you're having a good time!

> What is the current state of the simulator? Can we publish it next week?

I'm sure we can have it ready by the end of this week.

kisvegabor, 2019-06-24:
@amirgon 
> I suggest adding links to the online simulator in the docs, for each example, I think it can be very useful.

I agree, it would be great. If you sent a link I insert it to `index.rst` to see it in the practice.

> I can help with Micropython documentation and examples. 

Awesome, thank you!

There is a lot of information in https://github.com/littlevgl/lv_binding_micropython/blob/dev-6.0/README.md, I don't think we need to duplicate that.
In the docs themselves I suggest adding a link to this README and maybe add some basic Micropython explanations and examples.

I also meant it for an overview page with sections like: 
- What is MicroPython? (from the blog post?)
- Why use it?  (from the blog post?)
- The simulator
- A very simple 5 line example.
- A more Pythonic example
- Try on PC with the unix port (link to readme?)
- Something about the ESP port (link to readme?)

What do you think?

@embeddedt 
> Hope you're having a good time!

Thank you, it was great! :) 

> I'm sure we can have it ready by the end of this week.

Awesome!

amirgon, 2019-06-24:
> I also meant it for an overview page with sections like:
>
> * What is MicroPython? (from the blog post?)
> * Why use it?  (from the blog post?)
> * The simulator
> * A very simple 5 line example.
> * A more [...]

@kisvegabor Sounds good.
I'm not sure we can do everything this week (there are a lot of examples), but we can start.  
I don't think you must wait for all the docs before you release v6.0, we can continue the work afterwards.

btw I'm not sure about the sphinx syntax - I think the C example is gone now on https://littlevgl.com/sphinx/object-types/bar.html#example

kisvegabor, 2019-06-25:
> I'm not sure we can do everything this week (there are a lot of examples), but we can start.
I don't think you must wait for all the docs before you release v6.0, we can continue the work afterwards.

It's not a problem if we add Micropython examples later. I wanted to have at least one C example for every object type, the other examples can come later.

We can add new C examples later too. Actually, it would be the best if users would contribute with examples. Later we can make the [examples](https://github.com/littlevgl/docs/tree/sphinx/examples) folder to submodule from the `lv_examaples` repo.

Finally, I finished the docs (except the overview of the Images). Python Bar example is working too.
https://littlevgl.com/sphinx/object-types/bar.html#example
  
What is still missing:
- a better welcome page with some links
- update the header file comments for Doxygen (use `/** comment */`). But it's not urgent.

embeddedt, 2019-06-26:
> a better welcome page with some links

The [welcome page](https://littlevgl.com/sphinx/index.html) already looks great to me! Is there something else you wanted to add?

> update the header file comments for Doxygen

As far as I can tell, most of the header files are already Doxygen-enabled (i.e. [`lv_core/lv_disp.h`](https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_core/lv_disp.h)). Could you please clarify?

kisvegabor, 2019-06-27:
> The welcome page already looks great to me! Is there something else you wanted to add?

I've updated it yesterday so it's already the new, improved welcome page.

> As far as I can tell, most of the header files are already Doxygen-enabled (i.e. lv_core/lv_disp.h). Could you please clarify?

It's mainly for the enums and structs. I updated [lv_btn.h](https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_objx/lv_btn.h) to show what I mean. This way there are comments in the [API doc](https://littlevgl.com/sphinx/object-types/btn.html)

kisvegabor, 2019-06-27:
@amirgon 
I'm creating the release notes of v6.0. Could you write three lines about the MicroPython binding? Like: "Now LittlevGL supports ...", "You find it here and here...", "Learn more here...", ("Simulator...")

embeddedt, 2019-06-27:
> It's mainly for the enums and structs.

I will try and review all the header files today then.

kisvegabor, 2019-06-27:
> I will try and review all the header files today then.

Okay, but we don't need to hurry with that. It can be done in a convenient pace after the release too.

embeddedt, 2019-06-27:
@kisvegabor Did you upload the `Doxyfile` that you use with Sphinx?

kisvegabor, 2019-06-27:
> Did you upload the Doxyfile that you use with Sphinx?

I uploaded it here: https://github.com/littlevgl/lvgl/blob/dev-6.0/scripts/Doxyfile

> Also, how do you want to handle documenting the enums, since the enum and typedef are split?

I just documented the enum values like in [lv_btn.h](https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_objx/lv_btn.h#L42)

> lvgl/src/lv_core/lv_group.h

What is the question here? Maybe I just automatically understand my non-sense comments :smile:

embeddedt, 2019-06-27:
@kisvegabor Should the Doxygen comment for `LV_KEY_UP` just be `0x11`? That seems like it won't be very helpful for the majority of users.

kisvegabor, 2019-06-27:
> That seems like it won't be very helpful for the majority of users.

It's really not so helpful. In these simple cases, I think the name and value are enough.

embeddedt, 2019-06-27:
Is this still planned?

https://github.com/littlevgl/lvgl/blob/fc42d02e970a9d5e63bdb0f3cd21cf619d8dbbcb/src/lv_core/lv_style.h#L57

kisvegabor, 2019-06-27:
No, it isn't.

embeddedt, 2019-06-27:
@kisvegabor Okay, I replaced it with a more descriptive comment.

embeddedt, 2019-06-27:
All of the header files inside the `littlevgl/lvgl` should now have been updated with proper Doxygen comments.

I have also updated the `sphinx` branch.

kisvegabor, 2019-06-28:
@embeddedt Amazing! Thank you very much! 
Have you intentionally skipped the comments of `lv_..._ext_t` variables?

It seems you made sphinx work which is also amazing without any documentation about what needs to be installed. :clap: :clap: :clap: 

As setting up the sphinx environment was an iterative process for me I don't know what really needs to be installed. Do you still remember or could you check it in command line history and them to the readme?

embeddedt, 2019-06-28:
@kisvegabor I updated the `README.md` file. https://github.com/littlevgl/docs/tree/sphinx

Let me know if anything is unclear.

kisvegabor, 2019-06-28:
Looks perfect! Thank you very much!

What about this question:
> Have you intentionally skipped the comments of lv_..._ext_t variables?

embeddedt, 2019-06-28:
Yes, because normally users would use the APIs to control them so they shouldn't care about the contents.

kisvegabor, 2019-06-28:
> Yes, because normally users would use the APIs to control them so they shouldn't care about the contents.

I agree. If someone really needs those fields to develop something custom he will look into the header files anyway.

kisvegabor, 2019-06-30:
The English version of the new documentation is ready and published at https://docs.littlevgl.com/

I'm sure that there are some typos in it but they should be fixed later as we see them.

Thank you very much for your contribution!

If you are interested in translating the documentation please go to https://github.com/littlevgl/docs/issues/21

closed 2019-06-30 by kisvegabor