lvgl/lvgl #1102: Behaviour of lv_roller_set_selected() function.

pete-pjb, 2019-06-13:
Hi,

I am unsure if this is the intended behaviour or not, but the `lv_roller_set_selected()` function does not generate a _LV_EVENT_VALUE_CHANGED_ or _LV_EVENT_SELECTED_ event.  I assumed that if would.  Can you confirm if this is the case please?

Kind Regards,

Pete

kisvegabor, 2019-06-14:
Hi,

it wasn't intentional. I fixed it for `roller` and `ddlist`.

I also removed `LV_EVENT_SLECTED` and used `LV_EVENT_VALUE_CHANGED` instead.

pete-pjb, 2019-06-14:
Hi @kisvegabor ,

I can now confirm I am getting the expected behaviour when calling `lv_roller_set_selected()` function.  Thank you!

Kind Regards,

Pete

closed 2019-06-14 by pete-pjb

kisvegabor, 2019-06-20:
Hi @pete-pjb 

I tried to apply the same behavior to other objects too but it became chaotic. For consistency, I reverted this for roller too.

I can say these events are sent by the user when he is using the UI and not from the code. However, you can still use `lv_event_sent(roller, LV_EVENT_..., NULL);` to manually send an event if required.`

pete-pjb, 2019-06-20:
Hi @kisvegabor,

Okay no problem I will change my code to send the event manually. 

Kind Regards, 

Pete

MauricioEmilio, 2024-02-10:
Hello, I'm using graphic UI library with a roller, and I'm calling the function lv_roller_get_selected from main code, the result is I'm getting a high integer like 16535 all the time no matter what the position of the roller actually have in execution time, also I can't set a manual selection using lv_roller_set_selected called from main.