lvgl/lvgl #1105: [dev-6.0] Image caching when using the image decoder

amirgon, 2019-06-19:
[Related forum thread](https://forumtest.littlevgl.com/t/using-the-image-decoder/81).

Image decoder calls `open_cb` multiple times in order to display a single image. In each case the decoder decodes the entire image.  
This is even worse with animation - on every frame `open_cb` is called multiple times.  

A non trivial image format (PNG or JPG for example) takes time to decode.
On architectures with enough ram to store a decoded image, it does not make sense to decode the same image again and again.

### I suggest adding a caching logic into lvgl img to decode an image only once, until a different image is set (by `lv_img_set_src`).

Every time a new image is set (by `lv_img_set_src`), the cache should be invalidated.

When testing this, my experience shows a very significant performance improvement when the decoded image is cached.

kisvegabor, 2019-06-19:
I've just realised that caching this way is not working well with more than one image. Let's say there are 2 pcs  200x200 PNG images. The VDB is set to 20 lines and now the whole screen is being refreshed. In this case, when drawing the first chunk (first 20 lines of the display) both images needs to be decoded. Therefore caching is pointless with more than one image.

If you have enough RAM you can call `lv_img_decoder_open` in your code to get the decoded image and build an `lv_img_dsc_t` variable with the decoded data. It would give the biggest flexibility.

Or you can use a screen sized VDB. In would ensure that an image can be drawn with one open/close because the screen always will be refreshed in one big chunk. However, the images still need to be decoded on every new frame...

amirgon, 2019-06-19:
@kisvegabor With enough ram why not cache multiple images? We can set a threshold on the number of images (or number of KB) we would like to reserve for the cache.  

> If you have enough RAM you can call `lv_img_decoder_open` in your code to get the decoded image and build an `lv_img_dsc_t` variable with the decoded data. It would give the biggest flexibility.

You don't need lvgl image decoder if all you want is to decode the image beforehand and set_src the decoded image.  
More generally - What is the use case of lvgl image decoder? Is anyone using the image decoder today? If yes, how did they handle this problem?   

My point is that **I'm not sure image decoder is usable at all** for any practical purpose if the same image is decoded multiple times. It would just become too slow and waste significant amount of cycles, to the point where I'm not sure anyone would want to use it.

kisvegabor, 2019-06-19:
> More generally - What is the use case of lvgl image decoder? Is anyone using the image decoder today? If yes, how did they handle this problem?

1. They can be used to decode any custom image format. They don't have to be as complex as a PNG or JPG. E.g. a simple RLE coding or so.
2. The image decoder is an abstraction. You can buffer/cache images as you want. E.g. if you have RAM to store 10 uncompressed icons you can cache the last 10 decompressed icons but you can have 100 PNG icons in the flash each stored as an `lv_img_dsc_t` variable.  I really didn't consider the situation when an image uses a user created `lv_img_dsc_t` and the user "silently" changes the image data. I think it's a special case but of course, I'm open to improve it. However, as it's related to how buffering works in the decoder, maybe a custom function clear the cache or drop only one image would be the clearest solution. 

> Is anyone using the image decoder today?

I don't know.  :)

amirgon, 2019-06-19:
> 1. They can be used to decode any custom image format.

In my opinion, a decoder without cache would have very narrow usability.  

It would only be used for the simplest encodings where decoding is so fast that you don't care it's called many times for the same image.  
But I don't think it would be practical to use an image decoder without caching for any non trivial encoding (PNG, JPG, etc.).

> 2\. The image decoder is an abstraction. You can buffer/cache images as you want

@kisvegabor I agree that caching could be implemented either on the image decoder or on lvgl. But does it make sense to implement caching multiple times for each image decoder?  

Imagine - that in the future you have multiple decoders for various image formats. Would each decoder implement its own caching? Would the user need to configure caching parameters individually for each decoder? (enable/disable and threshold)  
Why not implement caching only once in lvgl?  then the user would have a centralized control over the caching configuration.

> I really didn't consider the situation when an image uses a user created `lv_img_dsc_t` and the user "silently" changes the image data.

If you insist not implemeting caching in lvgl, and implement it in each decoder instead, I still think that the trigger for invalidating the cache for an individual image should be a call to `lv_img_set_src`.  
The simplest implementation would just require `lv_img_set_src` to clear some flag that could be set and examined by the decoder.

kisvegabor, 2019-06-19:
I see your points. A cache like "don't close the last X images" can work. The cache entries can be timestamped and always the oldest would be overwritten.

However, it would be very unpredictable regarding memory usage. Let's say "X=3" so the last three images are left open. If the last three images are simple built images then the cached memory size is 0 bytes. But if three PNG images then it could be huge. It would be nice to have a memory limit besides a number of opened images but it's not trivial to tell the memory used by an opened image. From the above example, it's clear that the used memory is not related to the image size.

So if you agree, I can easily add this caching mechanism. If the user wants his own caching then he can set "X = 0".

The cache clean on `lv_img_set_src` should be optional because typically `lv_img_dsc_t`s are not modified in runtime. Thus cleaning the cache on every `lv_img_set_src` would cause unnecessary cache cleans.

What do you think?

amirgon, 2019-06-19:
> However, it would be very unpredictable regarding memory usage.

I agree. I also think that memory limit is better than image count.  
What if we let the user configure the cache size in KB and track image sizes? built in images would consume 0 KB, while other images would consume according to `data_size` in `lv_img_dsc_t`. The caching logic would cache an image only when adding its size to the total (current) cache size does not exceed the configured cache size.

> The cache clean on `lv_img_set_src` should be optional because typically `lv_img_dsc_t`s are not modified in runtime. Thus cleaning the cache on every `lv_img_set_src` would cause unnecessary cache cleans.

I'm not sure I understand this. Why would it cause unnecessary cache cleans? Usually image data is only set once using `lv_img_set_src` per image. If you call `lv_img_set_src` twice for the same image, it probably means you changed the image source, isn't it?  
Why not update the "invalidate cache" flag in `lv_img_decoder_dsc_t` every time `lv_img_set_src` is called?  
`lv_img_decoder_dsc_t` is kept per image per decoder right? so we would know when image data has changed.
This would allow a decoder to know if it could use the cache or not, in case we decide implementing the cache in the decoder. And the same flag could be used for the generic caching logic as well, when it's 
enabled.  

Does it make sense?

kisvegabor, 2019-06-19:
> ... while other images would consume according to data_size in lv_img_dsc_t. The caching logic would cache an image only when adding its size to the total (current) cache size does not exceed the configured cache size.

It's not entirely true. Imagine that a PNG file is decoded from an SD card. E.g `lv_img_set_src(img, "S:pic.png")`. In this case, lodepng first reads the whole file, allocates the output buffer and allocates other buffers to process the image. So what if you let 100kB for caching and 20 kB left. The image is 19kB so it seems ok, but a lot more memory can be allocated during decoding.

Or if `decocder_line_read_cb` is used probably the image doesn't need to be fully decoded. Maybe a line can be returned immediately. Maybe the previous line is also required to decode (e.g. they are XORed). So there are even more unknown things here.

Or when built-in images are loaded as files there is nothing to cache because they can be read directly line-by-line, so it seems 0 kB cache is used. The library can think that any number of images from files with built-in color format can be opened. But if the file is left open then it also consumes memory.

In light of these uncertainties, I think memory limit can't work. It's even difficult to predict the used memory in the decoder.

IMO it might be the user's responsibility to ensure that he will have enough memory even if the 3 largest images are loaded at the same time. (assume he allows 3 images to be cached)

> I'm not sure I understand this. Why would it cause unnecessary cache cleans? Usually image data is only set once using lv_img_set_src per image. If you call lv_img_set_src twice for the same image, it probably means you changed the image source, isn't it?

You can use the same image source in multiple image objects. It's quite common. Think about icons or any type of symbols. For example, a water system which has a lot of valves. In this case, only two image sources are used (`valve_open` and `valve_close`) but for many valves.  Cleaning the cache always when a valve changes state wouldn't be reasonable because we still have the same two images.

Or imagine a file system browser with file and folder icons. It's the same case. 

So when happens this issue? 
1. You have an `lv_img_dsc_t` variable (call it `dsc`)
2. Store some image data in  `dsc->data` field
3. Use the `dsc` in an image object with `lv_img_set_src`
4. Change the content of `dsc->data` to an other image
5. Set the image again with `lv_img_set_src`
6. And the cache is not invalidated because dsc is still the same and the image is not updated.

So my question is once you already work with `dsc->data` why is it a big deal to call an function like `lv_img_decoder_invalidate_src(&dsc)`? It seems quite straight forward to me. You need to notify the library if you change an alredy used resource.

amirgon, 2019-06-20:
> So my question is once you already work with `dsc->data` why is it a big deal to call an function like `lv_img_decoder_invalidate_src(&dsc)`? It seems quite straight forward to me. You need to notify the library if you change an alredy used resource.

@kisvegabor I see your point. I didn't think of the case of using the same image source on multiple image objects, but as you put it, it is probably very common.  

So I'm convinced the best solution is, as you suggested, expect the user to call an `invalidate_src` function.  

I noticed you called the function "`lv_img_decoder_invalidate_src`".
Does it have to be a decoder function? Suppose the user is not aware of the decoder, he is just setting sources to images and those sources could be either PNG or JPG images. Maybe the user is not even aware whether the image source he is holding is PNG or JPG. To invalidate the cache, if it's a decoder function, we would need to associate an image source with the right decoder.  
Or maybe you meant that this would be some global function unrelated to specific decoder?

kisvegabor, 2019-06-20:
> Or maybe you meant that this would be some global function unrelated to specific decoder?

Yes. For example
- `lv_img_decoder_invalidate_src(&my_dsc)`
- `lv_img_decoder_invalidate_src("S:pic.png")`

The image drawing module can have an array to hold the opened `lv_img_decoder_dsc_t` variables. `lv_img_decoder_invalidate_src` would check the decoder dscs in the array and if a dsc's source matches it closes that one. 

It's really not the best name as the cache should be managed by the drawing module. `lv_img_draw_invalidate_src` doesn't sounds much better. Do you have any idea?

Anyway, independently from the name it seems we agreed on how this feature should work. Or is there anything else?

amirgon, 2019-06-20:
> if a dsc's source matches it closes that one.

@kisvegabor Closing an image is invalidating it? Will you keep images open until they are invalidated? (because I think that today they are repeatedly opened and closed).  
Will you call `open` only once for an image, until it's invalidated (or deleted)? 

> Anyway, independently from the name it seems we agreed on how this feature should work. Or is there anything else?

Yes, I think we agree.  
Regarding the name, how about `lv_img_invalidate_src`?

kisvegabor, 2019-06-20:
> Closing an image is invalidating it?  Will you keep images open until they are invalidated?  Will you call open only once for an image, until it's invalidated (or deleted)?

Yes, for all of them. :)

> Regarding the name, how about lv_img_invalidate_src?

It might give the false impression of this function is related to the image object, however, is something more generic and lower level.

embeddedt, 2019-06-20:
@amirgon @kisvegabor If the function is related to the decoding logic, then I think for consistency, it's best to call it `lv_img_decoder_invalidate_src`. It makes it clear that it relates to the decoding logic.

Someone who isn't aware of the decoder can still use the function when they see it written in the documentation.

kisvegabor, 2019-06-20:
I implemented it in [lv_img_cache.c](https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_draw/lv_img_cache.c). When it opens an image it measures the time to open and try to keep the slow to open images in the cache. I added a life time parameter which means images used earlier than this time are considered unused and more probable to overwritten in the cache.

I thought it's worth to be in a new module so I used the `lv_img_cache_invalidate_src` name.

The default cache size and life time can be set in `lv_conf.h` but can be changed in run time too.

What do you think?

@amirgon Can you test the performance?

amirgon, 2019-06-20:
@kisvegabor I didn't check it yet, but I noticed one problem:

https://github.com/littlevgl/lvgl/blob/a9c818a6efa7762202489e6ef7bffbfebb053e07/src/lv_draw/lv_img_cache.c#L27

`img_cache` must be declared as gc root, like the global linked lists, otherwise it will be collected by gc prematurely.

kisvegabor, 2019-06-20:
@amirgon 
I've fixed it. See ca1f21a
I'm not sure if it's valid: `lv_img_cache_entry_t * cache = LV_GC_ROOT(_lv_img_cache_array);`

I also updated the caching policy which decides which cache entry to (re)use. I removed the `life time` parameter because it's not used in the new algorithm. 

It works like this:
- every cache entry has a `life` and a `time_to_open` parameter
- every time `lv_img_cache_open` is called the life of all cache entries is decreased by 1. (make them older)
- if the image is found in the cache it's life is increased by `time_to_open` to make it "more alive"
- if there is no more space in the cache the entry with the smallest life will be reused. (the oldest)

This way the images which are more difficult to open or used frequently will live longer.

amirgon, 2019-06-21:
@kisvegabor It works great! Using the decoder with large enough cache, gives the same user experience as using native `TRUE_COLOR_ALPHA` image directly.  

I only checked performance with two images, though. I did not check how the cache is behaving when dynamically adding more images, whether aging behaves correctly etc.

Maybe we need to think of a test bench. It's not trivial to check it right.

kisvegabor, 2019-06-21:
@amirgon 
> It works great! Using the decoder with large enough cache, gives the same user experience as using native TRUE_COLOR_ALPHA image directly.

Glad to hear that! :) 

> I only checked performance with two images, though. I did not check how the cache is behaving when dynamically adding more images, whether aging behaves correctly etc.

I was testing it with the `demo` UI using its bubble wallpaper, the dices PNG image, and a mouse cursor.
So three images: 2 built-in, 1 PNG. I set the cache size to 2 and moved the cursor over the dices. This way all 3 images were drawn but the PNG was opened only once and the other two were opened/closed continuously. So basically it seemed working.

> Maybe we need to think of a test bench. It's not trivial to check it right.

Yes, a more advanced test would be great but it's really not trivial. 

To have a predictable test we should use `lv_img_cache_open` directly instead of clicking/dragging the images.  What do you think can we test it with 3 images and cache size = 2? 
We can create 3 dummy image decoders (for the three images) with three different delays inside their open function, call `lv_img_cache_open` a lot of times by pseudo-randomly selecting an image to open and finally check which two images are cached.

amirgon, 2019-06-21:
> Yes, a more advanced test would be great but it's really not trivial.
>
> To have a predictable test we should use `lv_img_cache_open` directly instead of clicking/dragging the images. What do you [...]

Yes as you said - testing it right is not going to be trivial.  
We can try what you suggest.

amirgon, 2019-06-22:
@kisvegabor 

https://github.com/littlevgl/lvgl/blob/d79df213a16d2e18de1c2e074c31de3855568150/src/lv_draw/lv_img_decoder.h#L40

Why didn't you defined `LV_IMG_DECODER_OPEN_FAIL` as `NULL`?

Casting a pointer to -1 is awkward and causes a problem in the Micropython binding.
I support converting `None` to `NULL`, but I would like to avoid converting any integer value to pointer, as this would cause a segfault if someone puts integer instead of a pointer by mistake.

amirgon, 2019-06-23:
@kisvegabor Here is another idea for the image decoder:  

Image loading might be slow. If an image is read from the network - could be very slow.  
Today everything is synchronous - everything is blocked until the image is loaded.   
**How about supporting asynchronous image loading?**

Something like this:
- `open_cb` reports it started downloading the image asynchronously, and returns.
- lvgl displays some placeholder (spinner?) instead of the image
- When the image completed downloading, the system signals this to lvgl
- lvgl replaces the image placeholder by the image.

What do you think?

kisvegabor, 2019-06-24:
> Why didn't you defined LV_IMG_DECODER_OPEN_FAIL as NULL?

It's really awkward. I updated it. Not it returns `LV_RES_OK/INV` and the image data is set in dsc->imd_data. 

> How about supporting asynchronous image loading?

Sounds interesting. I've found an easy way to support it. Now, if the image can't be drawn the "No data" text is displayed as a placeholder. I added a the `dsc->error_msg` field to set a custom placeholder.

If the `error_msg` is not NULL then it will be displayed instead of the image. The `decoder_open` function still needs to return `LV_RES_OK`.

When the image arrives you can invalidate the cache entry and now open the image normally.

amirgon, 2019-06-24:
> I added a the `dsc->error_msg` field to set a custom placeholder.
> 
> If the `error_msg` is not NULL then it will be displayed instead of the image.

@kisvegabor What if I want to display a spinner or another image as a placeholder, instead of a label?

> > (btw, it would require a change in the way you measure opening time in the caching logic)
> 
> I don't think so. Open time still shows how fast it was to open the image. If it has placeholder, it's fast to open and it's still valid to say "this image was easy to open, let's cache something more complicated instead".

When the image arrive from the network and I invalidate the cache to display it, how would you know how much time it took to fetch it from the network? I would want it to have high score in the cache because it took very long to fetch it from the network.

kisvegabor, 2019-06-25:
> What if I want to display a spinner or another image as a placeholder, instead of a label?

A spinner would be too complex. A label simply can be drawn on the fly (instead of the image) which consumes no extra memory and means no extra complexity (like an addition animation).  So for now, I'd like to keep the text version. 

> When the image arrive from the network and I invalidate the cache to display it, how would you know how much time it took to fetch it from the network? I would want it to have high score in the cache because it took very long to fetch it from the network.

I see. I updated the image cache module. Now you `time_to_open` is part of the `decoder_dsc`. In your `decoder_open` you can set `time_to_open` to any value. If left `0`, LittlevGL set the measured time.
It seems quite flexible and easy to use for me. What do you think?

Here is the updated png decoder: 
[png_decoder.zip](https://github.com/littlevgl/lvgl/files/3325853/png_decoder.zip)

amirgon, 2019-06-25:
> A spinner would be too complex.

@kisvegabor could you explain the difficulty? Isn't it just another lvgl object that is instantiated in such occasion, and occupies the same coordinates?

> Here is the updated png decoder:
> [png_decoder.zip](https://github.com/littlevgl/lvgl/files/3325853/png_decoder.zip)

Why don't you add it to github, instead of zip? I would suggest using lodepng repo as git submodule.  

btw, I already have this on `lv_binding_micropython/driver/png/`, it's built by `lv_micropython` and I have a working pure-python decoder (that uses lodepng) on https://github.com/littlevgl/lv_binding_micropython/blob/dev-6.0/examples/example2.py

embeddedt, 2019-06-25:
I think instantiation of another object may be the complexity.

amirgon, 2019-06-25:
> I think instantiation of another object may be the complexity.

@embeddedt Why is this a problem? You can have a function pointer to a "default placeholder" init function. That function would create an configure the placeholder object.  
If the user wants a different placeholder, all he needs to do is to implement such function and register it in that function pointer. (You may need also a "deinit" function to free resources when the placeholder is removed).  

I think it would be nice if the default placeholder displayed a spinner, but it could display a label by default if you think it's better, as long as the user has the power to configure a different placeholder.

kisvegabor, 2019-06-26:
> I think it would be nice if the default placeholder displayed a spinner, but it could display a label by default if you think it's better, as long as the user has the power to configure a different placeholder.

IMO loading images from the web is a very special case so adding generic support for it would be overkill.
In addition, the `placeholder_init` function would be called when drawing is in progress and you can't create new object while drawing because it would mess up all the previous optimizations and calculations about the process of rendering.

However, it seems you to me you can do it manually:
1. Create an image
2. Create a preloader object on it
3. When the image has arrived remove the preloader.

kisvegabor, 2019-06-27:
@amirgon 
I'm trying to close the pending issue before the release. As I see the original and important issue has been solved by the implementation of caching.

We can continue the discussion about the placeholder if you don't like the above propoesed solution but it's already independent from caching itself. So if you'd like to improve the placeholder, please open an other issue for it.

And thank you for the contribution, caching was really a missing feature! 

BTW, I've written about it here: https://littlevgl.com/sphinx/overview/image.html#image-caching

closed 2019-06-27 by kisvegabor