lvgl/lvgl #1114: Stack usage concerns

embeddedt, 2019-06-23:
@kisvegabor There are a few places in LittlevGL where we are using >= 1024 bytes of stack in a single function. This concerns me, since not everyone's platform may be configured with a large stack by default.

(These measurements are with `LV_HOR_RES_MAX` and `LV_VER_RES_MAX` set to 480 and 320, respectively.)

--------------------------------------

Around 1 kilobyte of stack is used by each of these two functions.

https://github.com/littlevgl/lvgl/blob/7f7034f50fd62b3a3335e72510e129dab4cc3745/src/lv_draw/lv_img_decoder.c#L366

https://github.com/littlevgl/lvgl/blob/7f7034f50fd62b3a3335e72510e129dab4cc3745/src/lv_draw/lv_draw_img.c#L473

I think too much of the stack is being used here. IMO this should either be dynamically allocated with `malloc`, or, even better, since it's only running on one thread at a time, statically allocated.

----------------

`lv_draw_shadow_bottom` uses 1 kilobyte of stack.

`lv_draw_shadow_full` uses over 3 kilobytes of stack. I think this is too much for the typical embedded stack - which could only be 1 or 2 kilobytes (less in some environments).

I'm not quite sure how to optimize these - there are several VLAs in use that can't be allocated statically. Possibly dynamic allocation is an option here?

C47D, 2019-06-23:
Can the stack usage be profiled somehow? Maybe there are more spots that uses lots of stack memory.

If those large arrays are allocated inside functions maybe you can move them outside those and made them file scoped by adding the ```static``` qualification. As for VLAs, i think you can avoid them by allocating the 'worst possible' scenario, the linux kernel is VLA as far as i know, so lvgl can be also free of those.

I'm surely suggesting a naive thing but i'm interested to know how this will be solved.

kisvegabor, 2019-06-24:
>  lv_color32_t palette_tmp[256]; 

I was able to improve it. The palette dynamically allocated later so I've just used that dynamically allocated memory instead of a local variable.

>  uint8_t buf[LV_HOR_RES_MAX * ((LV_COLOR_DEPTH >> 3) + 1)]; /*+1 because of the possible alpha byte*/ 

I can't see a good solution here. However, it is used only when
- image is not `TRUE_COLOR`
- `LV_COMPILER_VLA_SUPPORTED == 0` 

> lv_draw_shadow_bottom uses 1 kilobyte of stack.
lv_draw_shadow_full uses over 3 kilobytes of stack. I think this is too much for the typical embedded stack - which could only be 1 or 2 kilobytes (less in some environments).

These are also only when `LV_COMPILER_VLA_SUPPORTED = 0`, right?

IMO, the `LV_COMPILER_VLA_SUPPORTED == 0`is not a big issue. All embedded C compilers what I saw so far supported it. I added this option because of Visual Studio.

As @C47D suggesteed we can move these variables out of the functions but only if VLA is not supported. With VLA consuming extra 3..4 kB RAM continuously is worse IMO. 

What do you think?

@C47D 
> Can the stack usage be profiled somehow? Maybe there are more spots that uses lots of stack memory.

I don't know but it'd be an interesting experiment.

embeddedt, 2019-06-24:
@kisvegabor Is it correct to say that when VLAs are supported the array uses the width/height of the object instead of `LV_[HOR/VER]_RES_MAX`? If so then I suppose that is acceptable.

> we can move these variables out of the functions but only if VLA is not supported. 

I agree.

C47D, 2019-06-24:
I did checkout the use of:
https://github.com/littlevgl/lvgl/blob/7f7034f50fd62b3a3335e72510e129dab4cc3745/src/lv_draw/lv_img_decoder.c#L366

And it's only used when lvgl is configured to use a file system (`LV_USE_FILESYSTEM != 0`). According with the function where it is allocated (`lv_img_decoder_built_in_open`), this function open an image file and tries to decode it (this is when `buf` is created), if it can't decode it, it will return NULL and the image will be decoded line by line using the `decoder_read_line` function pointer. If i understand correctly the default behavour can be decode the image line by line so `buf` allocation can be avoided?

Please let me know if i am misunderstanding, i'm in a break at job (:

embeddedt, 2019-06-24:
@C47D Decoding the image line by line is probably much slower - it would be more ideal to statically or dynamically allocate the buffer, since LittlevGL is single-threaded.

embeddedt, 2019-06-24:
@C47D 

> We can build lvgl with the -Wvla compiler switch to warn on VLA usage so we can try to avoid the, as much as possible.

Basically, we either use the stack for dynamic allocations (and expect people to increase their stack size) or we use the heap. Considering that the heap has overflow checking while the stack doesn't, using the heap is probably safer.

kisvegabor, 2019-06-25:
@C47D 
>  If i understand correctly the default behavour can be decode the image line by line so buf allocation can be avoided?

The behavior depends on 
- the color format: ALPHA and INDEXED images are read line-by-line but for TRUE_COLOR images only a pointer is returned if stored in the flash
- every image stored as a file (with built-in format) is read line-by-line to avoid buffering the whole image in RAM.

@embeddedt 
> Basically, we either use the stack (VLA) for dynamic allocations (and expect people to increase their stack size)

Using the heap people need to increase the heap size, isn't it?. 
However, usually, the minimum stack size is defined. So if your RAM is not full probably you don't have to do anything.

> Considering that the heap has overflow checking while the stack doesn't, using the heap is probably safer.

Is `lv_mem_alloc` really faster than allocating something in the stack? I don't know for sure but I think a VLA is just jumping the stack pointer.

So for no VLA I think using `static` variables is a good solution. But do you think we should remove VLAs at all?

embeddedt, 2019-06-25:
My concern is simply that people aren't likely to be using/expecting a 5 kilobyte stack. We need to have some way of capping the amount of stack we'll use... VLAs make that difficult. 

Allocating on the heap is slower, but safer.

kisvegabor, 2019-06-25:
> My concern is simply that people aren't likely to be using/expecting a 5 kilobyte stack. We need to have some way of capping the amount of stack we'll use... VLAs make that difficult.

True, it really could be an issue. However, we should take into account that nobody concerned about stack overflow so far. (Except some people with FreeRTOS but I think it's a different thing).  Probably it's because VLA is usually supported by the embedded compilers. 

Anyway, the biggest VLAs are the shadow related buffers and [file read buffer](https://github.com/littlevgl/lvgl/blob/7f7034f50fd62b3a3335e72510e129dab4cc3745/src/lv_draw/lv_draw_img.c#L473) for images, right?

For shadow buffers, we can always reallocate a buffer if the current is too small. This way only the required size will be used. (Shadow is usually < 10 px so the buffer will be < 100 bytes)

We can do the same thing with the file buffer. Its max size is `4 * LV_HOR_RES_MAX` so 1-2 kB.

We can even generalize this thing to have a function like `lv_draw_get_tmp_buf(size)` which reallocates the buffer if it's current size is too small. Now one buffer is enough, but a function like this can hide multiple buffers.

What do you think?

C47D, 2019-06-25:
> My concern is simply that people aren't likely to be using/expecting a 5 kilobyte stack. We need to have some way of capping the amount of stack we'll use... VLAs make that difficult.
> 
> Allocating on the heap is slower, but safer.

Just found [this](https://wiki.sei.cmu.edu/confluence/display/c/ARR32-C.+Ensure+size+arguments+for+variable+length+arrays+are+in+a+valid+range) link which explain some problems of using VLAs. Dinamically allocated memory can be reallocated with realloc, while VLAs doesn't, the heap memory is usually bigger than stack.

kisvegabor, 2019-06-26:
> Just found this link which explain some problems of using VLAs. Dinamically allocated memory can be reallocated with realloc, while VLAs doesn't, the heap memory is usually bigger than stack.

Okay, we can update the code to remove VLA. What about my proposal in the above [comemnt](https://github.com/littlevgl/lvgl/issues/1114#issuecomment-505518387)?

embeddedt, 2019-06-26:
@kisvegabor It should be possible to just dynamically allocate both of the buffers you mentioned at the top of the functions. Is that the same idea you were saying? Sorry, I was a bit confused.

kisvegabor, 2019-06-26:
@embeddedt To better demonstrate what I mean I implemented if for shadow drawing. See  071bf4c

embeddedt, 2019-06-26:
@kisvegabor This is great! But it has one flaw. Once the buffer has grown it is never shrunk again for the lifetime of the system's operation.

Imagine that for some reason the buffer became 1 kilobyte due to a large object, and then that object was deleted. The draw buffer would still consume 1 kilobyte for the program's lifetime.

One solution could be to free the draw buffer at the end of `lv_refr_task`, or at least shrink it to a more reasonable size.

kisvegabor, 2019-06-26:
> Once the buffer has grown it is never shrunk again for the lifetime of the system's operation.

That's true.

> One solution could be to free the draw buffer at the end of lv_refr_task, or at least shrink it to a more reasonable size.

Sounds good. It shouldn't affect the performance too much.

Added here: f806f8c

kisvegabor, 2019-06-26:
I've replaced all VLAs here: https://github.com/littlevgl/lvgl/commit/b04367c702f9a07f96518ad461b4856c49162f6f

[Here](https://github.com/littlevgl/lvgl/commit/b04367c702f9a07f96518ad461b4856c49162f6f#diff-dd44be118be5a3842537d69d2f2aa0d8L574) and [here](https://github.com/littlevgl/lvgl/commit/b04367c702f9a07f96518ad461b4856c49162f6f#diff-dd44be118be5a3842537d69d2f2aa0d8R658) I kept the local variables because the draw buffer is already used [here](https://github.com/littlevgl/lvgl/commit/b04367c702f9a07f96518ad461b4856c49162f6f#diff-16cdd0d23c785e1aa11d13255bf67756R472).

embeddedt, 2019-06-26:
@kisvegabor Great! I will test it with `-Wstack-usage` again when I get a chance just to make sure there's nothing we missed.

kisvegabor, 2019-06-26:
> I will test it with -Wstack-usage again when I get a chance just to make sure there's nothing we missed.

Great!

I've tested the performance on STM32429 Discovery with the benchmark app. With all features enabled:
- dev-6.0: 62 ms
- vla: 61.3 ms

So surprisingly is seems we have improved the performance too. :)

embeddedt, 2019-06-26:
> So surprisingly is seems we have improved the performance too. :)

Strange... I would expect it to be slower because of the extra allocation!

Are we merging `vla` to `dev-6.0` today?

kisvegabor, 2019-06-26:
If you agree, yes!

kisvegabor, 2019-06-27:
@embeddedt I read your comment again and it seems you are happy with merging it so I've just merged it.

embeddedt, 2019-06-27:
@kisvegabor Hmm, I thought I replied. Anyways, yes, it's fine to merge it.

closed 2019-06-27 by embeddedt