lvgl/lvgl #1116: STM32 example projects with DMA2D

embeddedt, 2019-06-24:
@kisvegabor I am in the process of updating `stm32f746_discovery_no_os_sw4stm32` to the latest LittlevGL. I discovered that the GPU API has changed.

I checked the F4 project to see if you had already rewritten the GPU parts of the driver there but it seems incomplete.

The function in question is `gpu_mem_fill`: https://github.com/littlevgl/stm32f429_disco_no_os_sw4stm32/blob/c66d6546ffdc9cea3e8b7d6a3bbb0c32a49a39a3/hal_stm_lvgl/tft/tft.c#L225-L243

I think this means that all STM32 projects are currently broken on dev-6.0 when `LV_USE_GPU` is enabled. Any suggestions?

kisvegabor, 2019-06-25:
The only difference is now an area is passed to fill instead of a line. So the trivial solution would be to fill the area line-by-line.

However, I will update the STM32F429 Discovery project as I can to support "real" area fill.
Hopefully, it will help to update other STM projects too.

> Side note: it seems like the STM32 display drivers have much in common. At some point I think we should unify the identical parts.

E.g. in the driver repo?

kisvegabor, 2019-06-25:
I've quickly updated the driver in the mentioned "trivial" way. 

I haven't spent too much time with that but didn't find if the DMD2D can fill an arbitrary area of a memory so I've just called the "old" `fill` for every line of the area.

See https://github.com/littlevgl/stm32f429_disco_no_os_sw4stm32/commit/cab0cf23343db99e16a908c8aa06883f0abb1dd1

I also renanmed `mem_blend_cb` and `mem_fill_cb` to `gpu_blend_cb` and `gpu_fill_cb`. The `gpu_` prefix tells betters what they are.

embeddedt, 2019-06-27:
> E.g. in the driver repo?

Yes, seeing as even the F4 and F7 have very similar HALs.

We can do a slow migration after `v6.0`; there is no rush to do this. I just figured it was better than having 4 copies of the same thing.

> I've quickly updated the driver in the mentioned "trivial" way.

Great! Hopefully I can test it today.

kisvegabor, 2019-06-27:
> We can do a slow migration after v6.0; there is no rush to do this. I just figured it was better than having 4 copies of the same thing.

Okay, we will see :) 

> Great! Hopefully I can test it today.

Awesome!

embeddedt, 2019-06-27:
@kisvegabor Works perfectly, thank you!

closed 2019-06-27 by embeddedt

kisvegabor, 2019-06-27:
Great!