lvgl/lvgl #1117: Small error in lv_canvas_copy_buf parameter

xiongyumail, 2019-06-25:
HI,

Please notice the small differences in w, h, x, y in the parameters.

<https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_objx/lv_canvas.c>
```
void lv_canvas_copy_buf(lv_obj_t * canvas, const void * to_copy, lv_coord_t w, lv_coord_t h, lv_coord_t x, lv_coord_t y)
```

<https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_objx/lv_canvas.h>
```
void lv_canvas_copy_buf(lv_obj_t * canvas, const void * to_copy, lv_coord_t x,
                        lv_coord_t y, lv_coord_t w, lv_coord_t h);
```

embeddedt, 2019-06-25:
@xiongyumail Thanks for noticing! Strange that this happened. I've fixed it here: https://github.com/littlevgl/lvgl/commit/ff657882bc544374c372d086e9a9c3d4e2037648

closed 2019-06-25 by embeddedt

kisvegabor, 2019-06-25:
Thank you for pointing that out and for fix!