lvgl/lvgl #1124: Latest [Dev-6.0] branch broken?

pete-pjb, 2019-06-28:
Hi,

I just downloaded the latest Dev-6.0 branch to update my code and found I am getting a compilation error at line 1111 of indev.c saying `LV_SIGNAL_DRAG_BEGIN`  is undefined...

Upon investigation it looks like the header file lv_obj.h line 125 might have an issue with it's comments which has inadvertently commented out this define... I fixed the comment locally and it seems to be fine now.

Perhaps @kisvegabor or @embeddedt can take a look and check.

Kind Regards,

Pete

embeddedt, 2019-06-28:
Sorry about that... while block pasting I forgot that it would comment out the line as well. Fixed in latest `dev-6.0`.

closed 2019-06-28 by embeddedt

pete-pjb, 2019-06-28:
No Worries @embeddedt, thanks...

Kind Regards,

Pete