lvgl/lvgl #1125: [v6.0] Chart tick marks are not exactly aligned

amirgon, 2019-06-28:
When adding tick marks to chart, **they are not aligned exactly to the chart division lines**.

You can see the problem on [this online simulator example](https://cdn.statically.io/gh/littlevgl/lv_micropython/ef943a1ab40b32a8a77059d6e833a440446c51e8/ports/javascript/lvgl_editor.html?env=dev&script=https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/e5c6219ecc89c1585fb92072a0256f9ed9d8e47c/examples/chart_anim_example.py).
In this example, "3" is aligned correctly, "2" is too low, "4" is too high. "b", "c", "d" also have a small offset. 

Screenshot:
[image]

Relevant code:

```python
...
chart.set_width(scr.get_width() - 100)
chart.align(scr, lv.ALIGN.CENTER, 0, 0)
series1 = chart.add_series(lv.color_hex(0xFF0000))
chart.set_type(chart.TYPE.POINT | chart.TYPE.LINE)
chart.set_series_width(3)
chart.set_range(0,100)
chart.init_points(series1, 10)
chart.set_points(series1, [10,20,30,20,10,40,50,90,95,90])
chart.set_x_tick_texts('a\nb\nc\nd\ne', 2, lv.chart.AXIS.DRAW_LAST_TICK)
chart.set_x_tick_length(10, 5)
chart.set_y_tick_texts('1\n2\n3\n4\n5', 2, lv.chart.AXIS.DRAW_LAST_TICK)
chart.set_y_tick_length(10, 5)
chart.set_div_line_count(3, 3)
chart.set_margin(30)
```

embeddedt, 2019-06-28:
Reproduced in C as well.

```c
        /*Create a chart*/
        lv_obj_t * chart;
        chart = lv_chart_create(lv_scr_act(), NULL);
        lv_obj_set_size(chart, 200, 150);
        lv_obj_align(chart, NULL, LV_ALIGN_CENTER, 0, 0);
        lv_chart_set_type(chart, LV_CHART_TYPE_POINT | LV_CHART_TYPE_LINE);   /*Show lines and points too*/
        lv_chart_set_series_opa(chart, LV_OPA_70);                            /*Opacity of the data series*/
        lv_chart_set_series_width(chart, 4);                                  /*Line width and point radious*/

        lv_chart_set_range(chart, 0, 100);

        /*Add two data series*/
        lv_chart_series_t * ser1 = lv_chart_add_series(chart, LV_COLOR_RED);

        /*Set the next points on 'dl1'*/
        lv_chart_set_next(chart, ser1, 10);
        lv_chart_set_next(chart, ser1, 10);
        lv_chart_set_next(chart, ser1, 10);
        lv_chart_set_next(chart, ser1, 10);
        lv_chart_set_next(chart, ser1, 10);
        lv_chart_set_next(chart, ser1, 10);
        lv_chart_set_next(chart, ser1, 10);
        lv_chart_set_next(chart, ser1, 30);
        lv_chart_set_next(chart, ser1, 70);
        lv_chart_set_next(chart, ser1, 90);

        lv_chart_set_y_tick_texts(chart, "1\n2\n3\n4\n5", 2, LV_CHART_AXIS_DRAW_LAST_TICK);
        lv_chart_set_y_tick_length(chart, 10, 5);
        lv_chart_set_div_line_count(chart, 3, 3);
        lv_chart_set_margin(chart, 30);
        lv_chart_refresh(chart); /*Required after direct set*/
```

embeddedt, 2019-06-28:
@kisvegabor I did a little bit of debugging and it looks like the problem is that the algorithms used for drawing the chart lines and drawing the ticks are producing different results. I also suspect these lines are causing issues:

https://github.com/littlevgl/lvgl/blob/72cf43a176fd94f209766420ac8b4b4d1cba8ecf/src/lv_objx/lv_chart.c#L708-L709

https://github.com/littlevgl/lvgl/blob/72cf43a176fd94f209766420ac8b4b4d1cba8ecf/src/lv_objx/lv_chart.c#L731-L732

Could you have a look at it, please?

embeddedt, 2019-06-28:
I think this is a precision issue. More debugging shows that the gaps for both grid lines and tick lines can vary by 1 pixel because of imprecisions in the calculation.

If my theory is right, this could cause a slow increase in the inaccuracy of the tick marks.

kisvegabor, 2019-06-28:
Hi,

I've found several issues with the positions. I fixed it here: c1072a6 It seems working now.

There way a y invert in y tick marks. I removed it because it also caused some error. Due to this, you need to set text in reversed order. So:
lv_chart_set_y_tick_texts(chart, "5\n4\n3\n2\n1", 2, LV_CHART_AXIS_DRAW_LAST_TICK);
      
I've noticed that there is 1 pixel error on the y edges compared to the border of the chart. It's because of the line drawing algorithm because it can't decide how to align symmetrically a 2 px wide line. Fortunately, it is visible on the edges and only 1 px with even line widths. 

Is it working well for you too?

embeddedt, 2019-06-28:
@kisvegabor Thank you! It seems to be working normally here now.

Since @amirgon and I are basically using the same code base, I assume it will work for him as well.

kisvegabor, 2019-06-28:
> Thank you! It seems to be working normally here now.
Since @amirgon and I are basically using the same code base, I assume it will work for him as well.

Great!

amirgon, 2019-06-29:
Looks good now!

[image]

@kisvegabor  @embeddedt  - Any idea why the lower horizontal chart division line has different width? (thinner)

embeddedt, 2019-06-29:
I also note that the Y-axis minor tick marks (the smaller ones) do not have consistent heights.

kisvegabor, 2019-06-30:
Hm, strange. It works well for me:
[image]

Can you reproduce it in the simulator too?

amirgon, 2019-06-30:
@kisvegabor maybe it's related to the slider on the right? Could you try adding the slider and see if it still works fine?

kisvegabor, 2019-07-02:
I haven't implemented the animation, an scaling just periodically adding new values to the chart but it works here:

[image]

amirgon, 2019-07-02:
@kisvegabor very strange.
Maybe it's a local problem related to screen resolution on my machine.
Thank you for checking this!
Closing for now.

closed 2019-07-02 by amirgon

kisvegabor, 2019-07-02:
> Maybe it's a local problem related to screen resolution on my machine.

I also see it on my screen using the MicroPython simulator.

I adjusted the height of the chart by `chart.set_height(100)` and 
`chart.set_height(90)` and other lines were thinner. In C in Simulator it works well with different heights too.

I suspect its related to how the browser displays the image. Maybe it's scaled and some pixels are missing.

amirgon, 2019-07-02:
> I suspect its related to how the browser displays the image. Maybe it's scaled and some pixels are missing.

@kisvegabor Actually I experienced this on the Micropython unix port (which uses SDL), not on the browser.

kisvegabor, 2019-07-03:
> Actually I experienced this on the Micropython unix port (which uses SDL), not on the browser.

I've just found that with `buf1_1 = bytearray(480*11)` other lines are thinner. By modifying the buffer size I could reproduce it in C too and found and fixed the issue. See a78d048

Did it solve the issue in Micropython too?

amirgon, 2019-07-03:
> I've just found that with `buf1_1 = bytearray(480*11)` other lines are thinner. By modifying the buffer size I could reproduce it in C too and found and fixed the issue. See [...]

@kisvegabor Yes! this fix solves the problem. Thanks.

kisvegabor, 2019-07-04:
Great!