lvgl/lvgl #1127: Selecting a spinbox activates editing cursor, but spinbox is not in editing mode

gesture1968, 2019-06-29:
Using a rotary switch and a group containing 2 spinboxes, when selecting a spinbox with a touch screen, the spinbox is selected, but also the editing cursor starts blinking. But the spinbox is not in editing mode yet. Pressing the rotary switch will start the editing mode as expected. Selecting the spinbox with the turn of the rotary switch does not make the editing cursor active/blinking.

Selecting the active spinbox by pressing the touchscreen while it is not editing, the editing cursor also does not ativate/blink, only the inactive spinbox being activated by a touchscreen press makes the cursor blink.

It would be best to either not blink the cursor, or start editing mode directly after pressing the spinbox on the touchscreen. Now it is kind of in the middle.

embeddedt, 2019-06-29:
@gesture1968 It's quite possible that no one ever tried using the spinbox with a touchscreen and rotary switch simultaneously.

gesture1968, 2019-06-29:
Well, one has now :))

kisvegabor, 2019-06-30:
> Selecting the active spinbox by pressing the touchscreen while it is not editing, the editing cursor also does not ativate/blink, only the inactive spinbox being activated by a touchscreen press makes the cursor blink.

It was really a bug and I've fixed it in `dev-6.0`. The other things should be fine this way because the cursor is activated according to the type of the used input device. With ENCODER the cursor is not activated in edit mode because the spinbox still can't be edited. When clicking on it the library assumes you have a KEYPAD where there is no edit and navigation mode (only one). 

> I also noticed that the activation/blinking cursor happens when the program is started for the first time. The focused spinbox also starts with the blinking cursor. I have to flip the focus between the two spinboxes to get normal non-blinking spinbox digits.

By default the cursor is visible, however, you can use 
`lv_ta_set_cursor_type(spinbox, lv_ta_get_cursor_type(spinbox) | LV_CURSOR_HIDDEN);` to hide it.
In can be used in `LV_EVENT_FOCUSED/DEFOCUSED` events too to manage the cursor manually.

kisvegabor, 2019-07-03:
#As the issue should be fixed I close it.

If it's still not working as expected please comment here and reopen the issue.

closed 2019-07-03 by kisvegabor