lvgl/lvgl #1129: unnecessary include in .h files

asvitenkov, 2019-07-01:
I have found (probably) unnecessary includes "string.h" in two files
 lvgl/src/lv_misc/lv_anim.h line 16 #include <string.h> ([link](https://github.com/littlevgl/lvgl/blob/38cbc2fdeac7b4a2b41e95d5503b5b75801b3e54/src/lv_misc/lv_anim.h#L24))
 lvgl/src/lv_misc/lv_area.h  line 26 #include <string.h> ([link](https://github.com/littlevgl/lvgl/blob/38cbc2fdeac7b4a2b41e95d5503b5b75801b3e54/src/lv_misc/lv_area.h#L16))

I think It is better to move includes to appropriate .c files. 

I am trying to build lvgl library for EFI environment and it is create some problem for building.

kisvegabor, 2019-07-01:
It really would be better to have them in the C files. However, some inline function uses `memcpy` so they also need to be changed.

Just out of curiosity, what kind of issue has it caused?

asvitenkov, 2019-07-01:
Thak you for your reply.
Oh, sorry. It is my fault. I looked through the documentation and understand that memcpy defined in string.h (I used to think that definition of memcpy defined not in string.h)
So you can't remove this includes.

Problem is that you can't use standard C library and function for UEFI application. (Probably I can on my own implement thise functions to solve problems)
And there is own realization of memory allocation (thank you a lot for perfect define LV_MEM_CUSTOM)

closed 2019-07-01 by asvitenkov

kisvegabor, 2019-07-02:
> Problem is that you can't use standard C library and function for UEFI application. (Probably I can on my own implement thise functions to solve problems)

It can work. Only a few functions are used from string.h such as memcpy, memset, strlen, strcpy.

Using LittlevGL for UEFI boot is a very interesting use case. Could you write a few lines about in [My projects](https://forum.littlevgl.com/c/my-projects) category on the Forum? Of course, only what you can tell if you do it for a company.

asvitenkov, 2019-07-02:
Probably I can leave some notes when I finish porting

kisvegabor, 2019-07-02:
Awesome!

luka177, 2020-09-09:
How did it go? Way you able to get lvgl in uefi env?

asvitenkov, 2020-09-09:
Yes, I got lvgl on UEFI env. It works perfectly!

luka177, 2020-09-09:
Can you share your source?

asvitenkov, 2020-09-09:
No, sorry. I can't do it.
But I can help you if you have some questions/troubles.

luka177, 2020-09-09:
OK, some notes will be helpful. I didn't try to port yet, but soon i will.

asvitenkov, 2020-09-09:
you can use my email for direct communication (see my profile)

jief666, 2021-10-02:
Hi @asvitenkov. I think I want to compile lvgl for UEFI too. I read you can't share your sources. But could you help me ? Pointing me in the right direction ?
You had to write a driver, right ?

asvitenkov, 2021-10-05:
Hi @jief666 
First of all you need to compile lvgl library with UEFI. I used gnu-efi library (linux) for this purposes (also you can use EDK2 project). After that you have to write a screen driver (and keyboard driver if necessary).

luka177, 2021-10-30:
@jief666 i can share sources if you still need it?

jief666, 2021-10-30:
Yes. Would be great !

luka177, 2021-10-30:
https://github.com/Android-Boot-Manager/droidboot_generic_uefi/tree/master/Library/LVGL

jief666, 2021-10-30:
Is there an UEFI screen driver is it ?

luka177, 2021-10-30:
yes, use like:
```
static void EfiGopBltFlush(
    lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
  mGop->Blt(
      mGop, (EFI_GRAPHICS_OUTPUT_BLT_PIXEL *)color_p, EfiBltBufferToVideo, 0, 0,
      area->x1, area->y1, area->x2 - area->x1 + 1, area->y2 - area->y1 + 1, 0);

  lv_disp_flush_ready(disp_drv);
}

```