lvgl/lvgl #1132: [v6.1] General discussion

kisvegabor, 2019-07-03:
## Schedule
TBD

## Main new features

#### General 
The most important goals are to
- Rework the examples repo: https://github.com/littlevgl/lv_examples/issues/31  
Can be done independently from v6.1
- Create tutorials, videos to gain more presence  
- Fix bugs and make v6.x more mature and stable 

#### Architecture 
- Advanced masking: #1088 
- Right-to-Left writing: #899 

#### Object types
- Gauge needle from image
- Canvas zoom
- Canvas blur
- Use FontAwesome5 as symbols and add Backspace symbol

embeddedt, 2019-07-03:
> Canvas blur/zoom

Are you planning to implement this by having users layer an `lv_blur`/`lv_zoom` object on top of the canvas?

kisvegabor, 2019-07-04:
> Are you planning to implement this by having users layer an lv_blur/lv_zoom object on top of the canvas?

I was planning to do this similarly to rotate. So only for canvas. You draw something on the canvas and you can blur it or zoom into it.

embeddedt, 2019-07-04:
@kisvegabor Okay. It would be neat to have the ability to blur other objects as well, but I can see the complexities of such an implementation due to the need for buffering all of the object's pixels.

kisvegabor, 2019-07-05:
>  It would be neat to have the ability to blur other objects as well, but I can see the complexities of such an implementation due to the need for buffering all of the object's pixels.

It really would be nice but I suppose if you need these features you have some memory to buffer it. Anyway, I'll be thinking about it how to make these feature more generic and easier to use with other common objects. Maybe a function like `lv_canvas_draw_obj(canvas, obj, x, y)` could be handy to draw an object an all it's children to the canvas. 

I've just added an other nice canvas feature: texture mapping. It means the user gives an image and a 3D mesh and LittlevGL stretches the image to fit the mesh. It's enough to work with a 3D surface instead of a real object to avoid Z buffering. It can be used to add some perspective to objects or map a text a to arc, etc.

embeddedt, 2019-07-05:
> I've just added an other nice canvas feature: texture mapping. It means the user gives an image and a 3D mesh and LittlevGL stretches the image to fit the mesh.

I'm afraid I didn't quite understand your description. Is there a simple code example so I can try it?

kisvegabor, 2019-07-05:
Let's say you have an image with width and height = 100. and a polygon like `(0;0), (110, 0), (0,90), (120,130)`. Texture mapping should stretch the image to fit the polygon. With this, we could create distort the image to get a 3D-like effect.

embeddedt, 2019-07-05:
@kisvegabor The idea sounds neat!

puzrin, 2019-07-06:
I'm interested in more frequent releases. If sort ideas by value/cost ratio, i would suggest:

- 6.0.1 with font compression support

embeddedt, 2019-07-06:
@puzrin Usually minor releases happen more frequently (i.e. 2-3 months) depending on the list of features to implement. We drained that list in 6.0 so we don't really have that much to do for 6.1 in comparison.

Most of the focus will be on writing more documentation and examples, which can be separate from the release itself.

kisvegabor, 2019-07-08:
The last few months were a big hurry and very tough. So I'd like to slow down a little bit. 

For `v6.1` the most difficult features are:
- right to left support
- advanced masking
- texture mapping. 

The rest is quite straight forward. And takes only 1-2 days to implement. 

I'll create issues for all the planned features to see if somebody is interested in implementing them.

As now the main focus is on the examples, I'd say 3-4 month instead of 2-3 for the release.

embeddedt, 2019-07-09:
> Maybe a function like `lv_canvas_draw_obj(canvas, obj, x, y)` could be handy to draw an object an all it's children to the canvas.

That would be neat! How would it work, exactly, since objects are currently associated with a display?

kisvegabor, 2019-07-10:
> That would be neat! How would it work, exactly, since objects are currently associated with a display?

It's a nice opportunity which came with the multi-display support.
Have look at [lv_canvas_draw_rect](https://github.com/littlevgl/lvgl/blob/4a036d8c28725a4574ce3474bb960677ee3adde2/src/lv_objx/lv_canvas.c#L454).
It creates a dummy a display where the buffer is the canvas and draws the rectangle on it. In other words, it simulates the case when the library is refreshing the display but it's just dummy display now.
I haven't tested it with drawing real objects but it should work.

embeddedt, 2019-07-10:
@kisvegabor Would it be possible to get input on the canvas with a similar hack (i.e. if I drew a button on the canvas, would I be able to click it?)

kisvegabor, 2019-07-11:
Hm, interesting question. As usual, the canvas itself can receive events, however, as it doesn't keep track with the "things" drawn on the canvas can't search among them when a click happens.

embeddedt, 2019-07-12:
@kisvegabor I've been thinking about this some more. What if we found some way of keeping the dummy display around for longer? Then a fake input device could also be attached to the canvas.

kisvegabor, 2019-07-12:
How do you mean "Then a fake input device could also be attached to the canvas."?

If I understand the concept correctly, the problem is that on the canvas there are no objects, only pixels. So there is no way to tell if a clicked pixel belongs to button, label or anything else.

aenchevich, 2019-07-12:
Hello!  :)
Hope this is the correct place to ask this question:
 I was using lvgl 5.x until recently but then the usual thing happened "help, problem! -- we won't fix in 5.x, use 6.x!"  so I started porting my code and .. omg are things different in 6! There's many new featrues, which is great, but also there are changes which break existing code even though it could be avoided.
 So... I would like to ask/suggest that whenever possible all 6.x changes are made as non-obtrusive as possible and with maximum 5.x backwards compatibility to make people's life easier (yes, that includes me :)) 

For example:
 There's a new param LV_ANIM_OFF/LV_ANIM_ON in many existing functions, e.g. a call like::
`lv_sw_on (switch2);`
now has to be:
`lv_sw_on (switch2, LV_ANIM_OF);`

So... the problems I see with this are:
a) existing code will not compile
b) people who do not want to use animation are forced to nevertheless include lv_anim in their code
c) as a direct consequence of b) the resulting binaries are bigger then necessary
d) the LV_USE_ANIM define becomes fuzzy: what does it control exactly? These new params are totally independent of it which is simply logically incorrect: 
  -- the LV_ANIM_ON makes no sense when LV_USE_ANIM is not defined and is confusing to see this in the code without any #ifdef guards, using #ifdef guards inside the function signature, n the other hand is ugly and makes the code hard to read. 

To fix the above I propose the following
- add funcs with signature like this (and guarded by USE_ANIM!):
`#if defined ( LV_USE_ANIM )
    void lv_<OBJNAME>_use_anim( lv_obj * o, lv_anim_enable_t use_anim);
#endif
`
- then in your code, (if, and only if you want to use animation in your app!), call these funcs to enable animation for the objects when it is desired, e.g.:

`
#if defined ( LV_USE_ANIM )
#include "lv_anim.h"
#endif
  ...
#if defined ( LV_USE_ANIM )
lv_sw_use anim (switch2, LV_ANIM_ON);
#endif
lv_sw_on (switch2);`

This will solve all the problems I listed above:
 - existing code will work as before;
 - no need to worry about anim-related stuff in your code if you don't want it;
 - the new feature will be controlled at compile time by the USE_ANIM define and at run time - by the LV_ANIM_ON/OFF values like it should be.

kisvegabor, 2019-07-12:
@aenchevich We change major version (v5.x -> v6.x) when API breaking changes are introduced. On minor version (v6.0 -> v6.1) we keep backward compatibility. 

A major version change is really inconvenient but necessary when too many things are accumulated and it's necessary to change the architecture/API.

Your arguments are reasonable, but we can add them on in v7.0.

embeddedt, 2019-07-12:
@kisvegabor Right now in order to draw on the canvas a fake `lv_disp_t` is created just for the duration of that function. What if we provided a way to turn the canvas into a fully-functional display? That way objects could still be associated with it.

BrianPugh, 2019-07-12:
@kisvegabor @embeddedt now that v6.1 is underway, can we merge these 2 PRs into it? Thanks!

https://github.com/littlevgl/lvgl/pull/1119
https://github.com/littlevgl/lvgl/pull/1115

kisvegabor, 2019-07-15:
@embeddedt 
> Essentially, the canvas would be a display inside a display.

I almost understand :slightly_smiling_face: 
- If the drawn objects can be rotated, zoomed, transformed etc, their coordinates will be changed so the input device module won't know where the object really is.
- If the drawn objects can't be rotated, etc, then how is it different from using a canvas as simple parent obejcts.

kisvegabor, 2019-07-15:
@BrianPugh 
> now that v6.1 is underway, can we merge these 2 PRs into it? Thanks!

Sure, please send a new PR!

aenchevich, 2019-07-15:
Gabor, 

I volunteer to add the `lv_<OBJNAME>_use_anim` funcs to all objects if you agree to this?

It's true that breaking some code between major versions is inevitable, and just as equally true is that it should be avoided if possible? Since it could be avoided in this particular case I think that this change would be a "Good Thing (tm)" 

:)

kisvegabor, 2019-07-16:
@aenchevich 
Just to clarify for motivation:
- Would it'd be difficult (a lot of time) to migrate to v6.0 because of the changes in these prototypes? If so, please share your case. I can hardly imagine a UI where it would take more than half an hour.
- You don't use animations and it seems silly to type `LV_ANIM_OFF` every time. If so, the solution could be to change the prototype if the animations are disabled. (Not sure if it's really a good idea)

embeddedt, 2019-07-16:
> If so, the solution could be to change the prototype if the animations are disabled. (Not sure if it's really a good idea)

I would suggest not adopting this approach. Changing the prototype depending on configuration makes it difficult to define a solid ABI (binary interface), whereas the current setup can just ignore the animation parameter if it's disabled, with no changes to the prototype.

It isn't really that much more work, in my opinion, to type `LV_ANIM_OFF`.

kisvegabor, 2019-07-17:
> Changing the prototype depending on configuration makes it difficult to define a solid ABI (binary interface)

That's also true. It seems it really should remain for v7.0 :slightly_frowning_face:

puzrin, 2019-07-20:
Not sure if it's for 6.1, but it really worth to solve scripting questions about all lvgl repos:

1. Recommended scripting solution for repos. Possibilities:
   - `make` - should be available everywhere, but not sure about windows (latest one should support linux CLI)
   - nodejs-based solutions like [Makefile.js](https://github.com/eslint/eslint/blob/master/Makefile.js)
2. Recommended testing approaches
   - My personal opinion is PIO https://docs.platformio.org/en/latest/ci/travis.html + may be nodejs's mocha for quick scripting https://github.com/mochajs/mocha (not C-related tasks).

I'm not sure about node.js and don't wish to push anyone about javascipt (i understand someone may not like JS at all). My opinion is based on fact, that new tools lv_i18n and lv_font_conv uses nodejs, and any intermediate user/dev will install nodejs anyway.

But i'll be ok with classic `make`/bash too, and ok to use PIO's built in python.

If you inspect lv_i18n - it uses make/nodejs, based on my personal opinion. Because i can't wait and need to select something. But it would be nice to use some unified (recommended?) approach for lvgl org. For example, i'm sure, authors of `lv_arduino` & `lv_binding_micropython` will be happy to automate some tasks to minimise human errors.

embeddedt, 2019-07-20:
@puzrin Our goal is not to provide a centralized build solution. That makes it difficult to use LittlevGL in any project. One of the perks with LittlevGL is that you can use it by just copying it into a directory and compiling all of the source files.

puzrin, 2019-07-20:
Let's separate maintainers an end users needs. Things i speak about are for maintainers. They need to automate updates and do integrity checks on lvgl changes.

embeddedt, 2019-07-20:
> They need to automate updates

Updates are now almost entirely automated - the only manual steps are to write the release notes, update the PC simulator, and update the example projects.

> integrity checks on lvgl changes

There are plans to integrate CI at some point in the future, but at the moment, @kisvegabor has mentioned that he uses/tests LittlevGL nearly every day.

kisvegabor, 2019-07-22:
We do have makefile based build system. This repo has [lvgl.mk](https://github.com/littlevgl/lvgl/blob/master/lvgl.mk) which is used by the [simulator](https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/master/Makefile) and the [linux frame buffer](https://github.com/littlevgl/lv_linux_frame_buffer/blob/master/Makefile) projects. `lv_examples` and `lv_drivers` also have their mk files.

ShenRen, 2019-07-23:
When I work a Marlin UI project, I have port lvgl from C to C++ for convenience . 
Also ,I had  added some speciality to my code. like : smart pointer (detection lv_obj_t exist) , signal and slot (inspired by Qt) , and a improved lv_task . Those feature are easy to implement on C version lvgl.  I will post my code soon.
Last time , I had suggest a 3D feature which is in progress on v5.3 .

lvgl is not friendly for c++ class inherit the internal struct data . Because lv_obj_t `s memory allocate action is uncontrollable .   
At the first time , I try to use lv_obj_t as a class member . But convert a pointer from class to struct is verbose.
At the second time, I try to modified the lv_mem_allocate function and add new and delete function for class. Then all struct data can inherited by class . No more pointer conversion.

I hope , we can change the way lvgl create lv_obj_t .

embeddedt, 2019-07-23:
Are you trying to put an `lv_obj_t` in a class, or an `lv_obj_t *` (pointer) in the class?

The first method won't work because LittlevGL needs to handle its own memory allocation.

I recommend you store the pointer inside your own custom class and use `lv_obj_create` and `lv_obj_del` to manage its creation/deletion.

ShenRen, 2019-07-23:
@embeddedt  I put lv_obj_t * (pointer) in the class

First time :
`
class LVObject
{
    LV_OBJECT // memory func ( new and delete )

    friend lv_res_t lvobjectSignalFunc (struct _lv_obj_t * obj, lv_signal_t sign, void * param);
protected:
    lv_obj_t * m_this = nullptr; 
public:
.........
};
`

ShenRen, 2019-07-23:
Second time:

```
/**
 * @brief lv_obj_t 
 */
class LVObjectNode
        : public lv_obj_t       
        , public LVLLNodeMeta  
{
protected:
    LVObjectNode() {}
};
```

```
/**
 * @brief The LVObject class
 * LVObject memory layout
 *                 LVObject                       LVObjectNode         lv_obj_t
 *                 |                              |                    |
 *                 V                              V                    V
 * {LVMemoryHeader:[ (VTable){LVMemoryHeader:[ (lv_obj_t) (LVLLNodeMeta) (LVObjectEnhance) ] } (LVMemoryHeader) ] }
 */
class LVObject
        : public LVObjectEnhance 
        , public LVFakeMemHeader<LVObjectNode>  
        , public LVObjectNode   
{
    LV_MEMORY
    LV_NOCOPY(LVObject)

protected:
......
};
```

amirgon, 2019-07-23:
@ShenRen What's wrong with your "First Time"?
Just keep a pointer to lv_obj_t as a class member, call the right `lv_<obj>_create` function in your constructor and `lv_obj_del` in the destructor.  
What's not working for you?

ShenRen, 2019-07-23:
> @ShenRen What's wrong with your "First Time"?
> Just keep a pointer to lv_obj_t as a class member, call the right `lv_<obj>_create` function in your constructor and `lv_obj_del` in the destructor.
> What's not working for you?

It is work for me. 
I'm just saying that I try two ways.

kisvegabor, 2019-07-24:
> Also ,I had added some speciality to my code. like : smart pointer (detection lv_obj_t exist) , signal and slot (inspired by Qt) , and a improved lv_task . Those feature are easy to implement on C version lvgl. I will post my code soon.

I'm curiously waiting to see your improvments!

aenchevich, 2019-07-24:
> I would suggest not adopting this approach. Changing the prototype depending on configuration makes it difficult to define a solid ABI (binary interface), whereas the current setup can just ignore [...]

Regarding changing or not changing prototypes: actually it is my approach advocating NOT changing the prototype and keeping it as it was in 5.x. Adding new functions to implement new functionality is correct way and this is exactly what I am suggesting.

Regarding whether it is a lot of work: I started doing this. It is a lot of work (IMO). And given that it is unnecessary work and is breaking the interface I thought it would be a good idea to add the new function
- - -
(..and, yes typing LV_ANIM_OFF in all the functions even when not using it is kinda silly :D But really my goal is to keep the lib lean, mean  and clean - don't need something? don't waste bytes on it!)
(sorry I am responding slow - traveling a lot these days)

kisvegabor, 2019-07-24:
> Regarding changing or not changing prototypes: actually it is my approach advocating NOT changing the prototype and keeping it as it was in 5.x. Adding new functions to implement new functionality is correct way and this is exactly what I am suggesting.

Okay, but it's already changed in v6.0 :smile: If we would change it after the release it wouldn't work for some people. We need to be strict in this regard else the whole versioning process wouldn't make sense.

embeddedt, 2019-07-24:
> actually it is my approach advocating NOT changing the prototype and keeping it as it was in 5.x. Adding new functions to implement new functionality is correct way and this is exactly what I am suggesting.

That's what we do in minor releases. In major releases, usually some major change to the internals of the library has been performed.

Consider systems like Windows (or even Linux) which have been in production use for years. They have to keep a lot of compatibility logic around (even in Windows 10) to support applications designed for Windows XP.

That isn't suitable for an embedded system, in my opinion. In cases where internals are changed, eventually it makes more sense to change the prototype than to work around it's limitations.

We've been discussing animations. Here's an example of the difference between 5.x and 6.x:

5.x:

```c
/* Turn on a switch with animations */
lv_sw_anim_on(sw);
```

6.x:

```c
/* Turn on a switch with animations */
lv_sw_on(sw, LV_ANIM_ON);
```

I think we can agree that there isn't much difference between the two.

> don't need something? don't waste bytes on it!

I think the older 5.x approach actually takes up more code space, because (if you've felt/had the need to look at the disassembly) the compiler had to generate function prologue/epilogue for both functions. Also, I don't remember exactly what the code looked like, but it's possible that there was quite a bit of duplication between the two functions, which is even more code savings.

Another point: I don't know if this is true for every architecture, but on ARM, the compiler should only need to generate one more instruction for the extra parameters (2-4 bytes). That's a very small cost to pay for what (I think :man_shrugging: ) is a cleaner API.

> I started doing this. It is a lot of work (IMO).

Have you tried automating the process? If you use a regex you should be able to find all instances of `lv_sw_anim_on(*)` and replace them with `lv_sw_on(*, LV_ANIM_ON)`, regardless of the parameter name.

Let's look at another example: multi-display support. It just wouldn't be possible to cleanly keep the existing API in place without quite a bit of compatibility logic, which would take time to implement and probably wouldn't be worth the effort.

In summary, I understand that API breakage can be frustrating as a developer. But I think there comes a point when we have to drop old APIs to avoid code bloat.

aenchevich, 2019-07-25:
@embeddedt 

Ok, firstly and most importantly: I understand Gabor's reason to say "ok but it is too late to do this in v6" and I am no longer advocating to add this to v6. 

Secondly (and this is really just a philosophical discussion at this point since we all agree that this will not go into v6) ... I do not agree with anything you say.

Your quoted comparison between 5 and 6 is not valid because it disregards the use of the LV_USE_ANIM define and that makes all of the subsequent reasoning invalid. 

The fact that a particular compiler is smart enough to optimize bad code is not a license to write bad code (to me anyway).

Yes I have tried to automate the process. Yes, if I use regex I can find all instances of LV_ANIM_ON in lw_sw. And then in lv_ddlist and lv_tabview, lv_bar, lv_win ... etc. 

I am not saying anything about multi-display support and I don't think it is related to this discussion. Nor am I advocating keeping existing interfaces at all costs. I am advocating keeping them whenever possible unless it is really necessary to change them. In this case it was possible to not break the interface, improve the scope of the USE_ANIM define and make peoples' lives easier. That was all I wanted when I proposed this change.

So we both hate code bloat. So I was wrong to say I do not agree with anything you say :)

embeddedt, 2019-07-25:
>  am not saying anything about multi-display support and I don't think it is related to this discussion

So, for clarification, this discussion pertains purely to the animation prototypes? Fair enough.

> Your quoted comparison between 5 and 6 is not valid because it disregards the use of the LV_USE_ANIM define and that makes all of the subsequent reasoning invalid.

I didn't mention `LV_USE_ANIM` in that comparison, but I did have it in mind at the time.

In the event that `LV_USE_ANIM` is set to `0`, there is one unused parameter in that function call. That still looks clean in my opinion (a single API for turning on the switch, not two).

An unused parameter is not enough of a performance/code size drop to warrant having two functions (in my opinion).

> The fact that a particular compiler is smart enough to optimize bad code is not a license to write bad code

I wasn't referring to optimization with the 2-4 byte savings. I was just referring to the way most architectures work. Maybe I confused you.

> In this case it was possible to not break the interface, improve the scope of the USE_ANIM define and make peoples' lives easier.

I'll concede to that. There wasn't really an internal design choice that would have forced us to change it.

> So we both hate code bloat.

Correct. I do understand where you are coming from, but I also want to keep the LittlevGL API small, clean, and simple to use.

ShenRen, 2019-07-25:
> > Also ,I had added some speciality to my code. like : smart pointer (detection lv_obj_t exist) , signal and slot (inspired by Qt) , and a improved lv_task . Those feature are easy to implement on C version lvgl. I will post my code soon.
> 
> I'm curiously waiting to see your improvments!

I have upload my project at [panda-3d-printer](https://github.com/panda-3d-printer). 
[LittlevGL_CPPPort](https://github.com/panda-3d-printer/LittlevGL_CPPPort) is a c++ version lvgl.
[PandaMini_MarlinUI](https://github.com/panda-3d-printer/PandaMini_MarlinUI) is a [Marlin](https://github.com/MarlinFirmware/Marlin) GUI , base on _LittlevGL_CPPPort_
[LittlevGL_QtSimulator](https://github.com/panda-3d-printer/LittlevGL_QtSimulator) is a LittlevGL Simulator which no need SDL. But it depends on Qt.

I haven't had time to sort out the project yet . There is only Chinese comment and no clear instructions.

For signal and slot , the c++ version code is here [lvsignalSlot](https://github.com/panda-3d-printer/LittlevGL_CPPPort/blob/master/core/lvsignalSlot.hpp). I only use  ```lv_ll``` to storage the connection between ```LVSignal``` and ```LVSlot```.

example for using:
```
    LVSignal IdleSignal;
    LVSlot idleSlot;
    idleSlot.setSlotFunc([](LVSignal* signal)
    {
        LV_LOG_INFO("idleSlot");
    });
    connect(&IdleSignal,&idleSlot,Connection::QueueConnect);// or DirectConnect
   ...
   //just call emit()
  IdleSignal.emit(); // then the slot function will be executed .
```

kisvegabor, 2019-07-26:
@aenchevich 
Well, there wasn't any discussion when I've changed this API change so maybe I haven't considered all the possible problems. My reasons were the same what @embeddedt already mentioned:
- Avoid duplicating code internally. 
- Have fewer API functions.
- Make the API unchanged when configs are changing. It allows you to change the config without breaking your existing code.

kisvegabor, 2019-07-26:
@ShenRen 
Could open a topic for it in the [My projects](https://forum.littlevgl.com/c/my-projects) category in the forum?

cron1c, 2019-09-11:
can you add "°" (degree) char to default font in the next release?

embeddedt, 2019-09-11:
@cron1c That might be an extended Unicode character (beyond what basic ASCII supports by default). My guess is that it is deliberately missing from the default fonts in order to conserve flash space.

kisvegabor, 2019-09-11:
> My guess is that it is deliberately missing from the default fonts in order to conserve flash space.

Yes, that's the case. The built-in font covers only the ASCII range and some symbols. If you something else you can create a new font a disable the built-in one(s).

kisvegabor, 2019-09-20:
I've update symbol fonts to use FontAwesome V5.0. Originally the symbols are split into three files but I merged them:  [FontAwesome5-Solid+Brands+Regular.woff.zip](https://github.com/littlevgl/lvgl/files/3634377/FontAwesome5-Solid%2BBrands%2BRegular.woff.zip)

I've added `LV_SYMBOL_BACKSPACE` and `LV_SYMBOL_PASTE`.

The new list of symbols to be used in the font converter:
```
61441, 61448, 61451, 61452, 61453, 61457, 61459, 61461, 61465, 61468, 
61473, 61478, 61479, 61480, 61502, 61508, 61512, 61515, 61516, 61517, 
61521, 61522, 61523, 61524, 61543, 61544, 61553, 61556, 61559, 61560, 
61561, 61563, 61587, 61589, 61636, 61637, 61639, 61671, 61674, 61683, 
61724, 61732, 61787, 61931, 62016, 62017, 62018, 62019, 62020, 62099, 
62189, 62810 
``` 

[image]

kisvegabor, 2019-09-25:
The following symbols are added:
- EYE_OPEN
- EYE_CLOSE
- SD_CARD
- USB
- NEW_LINE

[FontAwesome5-Solid+Brands+Regular.woff.zip](https://github.com/littlevgl/lvgl/files/3651754/FontAwesome5-Solid%2BBrands%2BRegular.woff.zip)

```
      61441, 61448, 61451, 61452, 61452, 61453, 61457, 61459, 61461, 61465,
      61468, 61473, 61478, 61479, 61480, 61502, 61512, 61515, 61516, 61517,
      61521, 61522, 61523, 61524, 61543, 61544, 61550, 61552, 61553, 61556,
      61559, 61560, 61561, 61563, 61587, 61589, 61636, 61637, 61639, 61671,
      61674, 61683, 61724, 61732, 61787, 61931, 62016, 62017, 62018, 62019,
      62020, 62087, 62099, 62212, 62189, 62810, 63426, 63650
```

char32, 2019-10-21:
Hi,
just wanted to ask: is there a plan to provide a C++ wrapper similar to the one @ShenRen created?
I know littlevGL is fully C++ compatible but having such a wrapper to use it completely object-orientated would be nice :-)
Thanks!

embeddedt, 2019-10-21:
> I know littlevGL is fully C++ compatible but having such a wrapper to use it completely object-orientated would be nice :-)

It's not on the roadmap right now (as far as I know) but pull requests are welcome.

kisvegabor, 2019-10-21:
> I know littlevGL is fully C++ compatible but having such a wrapper to use it completely object-orientated would be nice :-)

Would it be like a binding to C++? Why not? 
We already have a Micropython binding (created by @amirgon) which has a script to parse lvgl API and generate Python-C binding. Maybe that script can be used to generate C-C++ binding too.

kisvegabor, 2019-10-28:
I've just pushed the new drawing engine and other new features to `dev-7.0`.

The new features should go to `dev-7.0`. WIP features will be merged to `dev-6.1` and it will be related in November 2019.

For the discussion of `dev-7.0` see https://github.com/littlevgl/lvgl/issues/1243

kisvegabor, 2019-11-19:
I'm planning to release `6.1`  next Monday (25 November). Let me know if you think something is still missing.

embeddedt, 2019-11-19:
I think all that's left is reviewing the documentation and making sure there is some basic information for all the new features.

I will try to write some documentation for the color picker this weekend.

kisvegabor, 2019-11-19:
> I will try to write some documentation for the color picker this weekend.

That would be awesome!

kisvegabor, 2019-11-21:
I've just created a `dev-6.1` branch in `docs`.

kisvegabor, 2019-11-25:
I've just released `v6.1`. 

The docs still need to be polished a little bit but the main things are already there.

I've updated the Eclipse simulator project and I will update the STM32f429 and STM32769 project tomorrow.

[22 more replies]

closed 2019-12-09 by kisvegabor