lvgl/lvgl #1134: Spinbox lags and (thick) lines are not drawn correct

gesture1968, 2019-07-04:
Using the v6.0 from 26-june I reported a small issue with the spinbox controller, specifically an editing cursor while selecting a spinbox on a touchscreen. The issue was addressed by kisvegabor, but I did not check it until yesterday. I downloaded the latest v6.0. Now with the same code the spinbox is pretty laggy when selecting and editing values by turning a rotary switch. Not just that, but I also have 6 small line arrays that make up 2x3 arrows with a linewidth of 5, These arrows now either completely vanish, or are garbled up. I use the arrows to indicate which spinbox is selected (they are toglled by the rotary and the selected group):

old situation (v6.0_26jun):
00.00 <<<  00.00  -> left spinbox is selected
00.00  >>> 00.00  -> right spinbox is selected

new situation(v6.0_latest):
00.00&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;00.00 -> initially (no arrows visible!)
00.00 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;>00.00                 -> right spinbox selected (some garbled arrow)
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;/
0000  \\&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<00.00                -> left spinbox selected (notice the line above the spinbox)

There seems to be something wrong in the way the screen is build up, and the lag of the spinbox is also weird.

kisvegabor, 2019-07-04:
Could you send a video about the issue(s)?

gesture1968, 2019-07-04:
Sorry for the inconvenience, but today it is all right. I've seen the STM32 behaving strange on the rotary encoder once before (lag/not responding) but after a hard reset this was solved. This was not the case yesterday, but today all seems well. So it's not a code bug after all.

embeddedt, 2019-07-04:
> it's not a code bug after all.

Maybe not in LittlevGL, but usually if hardware works after a reset, it means something is wrong somewhere.

kisvegabor, 2019-07-05:
Anyway, I think this issue can be closed.

If finally, you will conclude that it's related to LittlevGL, please comment here and reopen the issue.

closed 2019-07-05 by kisvegabor