lvgl/lvgl #1135: Message box

embeddedt, 2019-07-04:
**Describe the bug**

The `LV_EVENT_DELETE` event is triggered twice on a message box.

**To Reproduce**

Register an event callback on a message box and then call `lv_obj_del`.

**Expected behavior**

Only one delete event would be sent for the message box itself.

**Additional context**

The problem is occuring because the button matrix is sharing the event handler for the message box.

**Workaround**

Before processing an `LV_EVENT_DELETE` event, check that the object that it was sent to is the message box and not its button matrix.

kisvegabor, 2019-07-05:
> The problem is occuring because the button matrix is sharing the event handler for the message box.

That correct. The button matrix has `lv_obj_set_parent_event(ext->btnm, true);` so all events are sent to the parent message box. 

To solution could be to not use `lv_obj_set_parent_event` but add a custom event function to the button matrix which passes only the desired event. Similarly to https://github.com/littlevgl/lvgl/blob/6989a9c797e28d939e14aa92c32c8936390dd4e4/src/lv_objx/lv_page.c#L1077-L1088

embeddedt, 2019-07-05:
@kisvegabor Okay. I will change it to do things that way then. Do we want that set of events plus `LV_EVENT_VALUE_CHANGED`?

kisvegabor, 2019-07-05:
> Do we want that set of events plus LV_EVENT_VALUE_CHANGED?

Yes, I think that enough.

Thank you!

kisvegabor, 2019-07-07:
It should be fixed [here]( https://github.com/littlevgl/lvgl/commit/5ae4b62f4db0b89b815697b9d1d1679bc568aa86#diff-ef6815498dbe8df10fd77252701d59adR533)

So can we close this issue?

embeddedt, 2019-07-07:
Yes, sorry.

closed 2019-07-07 by embeddedt

kisvegabor, 2019-07-07:
No worries :slightly_smiling_face: