lvgl/lvgl #1138: Horizontal list problem

XQDD, 2019-07-06:
#### 1. One button is fine
[image]

```c++
void App::create_navigation_bar(lv_obj_t *parent) {
    lv_obj_t *list = lv_list_create(parent, NULL);
    lv_cont_set_fit2(list, LV_FIT_FLOOD, LV_FIT_TIGHT);
    lv_page_set_scrl_layout(list, LV_LAYOUT_ROW_T);
    lv_page_set_scrl_fit(list, LV_FIT_TIGHT);
    
    lv_list_add_btn(list, NULL, "button1");
}
```

---

#### 2.Two button, the button becomes very long
[image]
```c++
void App::create_navigation_bar(lv_obj_t *parent) {
    lv_obj_t *list = lv_list_create(parent, NULL);
    lv_cont_set_fit2(list, LV_FIT_FLOOD, LV_FIT_TIGHT);
    lv_page_set_scrl_layout(list, LV_LAYOUT_ROW_T);
    lv_page_set_scrl_fit(list, LV_FIT_TIGHT);

    lv_list_add_btn(list, NULL, "button1");
    lv_list_add_btn(list, NULL, "button1");
}

```
---

#### 3.And try to add other elements to list will cause exception 
[image]

```c++
void App::create_navigation_bar(lv_obj_t *parent) {
    lv_obj_t *list = lv_list_create(parent, NULL);
    lv_cont_set_fit2(list, LV_FIT_FLOOD, LV_FIT_TIGHT);
    lv_page_set_scrl_layout(list, LV_LAYOUT_ROW_T);
    lv_page_set_scrl_fit(list, LV_FIT_TIGHT);

    lv_list_add_btn(list, NULL, "button1");
    lv_list_add_btn(list, NULL, "button1");

    lv_label_set_text(lv_label_create(list, NULL), "Test add other element");
}
```

---

### 4. But I change this line https://github.com/littlevgl/lvgl/blob/8e9764532d866719b5a87f0e2181f7d57c501592/src/lv_objx/lv_list.c#L196 to
```c
lv_btn_set_fit2(liste, LV_FIT_TIGHT, LV_FIT_TIGHT)
``` 
### All the problemes were gone
[image]

kisvegabor, 2019-07-06:
The `lv_list` is not designed to be horizontal too. However, if it's that easy to do we can have an API function like `lv_list_set_orientation(list, LV_LIST_HOR/LV_LIST_VER)`;

Can you add a Pull request?

XQDD, 2019-07-06:
No idea how to implement it, but as I mentioned above, we can change 
```c
lv_btn_set_fit2(liste, LV_FIT_FLOOD, LV_FIT_TIGHT)
```
to
```c
lv_btn_set_fit(liste, LV_FIT_TIGHT)
```
let it suit for both vertical and horizontal, Idk if there some potential problems after the change

kisvegabor, 2019-07-06:
I was thinking about a better API function name. It could be `lv_list_set_layout()` and the button's fit should automatically change according to the layout.

> No idea how to implement it, ...

`lv_list_set_layout` and `lv_list_get_layout` needs to be added to `lv_list.c`. In `lv_list_set_layout` al lthe existing button's fit should be updated. To iterate through the buttons `lv_list_get_next_btn` can be used.

I don't want to push you to implement it just asked if you have time or interest for it.

Anyway, I leave it open for `v6.1` to implement once.

C47D, 2019-07-06:
I would like to help to get this done, i'm running the lvgl example using the ESP32 microcontroller and it isn't updated to 6.0, is the project update scheduled to an specific date?

embeddedt, 2019-07-06:
@C47D By this do you mean [updating the ESP32 project](https://github.com/littlevgl/esp32_ili9341/issues/8) or implementing a horizontal list?

C47D, 2019-07-06:
@embeddedt Hi, i meant implementing the horizonal list. Just saw @kisvegabor comment on the esp32 example:
> I'll update the project to the v6.0 version of LittlevGL next week. Hopefully, the problem will be solved then.

kisvegabor, 2019-07-09:
Implemented in #1140 
 Thank you @C47D!

closed 2019-07-09 by kisvegabor