lvgl/lvgl #1142: display problem when using new font in roller object

xcqq, 2019-07-09:
Have a display problem when using new font in roller object. If I use build-in font. This won't happen. And I have test the font using list object, it works well.

Here is my code:
```c
    lv_obj_t *roller = lv_roller_create(cont_backgroud, NULL);

    LV_FONT_DECLARE(font_chinese_20);
    lv_style_t *style_chinese = lv_list_get_style(roller, LV_ROLLER_STYLE_BG);
    style_chinese->text.font = &font_chinese_20;
    lv_label_set_style(roller, LV_ROLLER_STYLE_BG, style_chinese);
    style_chinese = lv_list_get_style(roller, LV_ROLLER_STYLE_SEL);
    style_chinese->text.font = &font_chinese_20;
    lv_label_set_style(roller, LV_ROLLER_STYLE_SEL, style_chinese);

    lv_roller_set_options(roller,
                        "快捷电压",
                        LV_ROLLER_MODE_NORMAL);
```

Here is the screenshot:
[image]

Thanks a lot!

kisvegabor, 2019-07-09:
Hi,

You should use `lv_roller_set/get_style` function for the roller.

```c
 lv_obj_t *roller = lv_roller_create(cont_backgroud, NULL);

    LV_FONT_DECLARE(font_chinese_20);
    lv_style_t *style_chinese = lv_roller_get_style(roller, LV_ROLLER_STYLE_BG);
    style_chinese->text.font = &font_chinese_20;
    lv_roller_set_style(roller, LV_ROLLER_STYLE_BG, style_chinese);
    style_chinese = lv_roller_get_style(roller, LV_ROLLER_STYLE_SEL);
    style_chinese->text.font = &font_chinese_20;
    lv_roller_set_style(roller, LV_ROLLER_STYLE_SEL, style_chinese);

    lv_roller_set_options(roller,
                        "快捷电压",
                        LV_ROLLER_MODE_NORMAL);
```

It's better to copy the style instead of changing a built-in style directly:
See https://docs.littlevgl.com/en/html/overview/style.html#create-new-styles

xcqq, 2019-07-09:
Got it. I used to use list in this page and I forgot to change them.
Thanks for your help!

closed 2019-07-09 by xcqq