lvgl/lvgl #1145: Possible problem with lv_spinbox object

barmaley70, 2019-07-11:
static void lv_spinbox_updatevalue(lv_obj_t * spinbox) can cause buffer overrun if ext->digit_padding_left value is bigger than LV_SPINBOX_MAX_DIGIT_COUNT + 8. 

The void lv_spinbox_set_padding_left(lv_obj_t * spinbox, uint8_t padding) does not do any validation.

kisvegabor, 2019-07-12:
True.
Could you send a PR with a fix for it?

barmaley70, 2019-07-15:
I think that modifying of the lv_spinbox_set_padding_left is not enough because lv_spinbox_set_digit_format(lv_obj_t * spinbox, uint8_t digit_count, uint8_t separator_position) parameters also participate in writing to the buffer.  The validation can be added to both functions but then their behavior/results will depend  on the order of calling. Is it possible to combine those two methods in one? Something like lv_spinbox_set_digit_format(lv_obj_t * spinbox, uint8_t digit_count, uint8_t separator_position, uint8_t padding)

embeddedt, 2019-07-15:
@barmaley70 

> Is it possible to combine those two methods in one? Something like lv_spinbox_set_digit_format(lv_obj_t * spinbox, uint8_t digit_count, uint8_t separator_position, uint8_t padding)

We cannot change API signatures in a backwards-incompatible manner until the next major release.

closed 2019-08-12 by stale[bot]