lvgl/lvgl #1148: Possible bug in table when changing font using style...

pete-pjb, 2019-07-12:
Hi,

I have been working with the table object and wish to set a smaller font for the table.  If I get the style from the table and change the font and set the table style to the new style it causes the system to crash.  I have created a cut down standalone project to test and it happens in that also.

I believe you will be able to reproduce the issue if you use the following code:

```c
static lv_obj_t		*track_table;

#define NUM_TRK_COLUMNS		11
static void status_create( lv_obj_t *parent ) {

    uint32_t				col_idx;
    lv_style_t				tab_style;

    track_table = lv_table_create(parent, NULL);
    lv_table_set_col_cnt(track_table, NUM_TRK_COLUMNS);
    lv_table_set_row_cnt(track_table, 1);
    lv_style_copy( &tab_style, lv_table_get_style( track_table, LV_TABLE_STYLE_CELL2 ) );
    tab_style.text.font = &lv_font_roboto_12;
    lv_table_set_style( track_table, LV_TABLE_STYLE_CELL2, &tab_style);
    lv_obj_set_pos(track_table, 10, 10);
    for( col_idx = 0; col_idx < NUM_TRK_COLUMNS; col_idx++ ) {
	lv_table_set_cell_type(track_table, 0, col_idx, 2);
    }
    lv_table_set_col_width(track_table, 0, 240);
    lv_table_set_cell_value( track_table, 0, 0, "Satellite Name" );
    lv_table_set_col_width(track_table, 1, 240);
    lv_table_set_cell_value( track_table, 0, 1, "Obs Station" );
    lv_table_set_col_width(track_table, 2, 50);
    lv_table_set_cell_value( track_table, 0, 2, "Az" );
    lv_table_set_col_width(track_table, 3, 50);
    lv_table_set_cell_value( track_table, 0, 3, "El" );
    lv_table_set_col_width(track_table, 4, 50);
    lv_table_set_cell_value( track_table, 0, 4, "Pol" );
    lv_table_set_col_width(track_table, 5, 90);
    lv_table_set_cell_value( track_table, 0, 5, "Bcn(dBm)" );
    lv_table_set_col_width(track_table, 6, 50);
    lv_table_set_cell_value( track_table, 0, 6, "Long" );
    lv_table_set_col_width(track_table, 7, 50);
    lv_table_set_cell_value( track_table, 0, 7, "Lat" );
    lv_table_set_col_width(track_table, 8, 50);
    lv_table_set_cell_value( track_table, 0, 8, "Alt" );
    lv_table_set_col_width(track_table, 9, 50);
    lv_table_set_cell_value( track_table, 0, 9, "Ra" );
    lv_table_set_col_width(track_table, 10, 50);
    lv_table_set_cell_value( track_table, 0, 10, "Dec" );
}
```
Kind Regards,

Pete

pete-pjb, 2019-07-13:
Sorry I'm going to close this I see my own error ... the style should be static... 

Kind Regards, 

Pete.

closed 2019-07-13 by pete-pjb

embeddedt, 2019-07-13:
To be a bit more specific, the style can't go out of scope while LittlevGL is running.

In a simple application where everything is created inside `main()`, and then a`while(1)` loop that calls `lv_task_handler` is at the end, it would be acceptable to not use a `static` style because `main()` won't return while LittlevGL is running.

```c
int main(void)
{
    lv_style_t my_style;
    // . . . code here . . .
    while(1) {
        lv_task_handler();
    }
    /* If this were ever to be reached, LittlevGL shouldn't be running anymore */
    /* `my_style` can safely go out of scope */
}
```

kisvegabor, 2019-07-15:
I was thinking about a macro like:
```c
LV_STYLE_CREATE(name, to_copy) static lv_style_t name; lv_style_copy(&name, &to_copy);
```

What do you think about it? IMO it can help avoid issues like this.

pete-pjb, 2019-07-15:
Hi @embeddedt,

Thank you for your comment, I am aware of this, sorry I didn't explain myself fully in the post, in my case my base system is running FreeRTOS and lv_task_handler() runs in it's own thread. The cut down example I built also was running in a minimal FreeRTOS configuration, so I needed to declare the variable static in the function for creating the GUI in order for it still to be available as the execution of the program continued.

@kisvegabor I think that would be a good idea, In my case I just had moments lapse of concentration and then spent a few hours scratching my head lol, I guess we all do it sometimes!  This might stop us older absent minded guys making silly mistakes :-) 

Kind Regards,

Pete

kisvegabor, 2019-07-15:
@pete-pjb I've just added it to `dev-6.1`
ea00b24

pete-pjb, 2019-07-15:
@kisvegabor many thanks!

kisvegabor, 2019-07-16:
You are welcome :slightly_smiling_face:

turoksama, 2019-07-19:
@kisvegabor
Hello there, 
what do you think of:
```
#define LV_STYLE static lv_style_t

LV_STYLE name;
```

kisvegabor, 2019-07-19:
It sounds reasonable but breaks the rule that type should have `lv_..._t` format.