lvgl/lvgl #1149: RA8875 support

knopserl, 2019-07-14:
I want to use littlevgl with an ESP32 and a RA8875 SPI based 5" display (from buydisplay.com).
Currently I'm using with a RA8875 library directly on an ESP32 and it works great, but it's low level graphic programming.
I have seen, that the existing driver set does not yet provide RA8875 support.
There is a porting guidance, but I'm not sure how difficult it is to provide a littlevgl driver for the RA8875 including the resitive touch.

Is there anybody who used it already with the RA8875 or could give me some more hints to start or check if this chip can be easily ported?

embeddedt, 2019-07-14:
----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2019-07-14 by embeddedt

knopserl, 2019-07-16:
@embeddedt sorry, will do so :-(