lvgl/lvgl #1160: lv_calendar_get_pressed_date not returning NULL

larenge2, 2019-07-24:
**Describe the bug**

lv_calendar_get_pressed_date(cal) not returning NULL value if no day is selected. Changing months do not work as expected.

embeddedt, 2019-07-24:
Please fill out the template entirely.

closed 2019-07-24 by embeddedt

embeddedt, 2019-07-24:
After you've filled out the template, you can reopen this issue.

larenge2, 2019-07-25:
**Describe the bug**

lv_calendar_get_pressed_date(cal) not returning NULL value if no day is selected. Changing months do not work as expected.

**To Reproduce**
According to your documentation : 

> In Input device related events lv_calendar_get_pressed_date(calendar) tells which day is currently being pressed or return NULL if no date is  pressed.

And example code : 
`static void event_handler(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        lv_calendar_date_t * date = lv_calendar_get_pressed_date(obj);
        if(date) {
            lv_calendar_set_today_date(obj, date);
        }
    }
}
`
Try click on calendar to change month. You will get non NULL pointer in event handler.

Pointer `date` is always memory address assigned in lv_calendar_create, so always returning non NULL value, which is not stated in your documentation.

**Expected behavior**
I noticed, that if clicked on calendar without selecting particular day, i get from lvgl values all zero.
To override this behaviour i modified my handler by following

```
    if ( LV_EVENT_CLICKED == event_cb )
    {
        lv_calendar_date_t * prss_date = lv_calendar_get_pressed_date(cal);
        if(prss_date->day != 0 && prss_date->month != 0 && prss_date->year != 0) 
        {
           //my action on calendar clicked
        }
     }
```
My concrete problem is in my action, where i want to minimize calendar on day selection. And when i click on changing month, calendar minimizes.

**Screenshots**
[image]

**Additional context**

No other context.

kisvegabor, 2019-07-26:
Hi,

I've fixed it the `master` branch. Now it should return `NULL` if you press the header.
See 3cfe72c