lvgl/lvgl #1161: next_child=0x0 / lv_obj_clean issue

way5, 2019-07-24:
Hello everyone!
I am having a problem with the following screen structure:
1. 3 Screens given:
      1.a.: First (main) screen contains Image buttons
      1.b.: Second screen tabview + overlay in `lv_layer_sys()` with image button ("go back")
      1.c.: Third screen tabview + overlay in` lv_layer_sys()` with image button ("go back")
2. Display:
     Every time user makes an action global  `lv_obj_t * screen` going to be cleared using: 
    `lv_obj_clean(lv_layer_sys());  
     lv_obj_clean(lv_scr_act()); `
3. Navigation:
   3.a: Main screen -> second screen -> third screen -> second screen -> Main screen -> second screen -> FAIL
   3.b.: Everything works perfect until the transition between the second and third screen that fails with:

```
void lv_obj_clean(lv_obj_t * obj)
{
    lv_obj_t * child = lv_obj_get_child(obj, NULL);
    lv_obj_t * child_next;
    while(child) {
        /* Read the next child before deleting the current
         * because the next couldn't be read from a deleted (invalid) node*/
        child_next = lv_obj_get_child(obj, child);    //!======> here child_next = 0x0 causes FAIL
        lv_obj_del(child);
        child = child_next;
    }
} 
```

Could you give me an advice what is going on here?

BrianPugh, 2019-07-24:
what version of lvgl are you running.

way5, 2019-07-25:
Last update was today's morning:
```
From https://github.com/littlevgl/lvgl
   639409b8..bf2453ce  master     -> origin/master     <-- current branch
   ea00b24c..2093b430  dev-6.1    -> origin/dev-6.1
Updating 639409b8..bf2453ce
```

kisvegabor, 2019-07-25:
@way5 please send a simple code to reproduce the issue

way5, 2019-07-25:
Hi Gabor! I've made a short example, please check attached
[https://www.dropbox.com/s/2lltyifao8ls47f/lvgl_test.zip?dl=0](url)
Let me know if something is missing

kisvegabor, 2019-07-25:
Could you isolate the issue to a function which I can try in the PC simulator? (only the most essential things, just a few buttons to navigate) 

---
I've updated your post because the link wasn't added properly. Now it is:
`[LINK](https://www.dropbox.com/s/2lltyifao8ls47f/lvgl_test.zip?dl=0)`

BrianPugh, 2019-07-25:
@way5 

few things:

On line 102:
```
switch(screen){
```
should be
```
switch(_current_screen){
```

Whenever you are creating `lv_obj` you should be checking for NULL incase you run out of memory.

Those things said, I don't see how this:
```
        child_next = lv_obj_get_child(obj, child);    //!======> here child_next = 0x0 causes FAIL
```
causes a crash? What line actually causes your code to crash?

BrianPugh, 2019-07-25:
Also, there are a few other variable typos; can you submit a minimalistic (ideally single file) working example?

embeddedt, 2019-07-25:
> working example?

By working, I think @BrianPugh means crashing. :wink:

way5, 2019-07-26:
Hi Guys! After some debugging I concluded that this is definitely memory overflow issue. I have not installed PC-simulator yet (not familiar with it), and all the tests for now on a bare-metal that's why may be it took a bit of time to understand. The problem apparently are the same tabviews with an elements on it. I was actually surprised that 50kb of free heap wasn't enough for that. As soon as I changed tabbed UI structure to a paginal, everything seems to be working fine. Will keep digging into the library.
Sorry for a drill and thanks for sharing the LvGL. 
Topic closed

closed 2019-07-26 by way5