lvgl/lvgl #1164: Add PIO config to master

puzrin, 2019-07-30:
https://github.com/littlevgl/lvgl/tree/platformio_test

This is tested and works good enougth. The only thing i suggest - consider change package name if possible. My personal preferences are:

1. `lvgl` (as this repo name)
2. `littlevgl` - (as original, but lower case)

If you don't like to change name - ok, merge as is please.

embeddedt, 2019-07-30:
https://github.com/littlevgl/lvgl/commit/1f4f333a01d25eb35571b878140bdb1bc2b0656c

closed 2019-07-30 by embeddedt

puzrin, 2019-07-30:
@kisvegabor can i drop this branch now https://github.com/littlevgl/pc_simulator_sdl_platformio/tree/library ? That should drop package from registry on next robot scan.

kisvegabor, 2019-07-30:
Let's find a new name for the library first and push it to the registry.

But something is strange here.
[LittlevGL in the registry](https://platformio.org/lib/show/6228/LittlevGL) now is the Arduino version and the "normal" version seems to be overwritten.

Probably the Ardunio version should be called `lv_arduino` and the "normal" version is `lvgl`.

cc @Pablo2048

Pablo2048, 2019-07-30:
Ok, I'll change the library name to ```lv_arduino``` and do a new release...

kisvegabor, 2019-07-30:
Thank you, Pablo!

Pablo2048, 2019-07-31:
Arduino library name change done, release done...