lvgl/lvgl #1173: The roller is not aligned in some cases

liangyongxiang, 2019-08-13:
**Describe the bug**

The roller is not aligned: first drag the roller, then click on another object when the roller has not stopped scrolling. At this time the roller may be misaligned.

**To Reproduce**

I run example `the lv_test_roller_1()` in lv_examples/lv_tests/lv_test_objx/lv_test_roller, then scroll roller3 and quickly click on roller2.

**Expected behavior**

When the roller is scrolling, click on other objects, the roller is still aligned.

**Screenshots**
[image]

**Additional context**

kisvegabor, 2019-08-14:
Do you use the latest `master`?

liangyongxiang, 2019-08-14:
I tested it using the master branch of the https://github.com/littlevgl/pc_simulator_sdl_visual_studio project. So the corresponding lvgl version should be [a421547](https://github.com/littlevgl/lvgl/tree/a42154777414bd768e753c55e3f80248f613de22).

liangyongxiang, 2019-08-14:
Just I switched to the master branch and tried the test. 
But there is a compile error: 
```
fatal error: lvgl/lv_conf_checker.h: No such file or directory
```

liangyongxiang, 2019-08-14:
I changed LV_INDEV_DRAG_THROW to a smaller value such as 2, which makes it easier to reproduce.

liangyongxiang, 2019-08-14:
I think the problem is here:
Drag roller3 and release,  proc->types.pointer.last_obj in indev_drag_throw() is roller3.

When proc->types.pointer.drag_throw_vect.y is not 0,  I clicked on roller2, 
and proc->types.pointer.last_obj is updated to roller2. 

But roller3 may be in an unaligned position at this time.

liangyongxiang, 2019-08-15:
I tried the 7c90b84560aa8cea64c1f8f08b4398d3fb29e7da version, which is the same problem. When the roller3 is not aligned, I clicked on roller2. Then I took a screenshot while dragging roller1

[image]

liangyongxiang, 2019-08-15:
7c90b84 version and Modification of lv_conf.h:

```
/* Drag throw slow-down in [%]. Greater value -> faster slow-down */
#define LV_INDEV_DEF_DRAG_THROW           2
```

Test Result:
[image]

liangyongxiang, 2019-08-15:
Another question, I am not sure if it is a bug.

[image]

If I click on roller3 while scrolling, I think it's better to stop it instead of scrolling (choose b instead of d). 

I can create a new issue if needed.

kisvegabor, 2019-08-15:
I was able to reproduce the issue with `LV_INDEV_DEF_DRAG_THROW` but meanwhile, I found other issues too with the roller.

I'll fix them today or tomorrow.

kisvegabor, 2019-08-16:
I've fixed the issues I've found but still thinking about misalignment issue.

What do you think, what would be expected behavior?
If you click somewhere else stop the roller and ling to the nearest option?

> If I click on roller3 while scrolling, I think it's better to stop it instead of scrolling (choose b instead of d).

I agree. I'll check how could be easily implemented.

embeddedt, 2019-08-16:
> If you click somewhere else stop the roller and fling to the nearest option?

I think that's the simplest method and it makes sense.

liangyongxiang, 2019-08-16:
>  If you click somewhere else stop the roller and ling to the nearest option?

Yes, I think so too.

kisvegabor, 2019-08-16:
I've fixed the misalignment issue in  `master`.
[image]

> If I click on roller3 while scrolling, I think it's better to stop it instead of scrolling (choose b instead of d).

I haven't found a good solution for it yet. Right now I'm working other features and don't have to immerse in it deeper. :frowning_face:

embeddedt, 2019-08-16:
> I haven't found a good solution for it yet. Right now I'm working other features and don't have time to immerse in it deeper.

I think the current fix you have is effective enough.

liangyongxiang, 2019-08-16:
> > I haven't found a good solution for it yet. Right now I'm working other features and don't have time to immerse in it deeper.
> 
> I think the current fix you have is effective enough.

In my application, the range of the roller is 0~59, so I need to scroll quickly to select the appropriate value. And I need to see the selected value in the middle of the quick scroll, so I can only click on the bottom of the roller to stop it.

kisvegabor, 2019-08-17:
> In my application, the range of the roller is 0~59, so I need to scroll quickly to select the appropriate value. And I need to see the selected value in the middle of the quick scroll, so I can only click on the bottom of the roller to stop it.

The problem comes from a feature of the roller: if you are not dragging the roller but just click on an option then it will be selected. The solution can be to have an option to disable this feature and stop+realign on click.

liangyongxiang, 2019-08-19:
> The problem comes from a feature of the roller: if you are not dragging the roller but just click on an option then it will be selected. The solution can be to have an option to disable this feature [...]

OK. Because I am using V5.3 in my project, I don't need to modify this part. 
My problem is solved(click feature is not necessary for my project).
This issue needs to be closed after the repair of the two features is supported. Or close now?

kisvegabor, 2019-08-20:
I'd look into this a little bit later but I'm quite busy now.

Let's close it now to keep the issues clear. If others will have the same request I'll treat it with a higher priority and reopen this issue.

closed 2019-08-20 by kisvegabor