lvgl/lvgl #1175: Delete the screen issue

Masi2017, 2019-08-19:
Hi All,

I have created a simple screen including two buttons, when I try to delete the screen, system crashes. I migrated from ver5.3 to ver6.00 and I haven't had the same issue in ver5.3.

Here is a simplified code to recreate the issue:

lv_obj_t *parent = NULL;

static void event_handler(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        printf("Clicked\n");
        lv_obj_del(parent);
    }
    else if(event == LV_EVENT_VALUE_CHANGED) {
        printf("Toggled\n");
    }
}

void lv_ex_btn_1(void)
{
    parent = lv_obj_create(NULL, NULL);
    lv_scr_load(parent);

    lv_obj_t * label;

    lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_event_cb(btn1, event_handler);
    lv_obj_align(btn1, NULL, LV_ALIGN_CENTER, 0, -40);

    label = lv_label_create(btn1, NULL);
    lv_label_set_text(label, "Button");

    lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_event_cb(btn2, event_handler);
    lv_obj_align(btn2, NULL, LV_ALIGN_CENTER, 0, 40);
    lv_btn_set_toggle(btn2, true);
    lv_btn_toggle(btn2);
    lv_btn_set_fit2(btn2, LV_FIT_NONE, LV_FIT_TIGHT);

    label = lv_label_create(btn2, NULL);
    lv_label_set_text(label, "Toggled");
}

kisvegabor, 2019-08-20:
Hi,

It happens because you delete the active screen. Therefore no valid active screen remains after the click.

I've modified you code to make it work:
```c
lv_obj_t *parent = NULL;
lv_obj_t *parent2 = NULL;

static void event_handler(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        lv_scr_load(parent2);
        printf("Clicked\n");
        lv_obj_del(parent);
    }
    else if(event == LV_EVENT_VALUE_CHANGED) {
        printf("Toggled\n");
    }
}

void lv_ex_btn_1(void)
{
    parent = lv_obj_create(NULL, NULL);
    lv_scr_load(parent);

    parent2 = lv_obj_create(NULL, NULL);

    lv_obj_t * label;

    lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_event_cb(btn1, event_handler);
    lv_obj_align(btn1, NULL, LV_ALIGN_CENTER, 0, -40);

    label = lv_label_create(btn1, NULL);
    lv_label_set_text(label, "Button");

    lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_event_cb(btn2, event_handler);
    lv_obj_align(btn2, NULL, LV_ALIGN_CENTER, 0, 40);
    lv_btn_set_toggle(btn2, true);
    lv_btn_toggle(btn2);
    lv_btn_set_fit2(btn2, LV_FIT_NONE, LV_FIT_TIGHT);

    label = lv_label_create(btn2, NULL);
    lv_label_set_text(label, "Toggled");
}
```

Masi2017, 2019-08-21:
@kisvegabor Thanks I did so and the problem is resolved.

closed 2019-08-21 by Masi2017