lvgl/lvgl #1176: ESP32 Pico v4 + SSD2805 won't work

ceebo2019, 2019-08-20:
Hi there,

I tried to set up an ESP32 Pico v4 with the "i2c_8080_lcd" from https://blog.littlevgl.com/2019-02-02/use-ipod-nano6-lcd-for-littlevgl

I connected it as shown in the description, but the Display stays black. The CTP works fine. 

I also set the DIP-Switches like this:
0 1 1 0 0 0 0 1 1

In my view, it is set up like this:
NC: 0
IF_SEL: 1
IF_MUX: 1
LANE_SEL: 0
PS3, PS2: 0, 0
PS1, PS0: 1, 1

What is the problem? I connected all Data Pins (D0-D7), the CS#, WR# and DC Pin, the RST Pin and the Power-Supply to the Display.

All Voltages needed, are there. 

And excuse my bad english, I'm from germany and had to learn the language piece by piece..

kisvegabor, 2019-08-20:
@techtoys Could you take look at it?

techtoys, 2019-08-21:
Try to set IF_MUX 0, that means jumper position up, to 'ON'. The layout is somehow a bit funny. Switching it to 'ON' means closing the switch, and hence short to ground '0'.

The other jumper settings are fine, for 8080 8-bit mode, 1 data lane.

The next important pin to use is VDDIO_CTLR. This is pin 39 on pin header P1. This pin is wired to a mosfet to control VDD_CORE supply to SSD2805. There is no need to use this pin in our case but we need to short VDDIO_CTLR to ground so that VDDIO_CORE is available at all time.

The last thing to check is backlight power and PWM for a backlight IC RT9293.
First make sure power supply to EXT_5V is OK, with 5V. This can be sourced from ESP32 PICO's USB port. PWM pin is for backlight ON/OFF. Set this pin to 3.3V to switch on backlight forever to see.

Both EXT_5V and PWM are available on P2 of SSD2805 breakout board.

ceebo2019, 2019-08-21:
@techtoys 

I tried your suggestion, to connect the VDDIO_CTLR-Pin to ground, but the VDDIO_CORE drops only around 0.3V, if the CTLR-Pin is NOT connected to ground. Should the Voltage VDDIO_CORE not be nearly 0V?
If I connect Pin 39 of P1 to ground, the VDDIO_CORE is around 3.29V, what should be enough in my opinion, to power up the SSD2805. The EXT_5V is powered by the ESP32 and is around 4.8V.

I also switched the DIP-Switch 3 to "On" for IF_MUX 0 and connected the PWM-Pin to 3.3V and the backlight lit up, but the display still shows no picture.

I don't know, what I'm doing wrong. :(

techtoys, 2019-08-21:
If CTLR-Pin floats, VDDIO_CORE may show a voltage close to 3V but it is not sourcing current. It is a mosfet control. As a result, there may be voltage level but not a proper power source if CTLR-Pin is left floating. Definitely you need to short CTLR-Pin to switch that mosfet on.

VDDIO_CORE at 3.29V is good.

EXT_5V at 4.8V is good, and PWM pin at 3.3V is also good. With backlight ON is good.

No picture because the firmware has issues. There are three examples, one for Touch Screen demo and the other for CTP only (SSD2541 driver test), one for LCD only (I2S 8080 LCD driver), and the last one for CTP + LCD(Project).  Those projects for LCD required a lot more wires than CTP demo. For CTP demo you only need I2C. But for LCD you need 8 wires + DC + CS + WR. I would recommend checking the jumper wires and make sure they are secured.

embeddedt, 2019-08-25:
>  I believe it is a kind of hardware or wiring problem rather than some issue with LittlevGL. It is not fair to borrow this platform for supporting the SSD2805 EVK.

Should this issue be closed/transferred to the forum then?

kisvegabor, 2019-08-26:
> Should this issue be closed/transferred to the forum then?

I agree, let's continue in the forum to make the discussion more visible for people.

techtoys, 2019-08-26:
> let's continue in the forum to make the discussion more visible for people

Have no intention to hide anything from this forum. Previous thinking is that the issue was hardware specific. I still believe a blank screen is tight to messy jumper cables or misconnection etc., since CTP is working but no display. That means project compile is good. But there is no message from ceebo2019 since then. If there is no problem to continue with hardware/board specific discussion in this forum, for sure continuing here is absolutely okay.

kisvegabor, 2019-08-27:
@embeddedt 
Wow, how did you do that? :smile:

kisvegabor, 2019-08-27:
@embeddedt Thank you!

ceebo2019, 2019-08-27:
Hi there, 

with the help of techtoys, I have written a short summary of my troubleshooting, that helped me to get to a solution. The display works and the quality ist quite high for this little thing. 

1. VDDIO_CTRL pin on SSD2805 EVK board should be short to GND, to enable VDD supply to SSD2805.
2. EXT_5V should be supplied with 5V (from usb is OK). This is for backlight controller IC.
3. PWM pin for backlight controller IC should be driven high (3.3V) to enable it.
4. DIP switch on SSD2805 EVK is, from left to right, 01000011. (0=ON on dip switch)
5. PCLK/RD# on SD2805 EVK board should be driven high, not floating. This pin is a pull-down pin by default. If there is no reading required, pull it up to drive RD# pin high forever.

Thanks a lot @techtoys