lvgl/lvgl #1177: New Font converter issue (ver6.00)

Masi2017, 2019-08-20:
Hi All,

I am trying to migrate from ver5.3 to ver6.0. I had created my own symbols and I am trying to convert my .ttf file using the lvgl converter. My symbols' unicodes are started at 0xf293 and ended at  0xf2bc.
I tested the ttf file with the old converter(ver5.3) and worked fine but the new converter keeps giving me the following error:
Font “NLTHomeIconsV3Regular.ttf” doesn’t have any characters included in range 0xf293-0xf2bc

I've attached the output from the old converter and also my ttf file.
[
[NLTCombinedFonts-F293.zip](https://github.com/littlevgl/lvgl/files/3522329/NLTCombinedFonts-F293.zip)

[NLTHomeIconsV1.zip](https://github.com/littlevgl/lvgl/files/3522330/NLTHomeIconsV1.zip)

](url)

embeddedt, 2019-08-20:
@kisvegabor and I have already seen your post on the forum. Please be patient; we'll respond when we get a chance.

Masi2017, 2019-08-21:
@embeddedt Since I wasn't able to add the attachment there, I created this. Thanks for your help and effort.

embeddedt, 2019-08-21:
No problem then.

puzrin, 2019-08-21:
If you open your font in FontForge, then try to export as truetype (file -> create font), you will get warning that font contains errors - self-cross contours, missed extreme points and so on.

I think, if you fix your font, convertor will eat it without problems.

```
./lv_font_conv.js --format dump --size 32 --bpp 4 --font NLTCombinedFonts-F293.ttf --range 0x01-0xFFFF -o ./1111
```

Glyph file names are correct, but content is missed due bad contours.

Masi2017, 2019-08-22:
@puzrin Thanks for replying. But as I mentioned I am able to convert the font using the older converter (ver5.3 and I had included the output) and also I have installed theat on the computer and I can see the symbols fine.

puzrin, 2019-08-22:
There are different TTF rendering engines. If one  engine can autofix errors, that does not mean other one must be able to do the same.

Please, understand me right - it's not rational to spend time for such fixes. It would be huge spends with almost zero added value.

Masi2017, 2019-08-22:
@puzrin Thanks for the information. I should mention that I have a .ttf file  (attached here) and I am able to convert it by the new converter but when I open that by FontForge and just try to generate another font without doing any changes, I receive those errors and also I am not able to convert it by the new converter. However, I'll try to fix those errors and see if I can convert it. 
[NLTHomeIconsV2.zip](https://github.com/littlevgl/lvgl/files/3531668/NLTHomeIconsV2.zip)

puzrin, 2019-08-22:
FYI, here is short instruction how to prepare SVG for fonts https://github.com/fontello/fontello/wiki/How-to-use-custom-images#importing-svg-images. Not too difficult.

Probably, some font editors (and may be fontforge) can have tools to autofix bad data. But i never used those, and can not give you exact recipe how to repair all with one click.

Masi2017, 2019-09-03:
Thanks, I'll try that.

closed 2019-09-03 by Masi2017