lvgl/lvgl #1188: lv_ta: When the long text is automatically wrapped, the text before the line break exceeds the display area.

QQ1113845021, 2019-09-10:
Possible error location：
     src/lv_misc/lv_txt.c
         lv_txt_utf8_prev()：This function fails
        lv_txt_utf8_size(): 5.3 update 6.0，Modified code

kisvegabor, 2019-09-10:
Please send a code to reproduce.

QQ1113845021, 2019-09-10:
parent: lv_btn *btton;
child: lv_img *img

click btton, img set src  123.png
click btton, img set src  234.png (Not the same as 123.png)

But img doesn't show 234.png

kisvegabor, 2019-09-10:
Please open a new issue for it as it's a different bug and add code teher too to reproduce it.

QQ1113845021, 2019-09-10:
The same code, 5.3 no error, 6.0 appears。
1、call lv_ta_set_text(); 
2、lvgl call lv_draw_label();
3、lv_draw_label(); call  lv_txt_get_next_line(); The return value is increased by one than expected.
4、lv_txt_get_next_line(); call lv_txt_encoded_prev(); Unable to get the byte position of the previous character.

diff 5.3 vs 6.0 above
`
static uint8_t lv_txt_utf8_size(const char * str)
{
    if((str[0] & 0x80) == 0) return 1;
    else if((str[0] & 0xE0) == 0xC0) return 2;
    else if((str[0] & 0xF0) == 0xE0) return 3;
    else if((str[0] & 0xF8) == 0xF0) return 4;
    return 0;
}
`
`static uint8_t lv_txt_utf8_size(const char * str)
{
    if((str[0] & 0x80) == 0)
        return 1;
    else if((str[0] & 0xE0) == 0xC0)
        return 2;
    else if((str[0] & 0xF0) == 0xE0)
        return 3;
    else if((str[0] & 0xF8) == 0xF0)
        return 4;
    return 1; /*If the char was invalid step tell it's 1 byte long*/
}`

kisvegabor, 2019-09-10:
I mean add a code which can be copy-pasted to the simulator to demonstrates the issue. Same for the [other issue](https://github.com/littlevgl/lvgl/issues/1189).

embeddedt, 2019-09-10:
----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

Use the **How-to** template and fill out the template completely. Otherwise we will not have enough information to solve your problem, and your post(s) may not get attention.

closed 2019-09-10 by embeddedt