lvgl/lvgl #1190: ESP-IDF NVS Write operations cause crashes after LVGL initialised

huming2207, 2019-09-11:
## Describe the bug

Hi there,

I've got a M5StickC module and I've ported LittlevGL to it. Everything works great except it doesn't get along ESP-IDF's NVS library. If I commit any NVS write operations after LittlevGL initialised, the program will crash.

Here's my setup: 

- uC: ESP32-PICO-D4 (ESP32 with 4MB SPI Flash SiP chip)
- Display: ST7735R, 160x80
- LittlevGL version: v6.0.2, https://github.com/littlevgl/lvgl/commit/eefd9c4e0d0d434aa08c658db9c1f397e87804c6
- ESP-IDF version: v4.1-dev-141-ga7e8d87d3
- Compiler: xtensa-esp32-elf-gcc 8.1, esp32-2019r1

## To Reproduce

Try this commit: https://github.com/huming2207/totp-vault/commit/8ffab59e38e3eaa32336eeea2924a3a285829b53

and then un-comment this line: https://github.com/huming2207/totp-vault/blob/8ffab59e38e3eaa32336eeea2924a3a285829b53/main/lvgl_main.cpp#L95

## Expected behavior

Should works fine, NVS saves the changes and LVGL continues to work

## ESP-IDF log

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6532
load:0x40078000,len:11552
load:0x40080400,len:6700
entry 0x4008077c
I (29) boot: ESP-IDF v4.1-dev-141-ga7e8d87d3 2nd stage bootloader
I (29) boot: compile time 22:43:52
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11d08 ( 72968) map
I (120) esp_image: segment 1: paddr=0x00021d30 vaddr=0x3ffb0000 size=0x02148 (  8520) load
I (124) esp_image: segment 2: paddr=0x00023e80 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/hu/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (128) esp_image: segment 3: paddr=0x00024288 vaddr=0x40080400 size=0x0bd88 ( 48520) load
I (157) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x29754 (169812) map
0x400d0018: _stext at ??:?

I (218) esp_image: segment 5: paddr=0x00059774 vaddr=0x4008c188 size=0x00a20 (  2592) load
0x4008c188: xTaskRemoveFromEventList at /home/hu/esp/esp-idf/components/freertos/tasks.c:3118

I (227) boot: Loaded app from partition at offset 0x10000
I (228) boot: Disabling RNG early entropy source...
I (230) cpu_start: Pro cpu up.
I (233) cpu_start: Application information:
I (238) cpu_start: Project name:     totp-vault
I (243) cpu_start: App version:      a5fb3d2-dirty
I (249) cpu_start: Compile time:     Sep 10 2019 22:36:47
I (255) cpu_start: ELF file SHA256:  5519f16cb738d8dd...
I (261) cpu_start: ESP-IDF:          v4.1-dev-141-ga7e8d87d3
I (267) cpu_start: Starting app cpu, entry point is 0x400812f4
0x400812f4: call_start_cpu1 at /home/hu/esp/esp-idf/components/esp32/cpu_start.c:280

I (0) cpu_start: App cpu up.
I (278) heap_init: Initializing. RAM available for dynamic allocation:
I (285) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (291) heap_init: At 3FFBE9F0 len 00021610 (133 KiB): DRAM
I (297) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (303) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (310) heap_init: At 4008CBA8 len 00013458 (77 KiB): IRAM
I (316) cpu_start: Pro cpu start user code
I (334) spi_flash: detected chip: generic
I (335) spi_flash: flash io: dio
I (335) cpu_start: Chip Revision: 1
I (336) pm_esp32: Frequency switching config: CPU_MAX: 240, APB_MAX: 240, APB_MIN: 40, Light sleep: DISABLED
I (347) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
I (359) m5stickc: I2C_0 initialised
I (369) m5stickc: PMIC initialised
I (369) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (409) token_model: Got token: 4
I (409) uri_parser: Free heap after parsing: 248488, Max heap block: 119648
I (409) uri_parser: Free heap after parsing: 248192, Max heap block: 119352
I (419) uri_parser: Free heap after parsing: 247936, Max heap block: 119016
I (459) esp-lvgl: Initialising LittlevGL
I (459) esp-lvgl: Initialising ST7789
I (459) st7735: Performing GPIO init...
I (469) st7735: GPIO initialization finished, resetting OLED...
I (719) st7735: Performing SPI init...
I (719) st7735: SPI initialization finished, sending init sequence to IPS panel...
I (1029) esp-lvgl: Initialising LVGL driver
I (1029) esp-lvgl: Registering LVGL driver
I (1029) esp-lvgl: Registering LVGL tick callback
I (1029) esp-lvgl: Starting LVGL main task
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400dbf1c  PS      : 0x00060034  A0      : 0x80082670  A1      : 0x3ffb0620  
0x400dbf1c: lv_tick_inc at /home/hu/Projects/totp-vault/build/../main/lib/lvgl/src/lv_hal/lv_hal_tick.c:53

A2      : 0x00000003  A3      : 0x00060023  A4      : 0x00060021  A5      : 0x00000020  
A6      : 0x0000002a  A7      : 0x3ffc05d0  A8      : 0x8008350c  A9      : 0x00000000  
A10     : 0x0000000a  A11     : 0x4008677c  A12     : 0x0000de60  A13     : 0x00000020  
0x4008677c: spi_flash_chip_generic_wait_idle at /home/hu/esp/esp-idf/components/spi_flash/spi_flash_chip_generic.c:253

A14     : 0x0000000a  A15     : 0x3ffc05b0  SAR     : 0x00000017  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x4000c050  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400dbf1c
0x400dbf1c: lv_tick_inc at /home/hu/Projects/totp-vault/build/../main/lib/lvgl/src/lv_hal/lv_hal_tick.c:53

ELF file SHA256: ad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0ba

Backtrace: 0x400dbf19:0x3ffb0620 0x4008266d:0x3ffb0640 0x4008bcb4:0x3ffb0660 0x4008a187:0x3ffb0680 0x4008a37d:0x3ffb06a0 0x40083106:0x3ffb06b0 0x4000c04d:0x3ffc0550 |<-CORRUPTED
0x400dbf19: lv_indev_get_next at /home/hu/Projects/totp-vault/build/../main/lib/lvgl/src/lv_hal/lv_hal_indev.c:119

0x4008266d: esp_vApplicationTickHook at /home/hu/esp/esp-idf/components/esp_common/src/freertos_hooks.c:41

0x4008bcb4: xTaskIncrementTick at /home/hu/esp/esp-idf/components/freertos/tasks.c:2473

0x4008a187: xPortSysTickHandler at /home/hu/esp/esp-idf/components/freertos/port.c:289

0x4008a37d: _frxt_timer_int at /home/hu/esp/esp-idf/components/freertos/portasm.S:303

0x40083106: _xt_lowint1 at /home/hu/esp/esp-idf/components/freertos/xtensa_vectors.S:1153

Rebooting...

```

## Additional info

For my ST7735R display HAL, it's located here: https://github.com/huming2207/totp-vault/blob/8ffab59e38e3eaa32336eeea2924a3a285829b53/main/lib/hal/lvgl_hal_st7735r.c

embeddedt, 2019-09-11:
So, I know nothing about ESP32 itself, but are you writing to the same flash that LittlevGL is running out of?

huming2207, 2019-09-11:
@embeddedt Yes. The NVS partition is in the internal flash, not external SPI flash chip. 

I suspect it may be some sorts of low-level racing issues (FreeRTOS/LVGL event queue ticks, interrupts, etc.).

embeddedt, 2019-09-11:
> I suspect it may be some sorts of low-level racing issues (FreeRTOS/LVGL event queue ticks, interrupts, etc.).

I'm suspicious that it's an issue with your port, and not LittlevGL itself.

Are you calling LittlevGL APIs from multiple threads? If so, then be sure to read the [relevant portion](https://docs.littlevgl.com/en/html/porting/os.html) of the documentation.

> The NVS partition is in the internal flash, not external SPI flash chip.

I am assuming you are running LittlevGL from internal flash? If so, maybe ESP32 doesn't support reading and writing to the same flash device at the same time? If that's the case, you could try running your code out of RAM instead.

huming2207, 2019-09-11:
Hello @embeddedt ,

> I am assuming you are running LittlevGL from internal flash? If so, maybe ESP32 doesn't support reading and writing to the same flash device at the same time? If that's the case, you could try running your code out of RAM instead.

No. ESP32 does support that. 

> Are you calling LittlevGL APIs from multiple threads? If so, then be sure to read the relevant portion of the documentation.

I'm running the LVGL main loop in another thread. But I don't think it really matters. Because the flash chip access use an independent, internal SPI peripheral that can't be accessed directly via ESP-IDF SPI peripheral drivers (but can be accessed through SPI flash APIs, e.g. NVS or SPIFFS). My display HAL implementation uses HSPI and it's completely different from the internal SPI peripheral used for read/write flash chip.

BrianPugh, 2019-09-11:
Add these to your `lv_conf.h` and remove
` esp_register_freertos_tick_hook(lv_tick_cb); ` (L81) in your `lvgl_main.cpp`. ESP-IDF has some weird thing with using `esp_register_freertos_tick_hook`, lvgl, and NVS at the same time.

```
#include "esp_attr.h"
#define LV_ATTRIBUTE_TICK_INC IRAM_ATTR
#define LV_ATTRIBUTE_TASK_HANDLER IRAM_ATTR

/*===================
 *  HAL settings
 *==================*/

/* 1: use a custom tick source.
 * It removes the need to manually update the tick with `lv_tick_inc`) */
#define LV_TICK_CUSTOM     1
#if LV_TICK_CUSTOM == 1
#define LV_TICK_CUSTOM_INCLUDE  "esp_timer.h"       /*Header for the sys time function*/
#define LV_TICK_CUSTOM_SYS_TIME_EXPR (esp_timer_get_time()/1000)     /*Expression evaluating to current systime in ms*/
#endif   /*LV_TICK_CUSTOM*/
```

huming2207, 2019-09-11:
Hello @BrianPugh ,

Thanks for your reply. I will have a try tonight.

I'll also try `esp_register_freertos_tick_hook_for_cpu()` as well and pins to Core 1, or just use a timer tick instead.

Regards,
Jackson