lvgl/lvgl #1191: focus on list object issue

Masi2017, 2019-09-11:
**Describe the bug**

Hi All

I've created a list and then when user presses one of the button in the list, list is deleted and another list is created , second list will be added to a group and list is focused. Once user presses one of the btn in the second list , list will be deleted and another list is created but this never happens and system creches  ( creating first list is ok - second is ok, third one never is created and I get "misaligned load".
I should mention that if I add all the list to the group or if I never add any of the lists to the group everything is fine, the issue is when one of the list is added to a group and focused.
I've added the code here:

**To Reproduce**

`
lv_obj_t * list_var = NULL;
void lv_ex_list_2(void);
void lv_ex_list_1(void);

static void list_event_handler(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        printf("Clicked: %s\n", lv_list_get_btn_text(obj));

        lv_obj_del_async(list_var);
        lv_ex_list_2();
      }
}

static void list_event_handler1(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        printf("Clicked: %s\n", lv_list_get_btn_text(obj));

      //  lv_group_remove_obj(list_var);
        lv_obj_del_async(list_var);
        lv_ex_list_1();
    }
}

void lv_ex_list_1(void)
{

    /*Create a list*/
    lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_size(list1, 160, 200);
    lv_obj_align(list1, NULL, LV_ALIGN_CENTER, 0, 0);
    list_var = list1;
    /*Add buttons to the list*/

    lv_obj_t * list_btn;

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_FILE, "New");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_DIRECTORY, "Open");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_CLOSE, "Delete");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_EDIT, "Edit");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_SAVE, "Save");
    lv_obj_set_event_cb(list_btn, list_event_handler);

//   lv_group_add_obj(g1, list1);
//   lv_group_focus_obj(list1);
}

void lv_ex_list_2(void)
{
  if (!g1)
  {
  printf("create grp\n");
    g1 = lv_group_create();
    lv_group_set_style_mod_cb(g1, nlt_focus_style_mod);
    lv_indev_set_group(kb_indev, g1);
  }

    /*Create a list*/
    lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_size(list1, 160, 200);
    lv_obj_align(list1, NULL, LV_ALIGN_CENTER, 0, 0);
    list_var = list1;
    /*Add buttons to the list*/

    lv_obj_t * list_btn;

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_FILE, "New test");
    lv_obj_set_event_cb(list_btn, list_event_handler1);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_DIRECTORY, "Open test");
    lv_obj_set_event_cb(list_btn, list_event_handler1);

    lv_group_add_obj(g1, list1);
    lv_group_focus_obj(list1);
}

`

**Expected behavior**

**Screenshots**
[image]

**Additional context**
MCU : K210 (FreeRtos)
lvgl revision : lvgl 6.0

embeddedt, 2019-09-11:
Just to be sure, are you using the latest version of LittlevGL (from the `master` branch)?

Masi2017, 2019-09-11:
I have downloaded the lvgl about last month ago,  do you want me to updated again?

Masi2017, 2019-09-11:
Updated, same issue.

Also modified the firmware to create the second list after LV_EVENT_DELETE is received in first list event handler and by doing this system is crashed when second list is created and focused but if I don't focous on the second list object everything is OK, here is the modified code:
`
lv_obj_t * list_var = NULL;
void lv_ex_list_2(void);
void lv_ex_list_1(void);

static void list_event_handler(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        printf("list 1 Clicked: %s\n", lv_list_get_btn_text(obj));
        lv_obj_del_async(list_var);
      }

      if (event == LV_EVENT_DELETE)
      {
        printf("LV_EVENT_DELETE lv_ex_list_2\n");
      }
}

static void list_obj_event(lv_obj_t * obj, lv_event_t event)
{
  if (event == LV_EVENT_DELETE)
  {
      printf("LV_EVENT_DELETE list_obj_event\n");
    lv_ex_list_2();
  }

}

static void list_obj_event1(lv_obj_t * obj, lv_event_t event)
{
  if (event == LV_EVENT_DELETE)
  {
      printf("LV_EVENT_DELETE list_obj_event1\n");
    lv_ex_list_1();
  }
}

static void list_event_handler1(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        printf("list 2 Clicked: %s\n", lv_list_get_btn_text(obj));

      //  lv_group_remove_obj(list_var);
        lv_obj_del_async(list_var);
    }

    if (event == LV_EVENT_DELETE)
    {
        printf("LV_EVENT_DELETE lv_ex_list_1\n");
    }
}

void lv_ex_list_1(void)
{

    /*Create a list*/
    lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_size(list1, 160, 200);
    lv_obj_align(list1, NULL, LV_ALIGN_CENTER, 0, 0);
    list_var = list1;
    lv_obj_set_event_cb(list1, list_obj_event);

    /*Add buttons to the list*/

    lv_obj_t * list_btn;

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_FILE, "New");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_DIRECTORY, "Open");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_CLOSE, "Delete");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_EDIT, "Edit");
    lv_obj_set_event_cb(list_btn, list_event_handler);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_SAVE, "Save");
    lv_obj_set_event_cb(list_btn, list_event_handler);

//   lv_group_add_obj(g1, list1);
//   lv_group_focus_obj(list1);
}

void lv_ex_list_2(void)
{
  if (!g1)
  {
  printf("create grp\n");
    g1 = lv_group_create();
    lv_group_set_style_mod_cb(g1, nlt_focus_style_mod);
    lv_indev_set_group(kb_indev, g1);
  }

    /*Create a list*/
    lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_size(list1, 160, 200);
    lv_obj_align(list1, NULL, LV_ALIGN_CENTER, 0, 0);
    list_var = list1;
    lv_obj_set_event_cb(list1, list_obj_event1);

    /*Add buttons to the list*/

    lv_obj_t * list_btn;

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_FILE, "New test");
    lv_obj_set_event_cb(list_btn, list_event_handler1);

    list_btn = lv_list_add_btn(list1, LV_SYMBOL_DIRECTORY, "Open test");
    lv_obj_set_event_cb(list_btn, list_event_handler1);

  //  lv_group_add_obj(g1, list1);
//    lv_group_focus_obj(list1);
}

`

kisvegabor, 2019-09-14:
You can simply delete the object as you did now.

---

Note that GitHub doesn't send a notification when you edit a comment. I saw in the notification email that "Thank you very much! works works fine now!" and just came here to close the issue. So it's better to add a new comment in similar cases.

Masi2017, 2019-09-16:
Thanks.
Oh sure I didn't know that.

closed 2019-09-16 by Masi2017