lvgl/lvgl #1192: True Double buffer should use/respect lv_disp_drv->set_px_cb

BrianPugh, 2019-09-12:
In true double buffer mode, the regions copied over to the second buffer do not respect the pixel mappings in `lv_disp_drv->set_px_cb`. For example, if we are using a monochrome display and store each pixel as a single bit in our vdb, then the following are wrong:

- Too much data (8x) is copied over
- `start_offs` is increased by too much (8x too much)

https://github.com/littlevgl/lvgl/blob/dev-6.1/src/lv_core/lv_refr.c#L201

```
                    for(y = disp_refr->inv_areas[a].y1; y <= disp_refr->inv_areas[a].y2; y++) {
                        memcpy(buf_act + start_offs, buf_ina + start_offs, line_length);
                        start_offs += hres * sizeof(lv_color_t);
                    }
```

kisvegabor, 2019-09-14:
You are right. I've begun to implement this but realized that it's not possible with the current API.
Here we need to copy from one buffer to another. We have `set_px_cb` to write the new buffer, but there is no `get_px_cb` to read the old buffer.

I see two solutions:
- add `get_px_cb`: I'm not sure it's worth it because it'd increase the API for one corner case
- state in the docs that it's not supported

BrianPugh, 2019-09-14:
I don't actually need it, I was just experimenting to see if double buffering would improve my draw performance, so updating the docs to say its not supported is fine by me. To not increase the API, we could do some clever stuff with having `set_px_cb` returning the value at pixel `x,y`; but it seems like an unintuitive hack.

kisvegabor, 2019-09-16:
I've updated the [docs](https://docs.littlevgl.com/en/html/porting/display.html#display-driver) to mention this configuration is not supported.

>  ... but it seems like an unintuitive hack.

I agree. My view is `set_px_cb` is relatively slow anyway, therefore, I wouldn't sacrifice simplicity for a minor speed up a slow configuration.

embeddedt, 2019-09-16:
> My view is `set_px_cb` is relatively slow anyway, therefore, I wouldn't sacrifice simplicity for a minor speed up in a slow configuration.

So, if my understanding is correct, `set_px_cb` is a method where you can reduce buffer size at the cost of speed?

BrianPugh, 2019-09-16:
@embeddedt It's my understanding that `set_px_cb` is a callback for storing the drawed pixels into the VDB. By default, this is an identity mapping (1pixel to 1 `lv_color`). However, for a monochrome screen, `lv_color` is still 8bits, so you can pack them in closer via bitshifting and modulo (which is inherently slower). Of course, the `flush_cb` also has to inherently have the reverse mapping to flush it to display, but since they go together it's no big deal.

BrianPugh, 2019-09-16:
as for my use case, I'd rather just not use it rather than increasing the size of my VDB by 16x 😛

kisvegabor, 2019-09-17:
> So, if my understanding is correct, set_px_cb is a method where you can reduce buffer size at the cost of speed?

Actually, yes. You can use bit buffer (which is smaller) but it's more costly to set pixels in it.

embeddedt, 2019-09-19:
@BrianPugh Can this issue be closed?

BrianPugh, 2019-09-19:
yup!

closed 2019-09-19 by BrianPugh