lvgl/lvgl #1193: Navigation issue when Kb Background style is changed

Masi2017, 2019-09-13:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

Hi All,

I've created a Kb object and added to a group. I've done some modifications in the keyboard background style (see the code below) and I noticed that this causes some issues in moving among the btns in the keyboard object using navigation keys ( hardware key). for example when I press DOWN key while 'e' button has been selected 'x' btn in the third row of the keyboard is selected. and also I am not able to move up when I press the UP key while any key in the second row has been selected ( e.g. if while 'g' is selected UP key is pressed it doesn't move to the upper row)

**To Reproduce**

`
void lv_ex_kb_1(void)
{
    /*Create styles for the keyboard*/
    static lv_style_t rel_style, pr_style, bg_style;
    /*Create a keyboard and apply the styles*/
    lv_obj_t *kb = lv_kb_create(lv_scr_act(), NULL);

    lv_style_copy(&rel_style, &lv_style_btn_rel);
    rel_style.body.radius = 0;
    rel_style.body.border.width = 1;
    rel_style.text.font = &lv_font_roboto_28;

    lv_style_copy(&pr_style, &lv_style_btn_pr);
    pr_style.body.radius = 0;
    pr_style.body.border.width = 1;
    pr_style.text.font = &lv_font_roboto_28;

     lv_style_copy(&bg_style, &lv_style_pretty);
     bg_style.body.padding.inner = 10;
     bg_style.body.padding.left = 0;
     bg_style.body.padding.right = 0;
     bg_style.body.padding.top = 0;
     bg_style.body.padding.bottom = 0;

    lv_kb_set_cursor_manage(kb, true);
    lv_kb_set_style(kb, LV_KB_STYLE_BG, &bg_style);
    lv_kb_set_style(kb, LV_KB_STYLE_BTN_REL, &rel_style);
    lv_kb_set_style(kb, LV_KB_STYLE_BTN_PR, &pr_style);

    lv_obj_align(kb, NULL, LV_ALIGN_IN_BOTTOM_LEFT, 0, 0);

    /*Create a text area. The keyboard will write here*/
    lv_obj_t *ta = lv_ta_create(lv_scr_act(), NULL);
    lv_obj_align(ta, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);
    lv_ta_set_text(ta, "");

    /*Assign the text area to the keyboard*/
    lv_kb_set_ta(kb, ta);
    lv_group_add_obj(g1, kb);
}
`

kisvegabor, 2019-09-14:
Thanks for the report. I've fixed it in the `master` branch.

Masi2017, 2019-09-16:
No problem Thank you!  tested works fine.

closed 2019-09-16 by Masi2017