lvgl/lvgl #1197: About adding attribute definitions such as LV_MEM_ATTR.

tkashi-github, 2019-09-19:
Hello.

First of all, thank you for LVGL!

Q. Are there any plans to add attribute definitions such as LV_MEM_ATTR to functions in lvgl/src / * files?

I am using IMXRT1060-EVK and FreeRTOS. I have added an attribute in lv_conf.h to place LVGL in TCM for performance (https://github.com/tkashi-github/IMXRT1060-EVK/blob/develop/IMXRT1060-EVK_APL/source/include/lv_conf.h).

I am also preparing a pull request for dev6.1. Please tell me if you need it.

BR,
T.Kashiwagi

kisvegabor, 2019-09-19:
Hi,

Do you mean to prefix all functions? E.g. `SOME_ATTR void lv_obj_set_width(lv_obj_t * obj, lv_coord_t w)`?

If so, can you measure or estimate the possible performance gain?

tkashi-github, 2019-09-19:
Thank you for your quick reply!

> Do you mean to prefix all functions? E.g. SOME_ATTR void lv_obj_set_width(lv_obj_t * obj, lv_coord_t w)?

Yes. I wrote as like follows against lvgl/src/* files. 
`LV_FUNC_OBJX_KB_ATTR void lv_kb_def_event_cb(lv_obj_t * kb, lv_event_t event)`

> If so, can you measure or estimate the possible performance gain?

Sorry,  I have no good idea so far about LVGL.

However, program execution from IMXRT SDRAM is known to have poor performance. I know that TCM is about 8 times better than SDRAM（https://community.nxp.com/message/1074457）.

Do you know which LVGL API is appropriate for measuring performance?

 BR,
T.Kashiwagi

embeddedt, 2019-09-19:
> I know that TCM is about 8 times better than SDRAM

I would expect that to be the case, especially if cache is disabled. TCM is part of the processor while SDRAM is (usually) a separate chip.

> I am also preparing a pull request for dev6.1.

If we were to accept this PR, I do not think we need individual definitions like `LV_VAL_OBJX_CONT_ATTR`. That just adds verbosity. I can't see anyone wanting to put the text area code in TCM and the calendar in SDRAM, for example.

However, while I think this is a highly specialized use case, I would tolerate adding the following attributes, since that is the only portable solution I can think of:

* `LV_FUNC_ATTR` - lvgl functions (`.text`),
* `LV_BSS_ATTR` -  zero data (`.bss`), and
* `LV_DATA_ATTR` - initialized data (`.data`).

> Do you know which LVGL API is appropriate for measuring performance?

We usually use the benchmark from `lv_examples` as a rough estimate of performance.

tkashi-github, 2019-09-19:
Hi embeddedt-san.

Thank you for your quick reply!

> We usually use the benchmark from lv_examples as a rough estimate of performance.

OK. I will rework the code and measure the performance. I think it can be on the weekend. 

BR,
T.Kashiwagi

kisvegabor, 2019-09-19:
Probably the bottlenecks in performance are the drawing functions and not the simple API functions. E.g. `lv_obj_set_width` is called rarely, and typically runs only a few hundred lines of code. However, writing/reading/mixing hundreds of thousands of pixels 20 times in sec demands a lot of performance. 

So I think we can carefully pick a few core functions (10-20 I guess) which should be prefixed.

tkashi-github, 2019-09-21:
Hello !

I reworked code andd exectued benchmark.　The performance did not improve much. (20ms -> 19ms is 5% improved)

I attached  bench mark results memo.
[performance_report.pdf](https://github.com/littlevgl/lvgl/files/3638794/performance_report.pdf)

I will examine drawing functions 

BR,
T.Kashiwagi

kisvegabor, 2019-09-21:
It seems adding BG to TCM is more important than adding the program.

To thest the draw functions you should prefix [lv_mem_blend](https://github.com/littlevgl/lvgl/blob/master/src/lv_draw/lv_draw_basic.c#L567), [sw_color_fill](https://github.com/littlevgl/lvgl/blob/master/src/lv_draw/lv_draw_basic.c#L587) and [lv_color_mix](https://github.com/littlevgl/lvgl/blob/master/src/lv_misc/lv_color.h#L326)

tkashi-github, 2019-09-22:
Hi kisvegabor-san

Thank you for your reply!

> It seems adding BG to TCM is more important than adding the program.

That's right. When only the draw function was placed in the TCM, the performance improvement was only about 5%.

> To test the draw functions you should prefix lv_mem_blend, sw_color_fill and lv_color_mix.

Yes. Benchmarking was performed with these functions prefixed without lv_color_mix.
(Is attribute prefix valid for static inline functions?)

For now, it seems better to place image data such as BG in TCM than placing API to TCM.

The code used is 
**IMXRT1060-Projext** https://github.com/tkashi-github/IMXRT1060-EVK/commit/eca2080e4084c4f34cf98d622468dee09cbee431

**lvgl-6.0.2-fork** https://github.com/tkashi-github/lvgl/commit/b58f1696e1e0f5c803c0fd4f2904f22cebacd2ae
.

kisvegabor, 2019-09-22:
> Is attribute prefix valid for static inline functions?

I don't know. :sweat_smile: I can't imagine how it is possible to run an inlined RAM function from  SDRAM. But how knows what kind of magic happens... 

So the conclusion is rather an attribute (e.g. `LV_ATTRIBUTE_IMG_MAP`) should be added to the generated images?

embeddedt, 2019-09-22:
> I can't imagine how it is possible to run an inlined RAM function from SDRAM.

The attribute would have to be on the caller of the inlined function, not the inline function itself. Unless it is the kind of function that is only inlined in some cases.

kisvegabor, 2019-09-22:
> The attribute would have to be on the caller of the inlined function, not the inline function itself. Unless it is the kind of function that is only inlined in some cases.

It sounds logical.

tkashi-github, 2019-09-23:
> So the conclusion is rather an attribute (e.g. LV_ATTRIBUTE_IMG_MAP) should be added to the generated images?

Yes. Perhaps the image data will not fit in the L1 cache. (Conversely, I can guess that the API is in the L1 cache.)

From the results of this experiment, it seems efficient to store only critical image data in the TCM.

And, If attributes like LV_ATTRIBUTE_IMG_MAP are enabled, Cortex-M7 can expect a performance improvement of up to 30%. Do you need this feature?

kisvegabor, 2019-09-23:
What about defines like `LV_ATTRIBUTE_IMG_<NAME>` to individually control image map attribute?

tkashi-github, 2019-09-23:
> What about defines like LV_ATTRIBUTE_IMG_<NAME> to individually control image map attribute?

OK! Please wait a few days.

tkashi-github, 2019-09-26:
I checked the source code of "lvgl" and "lv_example".
"LV_ATTRIBUTE_IMG_ <IMAGE>" could be applied to files in lv_example / lv_apps and lv_example / lv_tutorial / 6_images.

The changes are as follows.
https://github.com/tkashi-github/lv_examples/commit/86767d779277fa0d4f384697ea3d73743a859a73

I don't think there is any file that can apply "LV_ATTRIBUTE_IMG_ <NAME>" to "lvgl / *" so far.

Should I PR to "lv_example"?

embeddedt, 2019-09-26:
> "LV_ATTRIBUTE_IMG_ " could be applied to files in lv_example / lv_apps and lv_example / lv_tutorial / 6_images.

The feature should be added to the [image converter](https://github.com/littlevgl/lv_utils/blob/master/img_conv_core.php) so that all new images will be generated with the attribute. Then the images in those directories can be remade using the new converter.

tkashi-github, 2019-09-26:
> The feature should be added to the image converter so that all new images will be generated with the attribute.

OK. I will check image converter.
I'm not good at php, so I think it will take a while.

kisvegabor, 2019-09-27:
The image converter is quite hacky so I've just added it myself :slightly_smiling_face: 
https://github.com/littlevgl/lv_utils/commit/ab67279c99d8271f90cef13195bfb4f13d201ddb

The online converter is updated as well.

tkashi-github, 2019-09-27:
> The online converter is updated as well.

Thank you!!  This issue will be closed.

closed 2019-09-27 by tkashi-github

kisvegabor, 2019-09-27:
Thank you for the contribution.