lvgl/lvgl #1203: issue when using LV_SYMBOL in lv_imgbtn

joedeveloper, 2019-09-25:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

Can't seem to use LV_SYMBOL derived lv_imgs in lv_imgbtn
It seems that the imgbtn renderer interprets the strings as filesystem paths even when using lv_img as an intermediary step.

**To Reproduce**

	lv_obj_t * img_rear_beep_mute = lv_img_create(actions_container, NULL);
	lv_img_set_src(img_rear_beep_mute, LV_SYMBOL_MUTE "");
	lv_obj_t * img_rear_beep_max = lv_img_create(actions_container, NULL);
		lv_img_set_src(img_rear_beep_max, LV_SYMBOL_DUMMY LV_SYMBOL_VOLUME_MAX "Beep");

	lv_imgbtn_set_src(img_btn_rear_beep, LV_BTN_STATE_REL , LV_SYMBOL_DUMMY LV_SYMBOL_VOLUME_MAX "Beep");
	lv_imgbtn_set_src(img_btn_rear_beep,LV_BTN_STATE_PR , &img_rear_beep_max);
	lv_imgbtn_set_src(img_btn_rear_beep, LV_BTN_STATE_TGL_PR, &img_rear_beep_max);
	lv_imgbtn_set_src(img_btn_rear_beep, LV_BTN_STATE_TGL_REL, &img_rear_beep_mute);
	lv_imgbtn_set_src(img_btn_rear_beep, LV_BTN_STATE_INA, &img_rear_beep_mute);

**Expected behavior**

I was hoping to be able to leverage the state derived image configurations available from lv_imgbtn, but using font symbol derived lv_imgs as source.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Add any other context about the problem here.

embeddedt, 2019-09-25:
Technically, this belongs on the forum. But I'll let it go this time. :smiley:

Did you read this? https://docs.littlevgl.com/en/html/object-types/img.html#image-source

The same information applies to `lv_imgbtn`.

So something like this should work (not tested):

```c
lv_imgbtn_set_src(img_btn_rear_beep, LV_BTN_STATE_PR, LV_SYMBOL_VOLUME_MAX);
```

Passing an image object as the source parameter in `lv_imgbtn_set_src` is **not supported** as far as I know.

kisvegabor, 2019-09-25:
Actually `lv_imgbtn` doesn't support "Symbols" as image source. :slightly_frowning_face:

embeddedt, 2019-09-25:
@kisvegabor We should add that to the documentation. Unless it's already there and I didn't see it?

kisvegabor, 2019-09-25:
It's really not documented now. 
I've updated it: https://github.com/littlevgl/docs/commit/6bf373342f39c5b68e683756c720c4972e7407c0

joedeveloper, 2019-09-26:
Thanks for adding it to the documentation, it did indeed imply that the imgbtn src behavior was the same as img - frankly I wish it was, this would allow for very easily having different Symbol icons for different LV_BTN_STATEs. I humbly suggest adding this in future releases as the boiler plate involved currently for this kind of behavior seems a bit excessive ( not to mention that it seems that it actually has to happen in event handlers and involves lv_set_label_text ).

kisvegabor, 2019-09-27:
I've looked into in and added symbol support in `dev-6.1`.
However, it works only in non-tiled mode (`LV_IMGBTN_TILED 0`)
 
[image]

```c
    lv_obj_t * ib = lv_imgbtn_create(lv_scr_act(), NULL);
    lv_imgbtn_set_src(ib, LV_BTN_STATE_REL, LV_SYMBOL_EYE_OPEN);
    lv_imgbtn_set_src(ib, LV_BTN_STATE_PR, LV_SYMBOL_EYE_CLOSE);
    lv_imgbtn_set_style(ib, LV_BTN_STYLE_REL, &lv_style_plain);
```

I'm not perfectly happy with the API now because once you enable tiled mode you lose the symbol support. But a correct fix would require API change. So it's another story for v7.0.

closed 2019-09-27 by joedeveloper