lvgl/lvgl #1210: Subpixel rendering

kisvegabor, 2019-09-30:
In continuation of the[ related topic](https://forum.littlevgl.com/t/subpixel-rendered-font-bitmaps/742) on the Forum.

cc @puzrin

kisvegabor, 2019-10-01:
@puzrin I've noticed that `box_w` and `adv_w` are 3 times larger. Now I'd like to deal only with rendering and not with subpixel kerning. Do you think it's ok to simply round up the widths? 
E.g. `box_w = 11` :arrow_right: `box_w = 4`?

However, IMO the `ofs_x` should be rounded down.
 E.g. `ofs_x = 4` :arrow_right: `ofs_x = 1`.
What do you think?

puzrin, 2019-10-01:
I'd suggest to keep things separate. Offsets in in glyphs are for floating bitmap. Instead of inventing fractional offsets, it's enouth to shift bitmap's content (and increase bounding box if needed). Bounding box base corner position can be always integer.

When you receive font with subpixel data, the only difference is that bitmap has 3x density/pixels in selected direction. So, nothing to do with existing offset values.

Explanations above are in context of data structures, not depending on implementation. Is this what you asked about?

kisvegabor, 2019-10-02:
> I'd suggest to keep things separate. Offsets in in glyphs are for "floating" bitmaps. Instead of inventing fractional offsets, it's enouth to shift bitmap's content (and increase bounding box if needed).

Still not fully clear. So do you suggest if `ofs_x = 4`, skip one pixel and skip the first color channel?

puzrin, 2019-10-02:
Probably i did not understand your question. I replied about font data.

If you asking about algorythm - i have no experience with those. If you wish to render subpixel font as ordinary one - that's not good idea. 

Proper image downscale needs filtering to keep result sharp. There are tons of reading about such topic in imagemagick docs. I say so as author of https://github.com/nodeca/pica. In short - any simple approach will give bad result, just don't do that.

kisvegabor, 2019-10-03:
Okay, thanks.
I can deal with it next week.

kisvegabor, 2019-10-03:
Just quick hacky experiment:
[image]

[image]

[image]

embeddedt, 2019-10-03:
I note that 2, 3, 5, and 6 appear out of place on the keyboard (i.e. too far to the right).

kisvegabor, 2019-10-03:
> I note that 2, 3, 5, and 6 appear out of place on the keyboard (i.e. too far to the right).

Thank you! Probably I make 1-pixel error in some cases. I'll investigate.

pete-pjb, 2019-10-03:
Also the spacing of abc button looks a little off.... 
If that is helpful...

puzrin, 2019-10-03:
@kisvegabor 

I'm not sure it's correct to check subpixel rendering via screenshot. AFAIK a real photo with high resolution needed, to show how things look in your eyes.

Did result on you LCD improved significantly or not?

embeddedt, 2019-10-03:
@puzrin As far as I know, screenshots captured on Linux simply retrieve the raw values of the logical RGB pixels. Therefore, it shouldn't affect the subpixel rendering.

Here's part of your comment scaled to 400% with no interpolation:

[image]

You'll have to click it to see it at proper size.

kisvegabor, 2019-10-04:
@embeddedt By zooming into the image the result really won't be better. At first, I just wanted to see the red-ish pixels on the left and blue-ish pixels on the right.

kisvegabor, 2019-10-04:
@puzrin 
Here is something I don't understand with `ofs_x` in the generated font. 
Both snippets contain only the "ABC" letters.

Normal:
```
    {.bitmap_index = 0, .adv_w = 0, .box_w = 0, .box_h = 0, .ofs_x = 0, .ofs_y = 0} /* id = 0 reserved */,
    {.bitmap_index = 0, .adv_w = 160, .box_w = 11, .box_h = 12, .ofs_x = 0, .ofs_y = 0},   //A
    {.bitmap_index = 43, .adv_w = 160, .box_w = 9, .box_h = 12, .ofs_x = 1, .ofs_y = 0},   //B 
    {.bitmap_index = 77, .adv_w = 160, .box_w = 10, .box_h = 12, .ofs_x = 0, .ofs_y = 0}   //C
```

with LCD flag:
```
    {.bitmap_index = 0, .adv_w = 0, .box_w = 0, .box_h = 0, .ofs_x = 0, .ofs_y = 0} /* id = 0 reserved */,
    {.bitmap_index = 0, .adv_w = 160, .box_w = 33, .box_h = 12, .ofs_x = 1, .ofs_y = 0},
    {.bitmap_index = 106, .adv_w = 160, .box_w = 26, .box_h = 12, .ofs_x = 1, .ofs_y = 0},
    {.bitmap_index = 188, .adv_w = 160, .box_w = 29, .box_h = 12, .ofs_x = 1, .ofs_y = 0}
```

With LCD flag all letters have `ofs_x = 1`. I don't know whether `ofs_x` has 1 px (I guess it is) or 1/3 px resolution but the difference seems odd.

If I manually change the `ofs_x` values the letter spaces are correct.
Original:
[image]

Manually fixed:
[image]

Do you have any idea?

puzrin, 2019-10-04:
[Here](https://github.com/littlevgl/lv_font_conv/blob/master/lib/freetype/index.js#L167-L242) is main draw code.

We just extract bitmap & coords without change, those are integer. May be in subpixel mode we should care about `lsb_delta` & `rsb_delta`? I have no idea and would like to avoid diving into C api of FT. If you say exactly what should be done - that will be updated.

puzrin, 2019-10-04:
May be `ofs_NN` of subpixel dimention is in `1/3` of pixels too?

kisvegabor, 2019-10-07:
I have no idea. It might have 1/3 or 1 resolution but IMO it shouldn't be the same for the 3 gylphs. I've [asked](https://forum.littlevgl.com/t/subpixel-rendered-font-bitmaps/742/25?u=kisvegabor) *Jon_Smirl* from the Forum to comment on this.

puzrin, 2019-10-07:
@kisvegabor please use version from master:

- There was a bug with missed X sign.
- AdvanceWidth missed fractional part (not fatal, but kerning could be affected).
- We added a lot of FreeType glyph properties into dump. That should help to check if something goes wrong or incorrect property used.

I hope, that should fix all issues you asked about. X seems to be in pixels.

kisvegabor, 2019-10-09:
@puzrin 
Thank you for the update! It seems it has solved the issue!
[image]

kisvegabor, 2019-10-10:
I've pushed the code to `feat-subpixel`. Now I set the type of font manually in the [draw function](https://github.com/littlevgl/lvgl/blob/7f7abe9a9336aa681d8720e9db96d5b6654e6eb5/src/lv_draw/lv_draw_basic.c#L304).

@puzrin Can you add a flag to [lv_font_t](https://github.com/littlevgl/lvgl/blob/68c752ab5097433bba30fd08723f0d2fae288f1f/src/lv_font/lv_font_roboto_12.c#L1800)? E.g. `subpixel = 1`.

For backward compatibility, it might be useful to add this option only if it's `1`. (Fields are initialized to 0 by default)

puzrin, 2019-10-10:
> Can you add a flag to [lv_font_t](https://github.com/littlevgl/lvgl/blob/68c752ab5097433bba30fd08723f0d2fae288f1f/src/lv_font/lv_font_roboto_12.c#L1800)? E.g. `subpixel = 1`.

What about vertical orientation? There are 2 options, `--lcd` & `--lcd-v`.

kisvegabor, 2019-10-11:
Do you insists on `--lcd-v`? I'd be happy with only `--lcd` too.

embeddedt, 2019-10-11:
@kisvegabor I think `--lcd-v` is necessary, because, depending on the orientation of the display, there might be a need to have vertical subpixel rendering.

kisvegabor, 2019-10-11:
Correct me if I'm wrong but as I see there are two options:
1. "RGB" in lines: we can have subpixel rendering horizontally
2. "RGB" in columns: we can have subpixel rendering vertically (IMO not really important and more complicated because lines should be buffed)

So we cannot do horizontal sub-pixel rendering on vertical RGB display with `--lcd-v`.

puzrin, 2019-10-11:
> Do you insists on `--lcd-v`? I'd be happy with only `--lcd` too.

I do not insist on implementation. But i would insist on reserving value in data design. Since it's technically possible to have at least 2 types of subpixels layout (+ none), this should be expressed in lvgl writer somehow.

For bin writer i planned store this info in "compression" byte, but now tend to add independent header field (this change will be breaking).

embeddedt, 2019-10-11:
> So we cannot do horizontal sub-pixel rendering on vertical RGB display with `--lcd-v`.

Right. It would have to be vertical sub-pixel rendering.

puzrin, 2019-10-12:
Landed non-breaking changes to 0.3.0 release of lv_convert, and pushed the rest to dev branch https://github.com/littlevgl/lv_font_conv/commits/dev

@kisvegabor, i need your decision about `subpixel` field size, to update lvgl writer. I'd suggest make it 2 bits wide at least.

jonsmirl, 2019-10-13:
Don't confuse portrait and landscape mode LCDs, both of those want horizontal subpixel rendering. The only time you would use vertical subpixels is if you mounted the LCD wrong or if your display rotates. Vertical subpixel rendering has little visible impact.  Check your display with a microscope and make sure the color bars are running horizontally, if they are you have it mounted correctly.

jonsmirl, 2019-10-13:
The current code is looking great. It should be sufficient for most users. As far as I know this is the first implementation of subpixel rendering for microcontrollers.

If you want to improve it more -- play around with higher order filtering functions. These help but the improvements are not huge.

Implement subpixel spacing and kerning. This doesn't improve legibility but it lets you pack more characters onto a tiny display without loss of legibility.

jonsmirl, 2019-10-13:
I also recommend asking for help on the Freetype list. I just dabble in this stuff, the people on the Freetype list have PhD's in it.  For example the Freetype PhD's will tell you to set the gamma of your LCD panel into the filter function. The gamma is constant and it should be on the LCD datasheet. If the filter function knows the gamma it can do a much better filter.  When you see halos around the letters on the desktop, you probably have the gamma of the monitor set wrong in the Xserver. That problem is disappearing now with monitor EDIDs that contain the gamma, but embedded systems don't have EDIDs. Your code is at 90% performance, these are just details to take it up the highest possible levels of legibility, they are not required.

kisvegabor, 2019-10-16:
@puzrin 
Let's use `.subpx = LV_FONT_SUBPX_NONE/HOR/VER/BOTH`. 
See the commit: 10b3463a

@jonsmirl 
Thank you for the hints!

puzrin, 2019-10-16:
@kisvegabor please confirm final variants:

1. Nothing for no subpixels (rely on zero init by default)
2. Add `subpixel = LV_FONT_SUBPX_HOR` for `--lcd`
3. Add `subpixel = LV_FONT_SUBPX_VER` for `--lcd-v`

Correct?

puzrin, 2019-10-16:
@kisvegabor https://github.com/littlevgl/lv_font_conv/commit/f95909bb67336b4c5f20c8caaeec838f898eaf63

Please check.

kisvegabor, 2019-10-18:
@puzrin Works great using the `dev` branch of the converter. I updated `lvgl` and merged `feat-subpx` into `dev-6.1`.

As the feature is added I close this issue. Thank you very much for your contribution. :+1:

closed 2019-10-18 by kisvegabor

puzrin, 2019-10-18:
@kisvegabor i had no chance to check subpixel rendering on device with LCD. What is your opinion? Did quality really improved, at least for black on white and white on black? What about perfomance? 3x slower?

kisvegabor, 2019-10-19:
I've just tested it on STM32F429 Discovery. Here are some photos: 
[subpx.zip](https://github.com/littlevgl/lvgl/files/3746391/subpx.zip)

They have very high resolution so if you zoom in you can see the RGB channels.

To my eyes, the subpixel rendered font looked a little bit better but there wasn't a huge difference. However this display has a relatively large DPI (2.4", 320x240), so probably it's not a good target to demonstrate the advantages of subpixel rendering.

Regarding the speed, the UI shown in the images:
- normal rendering: 36 ms
- subpixel rendering: 56 ms

puzrin, 2019-10-19:
Hm. It's strange, but difference is really not as big as expected. May be, difference should be more visible with  12 px font?

Hight DPI is 400+, your display is far from it. Should be good candidate to demonstrate subpixel rendering.

kisvegabor, 2019-10-19:
Here are some more photos  with 10px and 6px fonts. Still can't see a huge difference
[sub_px_10.zip](https://github.com/littlevgl/lvgl/files/3746427/sub_px_10.zip)
[sub_px_6.zip](https://github.com/littlevgl/lvgl/files/3746429/sub_px_6.zip)

10px was readable in all cases.
6px was not readable in any of the cases.

By high DPI I meant compared to other displays in this class. I saw a lot of 320x240 display in 3.5" size (114 DPI) where you can easily see the pixels. On this display, it's very hard to spot the pixels.

puzrin, 2019-10-19:
Hm... too many halos. Probably, better display gamma correction heeded on alpha blend.

Could you do last check? 10px negative - white letters on black background. May be that will hide some defects.

kisvegabor, 2019-10-19:
Here it is. 
[black_bg.zip](https://github.com/littlevgl/lvgl/files/3746715/black_bg.zip)

Subpx looks better in some cases, especially where there is no space between letters. E.g. "sed" in the second line.

However in general -at least on this screen - not that much better.

jonsmirl, 2019-10-19:
It is likely worthwhile to discuss this on the freetype list. Those guys do this stuff for their jobs and they are experts at it.  My guess is that the filter function on the subpixel display is not working quite right. The color halos on 'N' should not be visible.

There is a way to check your code. Have freetype generate fully subpixel rendered bitmaps of the same text. Then compare your pixel vales to theirs for random letters. That will show you where things aren't quite right.

The general algorithm appears to be working, you may simply have a bug somewhere. And where there is a bug the whole effect doesn't work.

puzrin, 2019-10-19:
@kisvegabor negative screen seems optimal for experiments. I remembered one more check. If you properly count fractional width (cursor position shift) and kerning - try `--autohint-strong`. If it will not break everything again, difference should be notable.

kisvegabor, 2019-10-22:
@puzrin I can do it tomorrow.

puzrin, 2019-10-22:
As far as i understand, with `--autohint-strong` there may be ~ 1 subpixel positioning error (1/3 px), because we do not process [lsb_delta](https://www.freetype.org/freetype2/docs/reference/ft2-base_interface.html#ft_glyphslotrec). Vusually, that may look as a bit broken kerning, but not as horrible as before subpixel mode (1 px error looks very bad).

If you find result worth to use, that error can be compensated almost to zero in convertor (instead of `lsb_delta` we can shift bitmap with 1/3 px precision)

jonsmirl, 2019-10-22:
I just checked out the black background sample.  I zoomed in on the letter 'm' and I see a red band in front of the 'm'.  That can't be right. I should see a blue band, not a red one.  This display appears to look like this:  RxxRGB it should xxBRGB. in the photo.

What's the easiest was to build a sample program with this enabled? I have a USB microscope and I can take pictures at 50x magnification. For hardware I have ESP32 with ILI9341 display. Or Unix with framebuffer.

C app is fine. I using the micropython build for my main work which is why I have not closely tracked this.

I took picture of current code which only dims full pixels and annotated it with approximate glyph outline. You can see that many sub-pixel are turned on that should not be on. I can take same picture using the new code.  This is from ILI9341 on ESP32. [image]
When looking at this not that the subpixels are being use in RGB groups and all subpixels have same brightness in each group.  Background is dark blue so it is on a little. It the button from the 'Hello World' example.

kisvegabor, 2019-10-23:
@puzrin 
Here are the autohinted images: 
[autohint.zip](https://github.com/littlevgl/lvgl/files/3760647/autohint.zip)

@jonsmirl 
Just set [LV_SUBPX_BGR](https://github.com/littlevgl/lvgl/blob/dev-6.1/lv_conf_template.h#L368) to 1.

jonsmirl, 2019-10-24:
It worn't compile on ESP32

CC build/lvgl/src/lv_draw/lv_draw_basic.o
/home/jonsmirl/aosp/esp32_ili9341/components/lvgl/src/lv_draw/lv_draw_basic.c: In function 'lv_draw_letter':
/home/jonsmirl/aosp/esp32_ili9341/components/lvgl/src/lv_draw/lv_draw_basic.c:420:37: error: 'struct <anonymous>' has no member named 'green'
                         res_color.ch.green = (uint16_t)((uint16_t)txt_rgb[1] * font_rgb[1] + (bg_rgb[1] * (255 - font_rgb[1]))) >> 8;
                                     ^

puzrin, 2019-10-24:
> Here are the autohinted images:

I don't see expected difference. No ideas what to do. Anyway, kick me if need more convertor updates to support subpixels.

jonsmirl, 2019-10-24:
I need to get this working and use the microscope on it. Once you see everything with microscope the bugs will be obvious. Right now I don't think it is working right due to those color halos in the photos.

kisvegabor, 2019-10-24:
> It worn't compile on ESP32

@jonsmirl I've fixed it in `dev-6.1`.

jonsmirl, 2019-10-24:
This look almost right. Something not quite right with background blend math.
[image]

See the areas I circled, those pixels should be on.

kisvegabor, 2019-10-24:
@jonsmirl 
The algorithm works like this:
```
 blue = (txt_blue * font_blue_opa + bg_blue * (255 - font_blue_opa)) >> 8;
```

Does it look correct?

It's a little bit hard to decide but it seems on your images only blue channels are missing. Am I see it right?

jonsmirl, 2019-10-24:
Can you print out the subpixel values you compute for a white letter 'W' blended with a black background? They should match the input 3x expanded bitmap.

jonsmirl, 2019-10-24:
This is white 'L' on black background. I don't understand why it is six subpixels at half brightness.
[image]

puzrin, 2019-10-24:
@jonsmirl if your question is about convertor:

- use code from `dev` branch
- add `--lcd` key

jonsmirl, 2019-10-24:
Isn't the 3X expanded font checked into the 6.1-dev branch? It is not obvious to me if it is.

puzrin, 2019-10-24:
Those looks old (NOT 3x).

See https://github.com/littlevgl/lv_font_conv/blob/dev/lib/writers/lvgl/lv_table_head.js#L73. There should be line `.subpx = LV_FONT_SUBPX_HOR`

jonsmirl, 2019-10-24:
That explains why it is not working.

jonsmirl, 2019-10-24:
We probably need smarter font logic. Always compile in the subpixel code, it is tiny.  When displaying look at the header of each font. If the font is subpixel enabled, then use it with the subpixel code. Otherwise render normally.

embeddedt, 2019-10-24:
@jonsmirl I think you need `dev-6.1` from https://github.com/littlevgl/lvgl and the `dev` branch from https://github.com/littlevgl/lv_font_conv.

puzrin, 2019-10-24:
> I'll see if I can figure out how to add them to my build.

If you copy-pasted lib folder, you can replace font files with new ones. Dirty but simple.

[30 more replies]