lvgl/lvgl #1212: error: narrowing conversion of '-2' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

Chronic788, 2019-10-01:
**Describe the bug**

Fonts that are generated using the Lvgl Font Converter tool that have negative numbers in the ofs_y field in the lv_font_fmt_txt_glyph_dsc_t struct located in lv_font_fmt_txt.h throw this error:

```
error: narrowing conversion of '[SOME NEGATIVE NUMBER]' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
```

**To Reproduce**

Generate a font using the LittlevGL font converter tool that has negative numbers in the ofs_y field in the  lv_font_fmt_txt_glyph_dsc_t struct in the generated file. In order to reproduce this accurately, one may generate any font and then change one of the values in the ofs_y field to a negative integer.

**Expected behavior**

The font can be integrated into a project without the described error.

**Screenshots**
No screenshots available.

**Additional context**
No additional context.

**The fix**

I have determined the fix for this issue but cannot submit a pull request for legal reasons:

The error lies in lv_font_fmt_txt.h on line 48.

Here is the affected struct:

```
/** This describes a glyph. */
typedef struct
{
#if LV_FONT_FMT_TXT_LARGE == 0
    uint32_t bitmap_index : 20;     /**< Start index of the bitmap. A font can be max 1 MB. */
    uint32_t adv_w :12;             /**< Draw the next glyph after this width. 8.4 format (real_value * 16 is stored). */
#else
    uint32_t bitmap_index;          /**< Start index of the bitmap. A font can be max 4 GB. */
    uint32_t adv_w;                 /**< Draw the next glyph after this width. 28.4 format (real_value * 16 is stored). */
#endif
    uint8_t box_w;                  /**< Width of the glyph's bounding box*/
    uint8_t box_h;                  /**< Height of the glyph's bounding box*/
    int8_t ofs_x;                   /**< x offset of the bounding box*/
    uint8_t ofs_y;                  /**< y offset of the bounding box. Measured from the top of the line*/
}lv_font_fmt_txt_glyph_dsc_t;
```

The variable type for ofs_y should be changed to int8_t from uint8_t like so:

```
/** This describes a glyph. */
typedef struct
{
#if LV_FONT_FMT_TXT_LARGE == 0
    uint32_t bitmap_index : 20;     /**< Start index of the bitmap. A font can be max 1 MB. */
    uint32_t adv_w :12;             /**< Draw the next glyph after this width. 8.4 format (real_value * 16 is stored). */
#else
    uint32_t bitmap_index;          /**< Start index of the bitmap. A font can be max 4 GB. */
    uint32_t adv_w;                 /**< Draw the next glyph after this width. 28.4 format (real_value * 16 is stored). */
#endif
    uint8_t box_w;                  /**< Width of the glyph's bounding box*/
    uint8_t box_h;                  /**< Height of the glyph's bounding box*/
    int8_t ofs_x;                   /**< x offset of the bounding box*/
    int8_t ofs_y;                  /**< y offset of the bounding box. Measured from the top of the line*/
}lv_font_fmt_txt_glyph_dsc_t;
```

embeddedt, 2019-10-02:
Is the bounding box offset supposed to be negative to begin with? Because adjusting it to be signed would cut the range in half.

puzrin, 2019-10-02:
Ups, i reported the same in #1213

Yes, negative offcets mean glyph shold have left shift from current cursor position. For example, if that's combined character.

If you doubt about anything in font - use font spec for reference https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md#table-head-font-header

> [1]  [Glyph BBox x/y bits length (signed value)]

kisvegabor, 2019-10-03:
Let's continue it in https://github.com/littlevgl/lvgl/issues/1213

closed 2019-10-03 by kisvegabor