lvgl/lvgl #1213: Warnings about sign conversion in font files

puzrin, 2019-10-02:
I see huge number of warnings with `-Wconversion` build flag:

```
src/fonts/my_font_roboto_12.c:648:87: error: negative integer implicitly converted to unsigned type [-Werror=sign-conversion]
     {.bitmap_index = 1670, .adv_w = 48, .box_w = 4, .box_h = 13, .ofs_x = 0, .ofs_y = -2},
                                                                                       ^

src/fonts/my_font_roboto_12.c:761:9: error: negative integer implicitly converted to unsigned type [-Werror=sign-conversion]
     -5, -6, 0, -2, -6, 0, -7, 0,
         ^
```

That's in `kern_class_values` & `glyph_dsc` definitions

embeddedt, 2019-10-02:
As long as it's safe to reduce the maximum range of these values (i.e. truncate `0-255` to `-128-127`) I don't have any objections to switching these members to signed variables.

puzrin, 2019-10-02:
I would also suggest to check code with all extra warnings enabled. On your choice. May be you find something else. Indirect problem is that lvgl poison output when i try to check my own code in PIO with enabled warnings. I can't say "build only my own src, without libs"

embeddedt, 2019-10-02:
> I would also suggest to check code with all extra warnings enabled. On your choice. May be you find something else. 

I do normally compile LittlevGL with a lot of warnings (probably more than most people) but for some reason I never get this error. Perhaps I have `-Wconversion` disabled for some reason.

puzrin, 2019-10-02:
Hm... here is my desired list https://github.com/speedcontrols/ac_sc_grinder/blob/master/platformio.ini#L29-L44. I'm not sure in correcnness, but may be you find something useful.

puzrin, 2019-10-02:
I see also a lot of warnings in image decoder with `-Wunused-parameter`.

kisvegabor, 2019-10-02:
@puzrin 

That's correct. Both `kern_class_values` and `glyph_dsc` should be updated. I'm not sure where to push the fix. In `master` (as @embeddedt  [pointed out](https://github.com/littlevgl/lvgl/issues/1213#issuecomment-537471009))  it can limit the range in existing projects.

puzrin, 2019-10-02:
I can switch to 6.1, if you can push fixes there. Also, please fix warnings in image decoder.

puzrin, 2019-10-02:
FYI https://github.com/puzrin/dispenser/blob/dev/.travis.yml - here is example of my travis config to run tests via PIO. It really worth add automated testing for lvgl to avoid such boring regressions.

embeddedt, 2019-10-02:
> It really worth add automated testing for lvgl to avoid such boring regressions.

It's on the roadmap, but no one has gotten around to it yet. :slightly_smiling_face: Contributions to add that are welcome.

embeddedt, 2019-10-03:
I checked my compiler flags - it appears that I disabled `-Wconversion` because it had too many false positives elsewhere in my codebase. I will enable it specifically for LittlevGL only and see what I find.

embeddedt, 2019-10-03:
`-Wconversion` is a bit too strict (IMO) but I've enabled `-Wsign-conversion` and `-Wunused-parameter`. I'm working on eliminating them. I have most of the other warnings in your list enabled already.

embeddedt, 2019-10-03:
@kisvegabor Should we fix things like this (`dev-6.1` HEAD)?

```c
lvgl/src/lv_draw/lv_img_decoder.c:677:17: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
             ofs += w * y + (x >> 3); /*First pixel*/
                 ^~
lvgl/src/lv_draw/lv_img_decoder.c:684:17: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
             ofs += w * y + (x >> 2); /*First pixel*/
                 ^~
lvgl/src/lv_draw/lv_img_decoder.c:691:17: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
             ofs += w * y + (x >> 1); /*First pixel*/
                 ^~
lvgl/src/lv_draw/lv_img_decoder.c:697:17: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
             ofs += w * y + x;  /*First pixel*/
                 ^~
lvgl/src/lv_draw/lv_img_decoder.c:715:42: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'lv_coord_t' {aka 'short int'} may change the sign of the result [-Werror=sign-conversion]
         lv_fs_read(user_data->f, fs_buf, w, NULL);
                                          ^~
```

I think it's complaining because of the fact that we use unsigned and signed integers a bit too interchangeably, but to be honest, I'm not really sure that this is a major issue, considering that I don't see anyone wanting an image 2 billion pixels wide.

I could go through and adjust all the types but it would be quite a bit of work, could possibly introduce unseen bugs, and may not be worth our time.

The font warning was a good catch though. What do you think?

puzrin, 2019-10-03:
https://docs.platformio.org/en/latest/projectconf/section_env_build.html#src-build-flags

Found option `src_build_flags` and moved warnings there. So, the only really critical warnings left are from custom fonts files. Everything else - would be nice to reduce, but right now not critical.

kisvegabor, 2019-10-03:
@puzrin I've fixed the issue in the fonts here: https://github.com/littlevgl/lvgl/commit/8fb484ab913ac98513a47c3327cfac57a71ae6bf

And also removed unused vars (There were only 3 in `lv_chart`)

I've tested  `Wsign-conversion` Really there are lot of warnings. Some of them are reasonable though. @embeddedt I suggest fixing the straight forward ones first and we see how many left.

puzrin, 2019-10-03:
@kisvegabor now much better, but still see 1 warning per font:

```
Compiling .pio/build/emulator/src/fonts/my_font_roboto_14.o
src/fonts/my_font_roboto_12.c:935:28: warning: pointer targets in initialization differ in signedness [-Wpointer-sign]
     .class_pair_values   = kern_class_values,
                            ^~~~~~~~~~~~~~~~~
src/fonts/my_font_roboto_12.c:935:28: note: (near initialization for ‘kern_classes.class_pair_values’)
```

puzrin, 2019-10-03:
Hm... type of value in kerning tables generated by convertor:

```
/*Kern values between classes*/
static const uint8_t kern_class_values[] =
{
...
```

How do you recommend to update that without creating problems for 6.0 users?

puzrin, 2019-10-03:
Also, places like this causes a lot of warnings with `-Wconvert`:

https://github.com/littlevgl/lvgl/blob/abb480f98864d9c1cf9c269032f4d10e257da5c8/src/lv_misc/lv_color.h#L232-L234

Problem is that's a header file. Even if i use extended warning for my code only, headers from LVGL are included into my compiled .c files and cause problem. IMO this should be fixed.

kisvegabor, 2019-10-04:
> Hm... type of value in kerning tables generated by convertor. That's from font file:

As the font converter doesn't have a `dev-6.1` branch to be compatible with lvgl's current development stage, IMO it'd be best to fix it in the `dev` branch. 

> Problem is that's a header file. Even if i use extended warning for my code only, headers from LVGL are included into my compiled .c files and cause problem. IMO this should be fixed.

That would be really great the start with the header files. I've found the same for `lv_area_get_width/height` (static inline functions).

Interestingly it produces a warning: 
```
return (lv_coord_t) area_p->x2 - area_p->x1 + 1;
```

And it doesn't:
```
return (lv_coord_t) (area_p->x2 - area_p->x1 + 1);
```

However, it doesn't work here:
```
lvgl/src/lv_hal/../lv_misc/lv_color.h:234:20: warning: conversion to ‘unsigned char:2’ from ‘unsigned char’ may alter its value [-Wconversion]
     ret.ch.blue  = (uint8_t) (color.ch.blue >> 3);  /* 5 - 2  = 3*/
```

@embeddedt Do you have any idea how to easily mitigate these warnings?

puzrin, 2019-10-04:
Try to add bitmask for used bits `(color.ch.blue >> 3) & 0x03`. As far as i understand common principle, warning disappear if you "comfirm manually" that you really wish to do something unusial.

embeddedt, 2019-10-04:
@kisvegabor Usually GCC points to a certain spot on the line when it prints out that warning. Do you know where it pointed to?

kisvegabor, 2019-10-07:
@embeddedt 
Here are some "strange-to-me" warnings:

```
lvgl/src/lv_hal/../lv_misc/lv_area.h: In function ‘lv_area_get_width’:
lvgl/src/lv_hal/../lv_misc/lv_area.h:85:36: warning: conversion to ‘lv_coord_t {aka short int}’ from ‘int’ may alter its value [-Wconversion]
     return area_p->x2 - area_p->x1 + 1;
            ~~~~~~~~~~~~~~~~~~~~~~~~^~~
```

```
lvgl/src/lv_hal/../lv_misc/lv_color.h:363:23: warning: conversion to ‘uint16_t {aka short unsigned int}’ from ‘int’ may alter its value [-Wconversion]
     uint16_t bright = 3 * c32.ch.red + c32.ch.blue + 4 * c32.ch.green;
                       ^
```

```
lvgl/src/lv_hal/../lv_misc/lv_color.h:364:29: warning: conversion to ‘uint8_t {aka unsigned char}’ from ‘int’ may alter its value [-Wconversion]
     return (uint16_t)bright >> 3;
            ~~~~~~~~~~~~~~~~~^~~~
```

puzrin, 2019-10-07:
> As the font converter doesn't have a `dev-6.1` branch to be compatible with lvgl's current development stage, IMO it'd be best to fix it in the `dev` branch.

Could you explain why those type changes can be a problem for 6.0? Could not find appropriate code. A var as i remeber, you used `(void *)`, it should not ve affected.

puzrin, 2019-10-07:
> Here are some "strange-to-me" warnings:
>
> ```
> lvgl/src/lv_hal/../lv_misc/lv_area.h: In function ‘lv_area_get_width’:
> lvgl/src/lv_hal/../lv_misc/lv_area.h:85:36: warning: conversion to ‘lv_coord_t [...]

Constants numbers are `int` by default => int result => warning. Try 1U or `(uint16_t)1`
 
> ```
> lvgl/src/lv_hal/../lv_misc/lv_color.h:363:23: warning: conversion to ‘uint16_t {aka short unsigned int}’ from ‘int’ may alter its value [-Wconversion]
>      uint16_t bright = 3 * c32.ch.red + c32.ch.blue + 4 * c32.ch.green;
>                        ^
> ```

The same

> ```
> lvgl/src/lv_hal/../lv_misc/lv_color.h:364:29: warning: conversion to ‘uint8_t {aka unsigned char}’ from ‘int’ may alter its value [-Wconversion]
>      return (uint16_t)bright >> 3;
>             ~~~~~~~~~~~~~~~~~^~~~
> ```

uint16_t >> 3 => still uint16_t

Try `return (uint16_t)(bright >> 3);`

kisvegabor, 2019-10-09:
I've also tried `+ 1U` but almost the same:
```
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_sim_master/lv_examples/lv_tests/lv_test_objx/lv_test_sw/../../../../lvgl/src/lv_hal/../lv_misc/lv_area.h:88:36: warning: conversion to ‘lv_coord_t {aka short int}’ from ‘unsigned int’ may alter its value [-Wconversion]
     return area_p->x2 - area_p->x1 + 1U;
            ~~~~~~~~~~~~~~~~~~~~~~~~^~~~
```

So it needs to be explicitly casted to `lv_coord_t` (int16_t by default)

> Could you explain why those type changes can be a problem for 6.0? Could not find appropriate code. A var as i remeber, you used (void *), it should not ve affected.

It has `const uint8_t * class_pair_values;` See [here](https://github.com/littlevgl/lvgl/blob/master/src/lv_font/lv_font_fmt_txt.h#L144).

puzrin, 2019-10-09:
> It has `const uint8_t * class_pair_values;` See [here](https://github.com/littlevgl/lvgl/blob/master/src/lv_font/lv_font_fmt_txt.h#L144)

But why this should be a problem? Looks like one more warning about sign conversion, it should not cause data loss or invalid work (if worked before).

kisvegabor, 2019-10-09:
It's a different warning (`-Wpointer-sign` which is part of `-Wall`)
```
../lvgl/src/lv_font/lv_font_roboto_16.c:4959:28: warning: pointer targets in initialization differ in signedness [-Wpointer-sign]
     .class_pair_values   = kern_class_values,
                            ^~~~~~~~~~~~~~~~~
```

puzrin, 2019-10-09:
Thank's for explanation. `-Wall` warnings are not nice at all, you are right.

May be you could consider land 6.1 (or 6.0.x+) "as is" with all small things done and and move pending tails for 6.2?  I'm not familiar with requirements of embedded devs. In JS world periodic releases are more preferable.

kisvegabor, 2019-10-11:
We have enough new things to release v6.1 but I still want to wait a few weeks with it.

kisvegabor, 2019-11-02:
Let's warm up this topic. 
Do you think it's really worth to bother with these warnings about signs? I think they are harmless and it'd be ugly to fix them.

embeddedt, 2019-11-02:
> Do you think it's really worth to bother with these warnings about signs? I think they are harmless and it'd be ugly to fix them.
What do you think?

It needs to be casted at the very minimum, otherwise it just pollutes build output.

puzrin, 2019-11-03:
> Do you think it's really worth to bother with these warnings about signs? I think they are harmless and it'd be ugly to fix them.

One of ways to check potential errors is to enable as much warnings as possible. When foreign lib pollute output, that's very inconvenient. IMO minimal requirement is fixing warnings in header files.

cpegeo, 2019-11-04:
I agree with @embeddedt that we need to cast at a minimum. Many development teams have policies that don't allow code that generates warnings. 

We probably want to turn on -Wall and see where our warnings are and resolve them. Most of them seem to occur because of mismatched types (signed/unsigned) which can usually be fixed easily by changing the types. For example, I have found that changing the 'kern_class_values' to int8_t to match the '.class_pair_values' field eliminates the majority of warnings that I had. If matching types is somehow not possible then cast or typedef a new type.

kisvegabor, 2019-11-04:
Okay, let's agree on the warnings flags we want to "support". Now there no warnings with `-Wall -Wshadow -Wundef -Wmaybe-uninitialized` on the `dev-6.1` branch.

embeddedt, 2019-11-04:
```
-Wmissing-prototypes -Wno-discarded-qualifiers -Wall -Wextra -Wno-unused-function -Wundef -Wno-error=strict-prototypes -Wpointer-arith -fno-strict-aliasing -Wno-error=cpp -Wuninitialized -Wmaybe-uninitialized -Wno-unused-parameter -Wno-missing-field-initializers -Wtype-limits -Wsizeof-pointer-memaccess -Wno-format-nonliteral -Wno-cast-qual -Wunreachable-code -Wno-switch-default -Wno-switch-enum -Wreturn-type -Wmultichar -Wformat-security -Wno-ignored-qualifiers -Wno-error=pedantic -Wno-sign-compare -Wno-error=missing-prototypes -Wno-cast-function-type -Wdouble-promotion -Wclobbered -Wdeprecated -Wempty-body -Wtype-limits -Wshift-negative-value -Wstack-usage=900 -Wno-unused-value -Wno-unused-parameter -Wno-missing-field-initializers -Wuninitialized -Wmaybe-uninitialized -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wtype-limits -Wsizeof-pointer-memaccess -Wno-format-nonliteral -Wpointer-arith -Wno-cast-qual -Wmissing-prototypes -Wunreachable-code -Wno-switch-default -Wswitch-enum -Wreturn-type -Wmultichar -Wno-discarded-qualifiers -Wformat-security -Wno-ignored-qualifiers -Wno-sign-compare
```

This is the warning set I compile LittlevGL with. Some of it might be overkill, but this set (and the stronger variant I use elsewhere in the project) does a pretty good job of eliminating simple bugs.

kisvegabor, 2019-11-04:
@embeddedt I've tested `dev-7.0` with your flags and found a [bug](https://github.com/littlevgl/lvgl/commit/e45d08343664e117d7c3e3758f9575e7817d446b#diff-6c14ef100b87dc4aaacb5056073f96cc) which would have been very difficult to find.

Now there are no warnings with these flags.

embeddedt, 2019-11-04:
> found a bug which would have been very difficult to find.

Wow. That was a really sneaky one. :wink:

> Now there are no warnings with these flags.

Great!

puzrin, 2019-11-04:
What about `-Wconversion`?

kisvegabor, 2019-11-05:
> What about -Wconversion?

So that's the question. Shall we really "support" it? 
I don't remember that we ever had issues related to sign/truncation.

puzrin, 2019-11-05:
I'd like to see no warnings in headers with `-Wconversion`. IMO that can show more potential problems in user code.

kisvegabor, 2019-11-08:
My concern about `-Wconversation` is it complains about implicit truncation too. It'd be very ugly to fix and even might cause a performance drop.

`lv_color_mix` is probably the most performance-critical function in the whole library. Here the bitfields of color channels (r,g,b) are written. To avoid `-Wconversation` warnings we should add `& 0x1F` (or so) to the channels. It might be slower than implicit truncation.

However, `Wconversion-sign` might be a reasonable update though because it requires only some casting.

embeddedt, 2019-11-08:
> It might be slower than implicit truncation.

I don't have any evidence to back it up, but I would think that explicit truncation and implicit truncation would have the same performance.

puzrin, 2019-11-08:
I would try to remove as much warning as possible, until this can be done without side effects. Because that's a library, and it should not restrict user's code.

The only way to know about performance - write a benchmark test.

If you don't like direct fixes, for some reasons, may be alternatives possible? For example:

- move this to "inline" attributed functions (in .c file), and leave declaration only in header.
- keep this in separate .h file, not included into lvgl.h, and included directly only when needed.

kisvegabor, 2019-11-08:
@embeddedt 
> I don't have any evidence to back it up, but I would think that explicit truncation and implicit truncation would have the same performance.

I also can't say anything for sure but I think it depends on the compiler and optimization settings.
I'd be afraid to add any extra operation to such a performance-critical function.  

@puzrin 
> The only way to know about performance - write a benchmark test.

A benchmark test would show the difference in given "hardware + compiler + settings". 
However (I'm speaking against myself) I've tested it on STM32F429 Discovery and the demo app with -O0˙ and `-O3`. I've measured exactly the same performance.

The disassembly was the same too (`-O0`):
```
No mask

080145ba:   push    {r7}
080145bc:   sub     sp, #20
080145be:   add     r7, sp, #0
080145c0:   strh    r0, [r7, #4]
080145c2:   strh    r1, [r7, #0]
080145c4:   mov     r3, r2
080145c6:   strb    r3, [r7, #3]
365           ret.ch.red = (uint16_t)((uint16_t)c1.ch.red * mix + (c2.ch.red * (256 - mix))) >> 8;
080145c8:   ldrb    r3, [r7, #5]
080145ca:   ubfx    r3, r3, #3, #5
080145ce:   uxtb    r3, r3
080145d0:   uxth    r2, r3
080145d2:   ldrb    r3, [r7, #3]
080145d4:   uxth    r3, r3
080145d6:   smulbb  r3, r2, r3
080145da:   uxth    r2, r3
080145dc:   ldrb    r3, [r7, #1]
080145de:   ubfx    r3, r3, #3, #5
080145e2:   uxtb    r3, r3
080145e4:   uxth    r1, r3
080145e6:   ldrb    r3, [r7, #3]
080145e8:   rsb     r3, r3, #256    ; 0x100
080145ec:   uxth    r3, r3
080145ee:   smulbb  r3, r1, r3
080145f2:   uxth    r3, r3
080145f4:   add     r3, r2
080145f6:   uxth    r3, r3
080145f8:   lsrs    r3, r3, #8
080145fa:   uxth    r3, r3
080145fc:   uxtb    r3, r3
080145fe:   and.w   r3, r3, #31
08014602:   uxtb    r2, r3
08014604:   ldrb    r3, [r7, #13]
08014606:   bfi     r3, r2, #3, #5
0801460a:   strb    r3, [r7, #13]

----------------------------------------------------------

Mask:

080145ba:   push    {r7}
080145bc:   sub     sp, #20
080145be:   add     r7, sp, #0
080145c0:   strh    r0, [r7, #4]
080145c2:   strh    r1, [r7, #0]
080145c4:   mov     r3, r2
080145c6:   strb    r3, [r7, #3]
365           ret.ch.red = (uint16_t)((uint16_t)(c1.ch.red * mix + (c2.ch.red * (256 - mix))) >> 8) & 0x1F;
080145c8:   ldrb    r3, [r7, #5]
080145ca:   ubfx    r3, r3, #3, #5
080145ce:   uxtb    r3, r3
080145d0:   uxth    r2, r3
080145d2:   ldrb    r3, [r7, #3]
080145d4:   uxth    r3, r3
080145d6:   smulbb  r3, r2, r3
080145da:   uxth    r2, r3
080145dc:   ldrb    r3, [r7, #1]
080145de:   ubfx    r3, r3, #3, #5
080145e2:   uxtb    r3, r3
080145e4:   uxth    r1, r3
080145e6:   ldrb    r3, [r7, #3]
080145e8:   rsb     r3, r3, #256    ; 0x100
080145ec:   uxth    r3, r3
080145ee:   smulbb  r3, r1, r3
080145f2:   uxth    r3, r3
080145f4:   add     r3, r2
080145f6:   uxth    r3, r3
080145f8:   lsrs    r3, r3, #8
080145fa:   uxth    r3, r3
080145fc:   uxtb    r3, r3
080145fe:   and.w   r3, r3, #31
08014602:   uxtb    r2, r3
08014604:   ldrb    r3, [r7, #13]
08014606:   bfi     r3, r2, #3, #5
0801460a:   strb    r3, [r7, #13]
```

embeddedt, 2019-11-08:
> I've tested it on STM32F429 Discovery and the demo app with `-O0˙ and -O3. I've measured exactly the same performance. The disassembly was the same too

Which means it will have no runtime effect if we add a mask to remove the warning, right?

puzrin, 2019-11-08:
> That means it will have no runtime effect if we add a mask to remove the warning, right?

In theory, some ancient compiler may be not so good with this. But, from another hand, any modern code analyzer will mark indirect truncations as potential error.

kisvegabor, 2019-11-08:
> That means it will have no runtime effect if we add a mask to remove the warning, right?

Yes, at least, not on STM and gcc. :slightly_smiling_face:

With `-O3` the function compiles to this:
```c
374           ret.ch.green = ((uint16_t)((uint16_t)c1.ch.green * mix + (c2.ch.green * (256 - mix))) >> 8);
0800af2a:   ubfx    r2, r1, #5, #6
0800af2e:   str     r2, [sp, #16]
376           ret.ch.blue = ((uint16_t)((uint16_t)c1.ch.blue * mix + (c2.ch.blue * (256 - mix))) >> 8);
0800af30:   ubfx    r2, r1, #0, #5
365           ret.ch.red = (uint16_t)((uint16_t)(c1.ch.red * mix + (c2.ch.red * (256 - mix))) >> 8);
0800af34:   ubfx    r11, r1, #11, #5
376           ret.ch.blue = ((uint16_t)((uint16_t)c1.ch.blue * mix + (c2.ch.blue * (256 - mix))) >> 8);
0800af38:   str     r2, [sp, #28]
```
So with some magic it becomes only 4 lines. Even one extra line due to any reason causes 25% performance loss.

Anyway, I'm not 100 % sure that my approach is correct. Maybe really are modern compiler will do the job correctly.

However,  we need to add fixes to all bitfields assignments too. E.g.
```c
typedef struct {
    uint8_t a:3;
}test_t;

int main(int argc, char ** argv)
{
   b = 20;
    test_t t;
    t.a = b; // -> t.a = b & 0x7;
    ...
```

1. It'd be a huge work because there are a lot of bitfields.
2. Masks should be defined and maintained like `LV_TEST_MASK_A 0x7`
3. As the API accepts values from enums (e.g. `LV_CHART_UPDATE_MODE_SHIFT`) there shouldn't be issues with it. 

I have to note that there were never complains due to `Wconversation` so I suppose it's not so important for people who use the library. Therefore I think it will be a minor improvement which requires a lot of tedious work and results in less readable code.

puzrin, 2019-11-09:
> I have to note that there were never complains due to `Wconversation` so I suppose it's not so important for people who use the library.

IMO such logic is https://en.wikipedia.org/wiki/Survivorship_bias.

> Therefore I think it will be a minor improvement which requires a lot of tedious work and results in less readable code.

That's not "minor improvement". Having no warnings from lib headers is essential thing like having tests. Code can work ignoring this, but that does not mean situation is normal.

https://github.com/littlevgl/lvgl/issues/1213#issuecomment-551840088 - here i suggested alternate approaches to isolate problem. Is that possible?

There is a big difference between fixing all lvgl code warnings and fixing `lvgl.h` warnings only. Second case may have alternate low cost solution.

kisvegabor, 2019-11-10:
> There is a big difference between fixing all lvgl code warnings and fixing lvgl.h warnings only. Second case may have alternate low cost solution.

Okay, I'm convinced. Let's fix `-Wconversation` warnings in header files. It should mainly affect `lv_color.h` and `lv_area.h`.

Can you contribute in it?

puzrin, 2019-11-10:
> Can you contribute to it?

Nothing changed from previous time, i still don't like C programming :). It's better to ask someone else.

embeddedt, 2019-11-10:
> Can you contribute to it?

I will fix them.

embeddedt, 2019-11-10:
I began working on it, but I'm concerned about security.

For the purpose of this comment, let's use the following three lines from `lv_color.h`:

https://github.com/littlevgl/lvgl/blob/6f762bb7c776894ca49f8143cf28dade4099d988/src/lv_misc/lv_color.h#L235-L237

`error: conversion to 'unsigned char:3' may change value [-Werror=conversion]`

The only way to silence `-Wconversion` in this case, from my research, is to manually mask the appropriate bits to inform GCC that there will never be a case where the value overflows.

Per [a question on Stack Overflow](https://stackoverflow.com/questions/3319717/is-there-a-bit-equivalent-of-sizeof-in-c), it is impossible to automatically determine the size of a bitfield at compile time. That means that the size of the bitfield would now have to be stored in at least two locations:

1. The structure definition (`uint8_t a : 3`).
2. Anywhere where the bitfield has a value assigned to it: `s->a = value & 0x7`. In this case the mask effectively holds the size of the bitfield.

If we ever wanted to change the size of the bitfield, there are two potential problems that could occur.

1. We make the bitfield smaller. In that case, `-Wconversion` will appear again, as the value could potentially be larger than the bitfield's new width (as far as GCC is concerned).
2. We make the bitfield larger. As a result of having already masked the value to a *smaller* width, however, `-Wconversion` will *not* inform us of this issue. This would inevitably lead to strange runtime behavior, where bits would be mysteriously dropped from values.

I'm not sure what the best solution is for this. Personally, I would just turn the warning off, but I know @puzrin has issues with that. Maybe we should use a pragma (conditioned on `__GNUC__`) in public LittlevGL headers to turn off that warning? Any other ideas?

puzrin, 2019-11-10:
What about move that code into `lv_color_inlines.h`? It it will be included from `.c` directly (not exists in `lvgl.h`), that should solve problem.

It it really necessary to keep those fn in `.h`? Can this optimisation be done by compiler?

puzrin, 2019-11-10:
One more idea - create guard to say should we use manual inlining via macro or automatic. Also, gcc allows forced inlining `void func_test() __attribute__((always_inline));`

PS. That's only an idea, i have to experience to decide how good it is.

embeddedt, 2019-11-10:
The `lv_color_toXX` functions are sometimes used by the public (in display drivers). So this would introduce a new requirement of them including `lv_color_inlines.h`. I guess we have to decide if that is acceptable or not.

kisvegabor, 2019-11-11:
> I will fix them.

Thank you very much! You are a hero! :slightly_smiling_face: 

> I began working on it, but I'm concerned about security.

@embeddedt I was referring to this problem in [this comment](https://github.com/littlevgl/lvgl/issues/1213#issuecomment-551978323). I don't know an easy to maintain approach either. However, if we fix these issues only in the header files then mainly the colors will be affected by bitfield masking. And the bitfields of colors are rarely changing so IMO it should be fine to hardcode the masks there.

embeddedt, 2019-11-15:
@kisvegabor @puzrin See #1271.

kisvegabor, 2019-11-15:
@embeddedt I merged it!

embeddedt, 2019-11-15:
@kisvegabor Great!

@puzrin Can you verify that you no longer see any warnings in public LittlevGL header files?

puzrin, 2019-11-15:
@embeddedt 

I see a some of this kind (but that's less than before):

```
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:173:31: note: in expansion of macro ‘LV_COLOR_SET_B16’
 # define LV_COLOR_SET_B(c, v) LV_COLOR_SET_B16(c,v)
                               ^~~~~~~~~~~~~~~~
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:494:5: note: in expansion of macro ‘LV_COLOR_SET_B’
     LV_COLOR_SET_B(color, b8 >> 3);
     ^~~~~~~~~~~~~~
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:118:48: warning: conversion to ‘unsigned char:6’ from ‘uint16_t {aka short unsigned int}’ may alter its value [-Wconversion]
 # define LV_COLOR_SET_G16(c, v) (c).ch.green = (uint16_t)((v) & 0x3F);
                                                ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:169:33: note: in expansion of macro ‘LV_COLOR_SET_G16’
 #   define LV_COLOR_SET_G(c, v) LV_COLOR_SET_G16(c,v)
                                 ^~~~~~~~~~~~~~~~
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:495:5: note: in expansion of macro ‘LV_COLOR_SET_G’
     LV_COLOR_SET_G(color, g8 >> 2);
     ^~~~~~~~~~~~~~
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:117:46: warning: conversion to ‘unsigned char:5’ from ‘uint16_t {aka short unsigned int}’ may alter its value [-Wconversion]
 # define LV_COLOR_SET_R16(c, v) (c).ch.red = (uint16_t)((v) & 0x1F);
                                              ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:167:31: note: in expansion of macro ‘LV_COLOR_SET_R16’
 # define LV_COLOR_SET_R(c, v) LV_COLOR_SET_R16(c,v)
                               ^~~~~~~~~~~~~~~~
.pio/libdeps/emulator/lvgl/src/lv_hal/../lv_misc/lv_color.h:496:5: note: in expansion of macro ‘LV_COLOR_SET_R’
     LV_COLOR_SET_R(color, r8 >> 3);
     ^~~~~~~~~~~~~~
```

gcc/g++ 7.2.1

Does your compiler hide those?

embeddedt, 2019-11-15:
@puzrin Strange. I don't see those. Can you pull the latest commit from `dev-6.1` and try again? I just changed those three again now.

[54 more replies]

PR #1271 (2019-11-15): Fix -Wconversion warnings
I changed two inline helper functions to use `int16_t` instead of `uint16_t`. I don't think it's a big issue because using an out-of-range value wouldn't have worked properly anyways.

This should fix #1213.
closed 2019-12-11 by puzrin