lvgl/lvgl #1214: Add empty font to simplify PIO configs

puzrin, 2019-10-02:
I use all custom fonts (no built-ins). It's difficult to make lvgl in PIO use outer dirs. So, i declare & add [fonts](https://github.com/puzrin/dispenser/blob/dev/src/fonts_custom.h) [manually](https://github.com/puzrin/dispenser/blob/dev/src/app.cpp#L13).

This works and acceptable. But i still have to select one of built-in fonts in config, to allow lvgl dependency build. Currently i use [uncsii](https://github.com/puzrin/dispenser/blob/dev/src/lv_conf.h#L271) (minimal size), but it would be better to have "zero-size stub".

embeddedt, 2019-10-02:
Perhaps we should allow `LV_FONT_DEFAULT` to be `NULL`? That would take even less space than a stub.

kisvegabor, 2019-10-02:
> Perhaps we should allow LV_FONT_DEFAULT to be NULL? That would take even less space than a stub

Good idea. 

@puzrin 
Could it work in your case?

puzrin, 2019-10-02:
Tested - it works. Just changed to NULL in config. The only thing i don't understand - why in saved 800 bytes (!!!) of RAM? 

```
#define LV_FONT_DEFAULT        &lv_font_unscii_8

DATA:    [========= ]  86.3% (used 14140 bytes from 16384 bytes)
PROGRAM: [======    ]  55.7% (used 73036 bytes from 131072 bytes)

#define LV_FONT_DEFAULT        NULL

DATA:    [========  ]  81.3% (used 13316 bytes from 16384 bytes)
PROGRAM: [=====     ]  54.8% (used 71764 bytes from 131072 bytes)
```

AFAIK font consists of constants, and should affect only code size, not memory size.

embeddedt, 2019-10-02:
A portion of the converted font might be stored in RAM - not sure.

puzrin, 2019-10-02:
800 (!) bytes of static vars. Very suspicious.

kisvegabor, 2019-10-03:
Earlier not every array of the font was const and I forget to update unscii. Updated in `master`

puzrin, 2019-10-03:
I have no more questions then.

closed 2019-10-03 by puzrin

kisvegabor, 2019-10-03:
Thanks for pointing out the RAM usage issue!