lvgl/lvgl #1215: 6.1 broken build - bad paths in updated fonts

puzrin, 2019-10-02:
- in [6.0](https://github.com/littlevgl/lvgl/blob/master/src/lv_font/lv_font_roboto_12.c#L1) - relative, works in PIO
- in [6.1](https://github.com/littlevgl/lvgl/blob/dev-6.1/src/lv_font/lv_font_roboto_12.c#L1) - requires specific folders structure, doesn't work in PIO.

https://github.com/littlevgl/lvgl/blob/master/scripts/built_in_font/built_in_font_gen.py seems your build script misses `sed` call to replace path with correct one.

embeddedt, 2019-10-02:
I think this was an intentional change. @kisvegabor?

puzrin, 2019-10-02:
I think not. https://github.com/littlevgl/lv_font_conv/blame/master/lib/writers/lvgl/lv_font.js#L37 that's default from convertor, the same as from been added.

kisvegabor, 2019-10-03:
It should be `../../lvgl.h` be ensure that the internal files can include what they need without any further project configuration.

`built_in_font_gen.py` should generate a font similar to built-in ones but used externally by the user. 
There `lvgl/lvgl.h` is fine.

puzrin, 2019-10-03:
Switched to 6.1-dev. Now works.

closed 2019-10-03 by puzrin