lvgl/lvgl #1223: The Chinese font shows normal when the bpp set to 1. but when the bpp greater than 1, it shows abnormal

incity, 2019-10-12:
i use these two command:
    lv_font_conv --font chinese_ui.ttf -r 0x9009 --size 48 --format lvgl --bpp 1 -o stsong_30.c
    lv_font_conv --font chinese_ui.ttf -r 0x9009 --size 48 --format lvgl --bpp 2 -o stsong_30.c
[image]
[image]
[image]
[image]

puzrin, 2019-10-12:
If you are on 6.0, add `--no-compress`, or switch to 6.1-dev. Compressor defaults are a bit ahead of lvgl releases.

incity, 2019-10-14:
> If you are on 6.0, add `--no-compress`, or switch to 6.1-dev. Convertor defaults are a bit ahead of lvgl releases.

very thanks, i try it, it shows ok, but it works only in a small range of unicode. when i choose a big range(e.g. 0x4E00-0x9009 or 0x4E00-0x9FA5), it also shows abnormal.

here is my command line:
lv_font_conv --font chinese_ui.ttf -r 0x00-0x9009 --size 48 --format lvgl --bpp 2 --no-compress -o stsong_30.c

puzrin, 2019-10-14:
Those effects are not related anyhow. I'd suggest you to create a reproducible sample, because described command is not enougth. Also try this things:

- use `--format dump` and check if dumped images are correct or not
- try narrow down your example to case when it still works and stop works after small change.

incity, 2019-10-14:
thanks again, the dumped images are correct.

[image]
[image]
[image]

puzrin, 2019-10-14:
Prepare a test repo please, with your font and commands to execute. Ideally, this ones:

- with max range, still working correctly
- with max range + 1, when glyph become bad (will in be only one bad glyph or all revious too?).

incity, 2019-10-14:
> Prepare a test repo please, with your font and commands to execute. Ideally, this ones:
> 
> * with max range, still working correctly
> * with max range + 1, when glyph become bad (will in be only one bad glyph or all revious too?).

glyph become bad from 0x5669.  range 0x4e00-0x5668 is ok.

[image]
[image]
[image]

puzrin, 2019-10-14:
@incity please, create a test repo as requested, that's important, because we need reproduce situation locally.

incity, 2019-10-14:
> @incity please, create a test repo as requested, that's important, because we need reproduce situation locally.
i just use the sample code in lvgl(lv_linux_frame_buffer\lv_examples\lv_tests\lv_test_theme)
and modify the code in lv_test_theme_2.c from line 135(sb_create function)
[lv_test_theme.zip](https://github.com/littlevgl/lvgl/files/3722756/lv_test_theme.zip)

incity, 2019-10-14:
@puzrin there is a overflow in lv_font_fmt_txt_glyph_dsc_t struct. i remove the bit field. and it shows ok now.
[image]
[image]

embeddedt, 2019-10-14:
> BTW, if you have resources for such big fonts, why do you use lvgl then? Seems you can run qt + freetype.

It's probably for at least one of these reasons:

* LittlevGL is more lightweight (and therefore faster).
* It might be easier to work with for simple GUIs.
* Maybe @incity has already used LittlevGL on another platform before.

incity, 2019-10-14:
i do like the ui style of the littlevgl, it more like android and ios, im not very familiar with QT, maybe it can also realize that, but, i dont know how much work will it take.
do u have any suggestions.

puzrin, 2019-10-14:
> LittlevGL is more lightweight (and therefore faster).

This 2 statements are not related. For example, lvgl has no commands abstraction layer (see nuklear for example), which can be effectively done at gpu if exists. More simple != more fast.

> do u have any suggestions?

I can not decide what is better for you. In my current projects i use LVGL due memory constrains.

kisvegabor, 2019-10-14:
@incity 
With a lot of characters, you should enable [LV_FONT_FMT_TXT_LARGE](https://github.com/littlevgl/lvgl/blob/master/lv_conf_template.h#L281).

embeddedt, 2019-10-14:
> More simple != more fast.

You need more context to effectively make a statement like that.

On a platform without a dedicated GPU, something that does simpler graphics operations will be faster. On a platform with a GPU, there are probably advantages to the complexity (because it will be handled by the GPU).

closed 2019-10-21 by incity