lvgl/lvgl #1229: Ploygon are not always filled correctly

fivdi, 2019-10-20:
**Describe the bug**

If `lv_canvas_draw_polygon` is used to draw and fill a polygon the polygon is not always filled correctly. In this particular case the polygon is a trapezoid and it should be drawn and filled in red on a silver background. Everything functions correctly except for a thin 1 pixel wide line from the top left of the trapezoid to the bottom right of the trapezoid which is silver rather than red.

This is a followup to #1083.

**To Reproduce**

The code below can be used to reproduce the issue on an ESP-WROVER-KIT with https://github.com/littlevgl/esp32_ili9341

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "lvgl/lvgl.h"
#include "esp_freertos_hooks.h"

#include "drv/disp_spi.h"
#include "drv/ili9341.h"

#define CANVAS_WIDTH 100
#define CANVAS_HEIGHT 80

static void IRAM_ATTR lv_tick_task(void)
{
  lv_tick_inc(portTICK_RATE_MS);
}

void ui_create()
{
  static lv_style_t style;
  lv_style_copy(&style, &lv_style_plain);
  style.body.main_color = LV_COLOR_RED;

  static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];

  lv_obj_t *canvas = lv_canvas_create(lv_scr_act(), NULL);
  lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_TRUE_COLOR);
  lv_obj_align(canvas, NULL, LV_ALIGN_CENTER, 0, 0);
  lv_canvas_fill_bg(canvas, LV_COLOR_SILVER);

  lv_point_t points4[] = {
    {.x=10, .y=10},
    {.x=40, .y=10},
    {.x=30, .y=40},
    {.x=20, .y=40},
  };
  lv_canvas_draw_polygon(canvas, points4, 4, &style);
}

void app_main()
{
  lv_init();

  disp_spi_init();
  ili9341_init();

  static lv_color_t buf1[DISP_BUF_SIZE];
  static lv_color_t buf2[DISP_BUF_SIZE];
  static lv_disp_buf_t disp_buf;
  lv_disp_buf_init(&disp_buf, buf1, buf2, DISP_BUF_SIZE);

  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.flush_cb = ili9341_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  esp_register_freertos_tick_hook(lv_tick_task);

  ui_create();

  while(1) {
    vTaskDelay(1);
    lv_task_handler();
  }
}
```

**Expected behavior**

The trapezoid should be completely filled in red.

kisvegabor, 2019-10-21:
I was able to reproduce the issue in the simulator.

I've found that if the `[0]`th point id the boom left point it works correctly. It seems like a limitation of the polygon drawing. 
```c

  lv_point_t points4[] = {
    {.x=20, .y=40},
    {.x=30, .y=40},
    {.x=40, .y=10},
    {.x=10, .y=10},
  };
```

I'm thinking about a better algorithm to avoid this issue but I can offer only this workaround for now.

fivdi, 2019-10-21:
> I'm thinking about a better algorithm to avoid this issue but I can offer only this workaround for now.

The workaround works for me too.

This also works:
```c
  lv_point_t points4[] = {
    {.x=20, .y=40},
    {.x=10, .y=10},
    {.x=40, .y=10},
    {.x=30, .y=40},
  };
```

kisvegabor, 2019-10-21:
Great! Thanks for the feedback.

closed 2019-10-21 by kisvegabor