lvgl/lvgl #1230: Memory leakage

terrycmchan, 2019-10-21:
I am using the latest Github source code for my development.

I created an idle screen to display the digital clock.
Everytime the clock was updated, I deleted all created objects and then created them again.
I found that the used RAM was increasing and the biggest free RAM was decreasing as below.
used:   1284 (  8 %), frag:   0 %, biggest free:  15100
used:   1388 (  9 %), frag:   0 %, biggest free:  14996
used:   1492 ( 10 %), frag:   0 %, biggest free:  14892
used:   1596 ( 10 %), frag:   0 %, biggest free:  14788
used:   1700 ( 11 %), frag:   0 %, biggest free:  14684
...

The 'delete' function is scrIdle_Exit() while the 'create' function is scrIdle_Enter().

```
static void scrIdle_Exit(Menu_Item_t* const NewMenu)
{
  if ((NewMenu == &NULL_MENU) || (NewMenu == NULL))
    return;

  DBG_PRINTF("[UI]: %s\r\n", __func__);

  if (objScrIdle_Time != NULL)
  {
    lv_obj_del(objScrIdle_Time);
    objScrIdle_Time = NULL;
  }
  if (objScrIdle_Date != NULL)
  {
    lv_obj_del(objScrIdle_Date);
    objScrIdle_Date = NULL;
  }
  if (objScrIdle_AmPm != NULL)
  {
    lv_obj_del(objScrIdle_AmPm);
    objScrIdle_AmPm = NULL;
  }
  if (objScrIdle_Bg != NULL)
  {
    lv_obj_del(objScrIdle_Bg);
    objScrIdle_Bg = NULL;
  }
}

void scrIdle_Enter(void *data)
{
  DBG_PRINTF("[UI]: %s\r\n", __func__);
}

void scrIdle_Select(void *data)
{
  uint16_t year;
  uint8_t month;
  uint8_t day;
  uint8_t weekday;
  uint8_t hour;
  uint8_t min;
  uint8_t sec;
  uint16_t ms;
  getDateAndTime(&year, &month, &day, &weekday, &hour, &min, &sec, &ms);

  DBG_PRINTF("[UI]: %s\r\n", __func__);

  secCounter = sec;

  objScrIdle_Bg = lv_img_create(lv_disp_get_scr_act(NULL), NULL);
  lv_img_set_src(objScrIdle_Bg, &background);
  lv_obj_set_pos(objScrIdle_Bg, 0, 0);
//  lv_obj_set_drag(objScrIdle_Bg, true);

  genericDisp gDisp;
  gDisp.config = 0;
  gDisp.configBit.time = 0;
  gDisp.configBit.battery = 1;
  gDisp.configBit.date = 0;
  gDisp.configBit.heart = 0;
  gDisp.configBit.mbutton = 0;
  gDisp.configBit.lbutton = 0;
  gDisp.configBit.lbutton_arrow = 0;
  gDisp.configBit.rbutton = 0;
  gDisp.configBit.rbutton_arrow = 0;
  createGenericDisp(gDisp);

  lv_style_copy(&styleScrIdle_Small, &lv_style_plain);
  styleScrIdle_Small.text.font = &lv_font_roboto_16;

  lv_style_copy(&styleScrIdle_Large, &lv_style_plain);
  styleScrIdle_Large.text.font = &lv_font_roboto_28;

  objScrIdle_Time = lv_label_create(lv_disp_get_scr_act(NULL), NULL);
  lv_label_set_style(objScrIdle_Time, LV_LABEL_STYLE_MAIN, &styleScrIdle_Large);
  memset(strTime, 0, sizeof(strTime));
  if (hour == 12)
    sprintf(strTime, "%02d:%02dpm", hour, min);
  else if (hour > 12)
    sprintf(strTime, "%02d:%02dpm", hour - 12, min);
  else
    sprintf(strTime, "%02d:%02dam", hour, min);
  lv_label_set_text(objScrIdle_Time, strTime);
  lv_obj_align(objScrIdle_Time, NULL, LV_ALIGN_CENTER, IDLE_TIME_X, IDLE_TIME_Y);
//  lv_obj_set_pos(objScrIdle_Time, IDLE_TIME_X, IDLE_TIME_Y);

  objScrIdle_Date = lv_label_create(lv_disp_get_scr_act(NULL), NULL);
  lv_label_set_style(objScrIdle_Date, LV_LABEL_STYLE_MAIN, &styleScrIdle_Small);
  memset(strDate, 0, sizeof(strDate));
  sprintf(strDate, "%02d-%02d", month, day);
  lv_label_set_text(objScrIdle_Date, strDate);
  lv_obj_align(objScrIdle_Date, NULL, LV_ALIGN_IN_LEFT_MID, IDLE_DATE_X + 30, IDLE_DATE_Y - 20);
//  lv_obj_set_pos(objScrIdle_Date, IDLE_DATE_X, IDLE_DATE_Y);
}
```

Please help. Thanks!

kisvegabor, 2019-10-21:
Please, provide an example that can be copy-pasted to the simulator. I.e. no specific functions and variables.

Can you reduce the scope of the issue by deciding the label or the image causes the issue?

And please be sure you are using the latest `master` version of `lvgl`.

terrycmchan, 2019-10-21:
Hi kisvegabor,

I found the cause of the problem. It is my coding fault.
Thank you for your quick response.

By the way, I would like to know how to update the content of the text object without deleting all objects and creating them again?
Since I have faced an fatal error, if I only deleted the text object and created it again in order to refresh the content of the text object.

Thanks!

embeddedt, 2019-10-21:
@terrycmchan I recommend that you read the [relevant part of the documentation](https://docs.littlevgl.com/en/html/object-types/label.html#set-text).

kisvegabor, 2019-10-21:
> I found the cause of the problem. It is my coding fault.
Thank you for your quick response.

God to hear that you've found the issue.

You should find the answer on the link posted by @embeddedt so I close this issue.

closed 2019-10-21 by kisvegabor