lvgl/lvgl #1238: Problematic locking code in lv_task_handler() on preemptive systems

Boldie, 2019-10-24:
While doing code reviews, I saw a kind of locking which will fail in some situations on a preemptive OS.  lv_task_handler() there is a section to lock from running from multiple threads / tasks:
```
    /*Avoid concurrent running of the task handler*/
    static bool task_handler_mutex = false;
    if(task_handler_mutex) return;
    /// -> here the fuck happens
    task_handler_mutex = true;
```
I added the line where the non good stuff may happen, if a schedule will happen there. I saw a lot of code of this kind which is bad. So this code shall better be removed since it is misleading and will not work every time. Here one needs a TestAndSet strategy to be save, but I know this can not be generally implemented in C due to missing support in standard.

embeddedt, 2019-10-24:
> I saw a kind of locking which will fail in some situations on a preemptive OS. lv_task_handler() there is a section to lock from running from multiple threads / tasks

That code segment should be removed, as that kind of usage of `lv_task_handler` is now forbidden (as far as I know). https://docs.littlevgl.com/en/html/porting/os.html#tasks-and-threads

kisvegabor, 2019-10-25:
It's is possible (but not recommended) to call `lv_task_handler` manually if you have a blocking process and still want to make things "alive". It should be fine in some cases, but not when you are already in `lv_task_handler`.

I've added  `if(task_handler_mutex) return` not prevent crash in these cases when you call `lv_task_handler` from `lv_task_handler`.

embeddedt, 2019-10-25:
> not prevent crash in these cases when you call `lv_task_handler` from `lv_task_handler`.

I suggest that we forbid that kind of behavior in 7.0 (it seems kind of complex). As @Boldie pointed out, there is no way to atomically check and set a variable in C without having the ability to disable/enable interrupts and/or scheduling.

Boldie, 2019-10-25:
> It's is possible (but not recommended) to call `lv_task_handler` manually if you have a blocking 
> I've added `if(task_handler_mutex) return` not prevent crash in these cases when you call `lv_task_handler` from `lv_task_handler`.

Ok, that can be a use-case. Although it seems to weird to me, but I know sometimes one will have such situations. For this case, I would suggest to rename the variables not to say mutex, more to say like "already running". So if one reads the code he will not think of a mutex in classical way, only a protection for running from the same thread again.

kisvegabor, 2019-10-28:
>  As @Boldie pointed out, there is no way to atomically check and set a variable in C without having the ability to disable/enable interrupts and/or scheduling.

@embeddedt 
There is no context switch/interrupt to make crach in the situation I've described above. Imageine a situation like this:
```
while(1) {
  lv_task_handler();
  sleep(5);

  if(make_process) process();
}

void process(void) 
{
   uint8_t i;
   for(i = 0; i < 100; i++) {
     process_element(i);  //Takes 100 ms to process
     lv_task_handler(); 
  }
}
```

> For this case, I would suggest to rename the variables not to say mutex, more to say like "already running". So if one reads the code he will not think of a mutex in classical way, only a protection for running from the same thread again.

@Boldie 
Good idea. I've just renamed it in 8fe92bcd

embeddedt, 2019-10-28:
> There is no context switch/interrupt to make crash in the situation I've described above.

I still don't completely understand but I trust that you architected it properly. :slightly_smiling_face:

kisvegabor, 2019-10-28:
> I still don't completely understand but I trust that you architected it properly. slightly_smiling_face

Let's discuss it. It wouldn't be the first time when I'm sure I'm right but I am not. :smile: 

Can you imagine the [example I've shown](https://github.com/littlevgl/lvgl/issues/1238#issuecomment-546795944)? (slightly updated)

embeddedt, 2019-10-28:
> Can you imagine the example I've shown? (slightly updated)

Now I understand the example, but I can't see a case where `lv_task_handler` would be re-entered. `process` is only called outside `lv_task_handler`.

So now I don't see a need for `already_running` at all. :confused:

kisvegabor, 2019-10-28:
> but I can't see a case where lv_task_handler would be re-entered. process is only called outside lv_task_handler.

It's not a problem in the example, but somebody might do the same in an `lv_task` which can cause undefined behavior.

embeddedt, 2019-10-28:
Okay, now I understand. :slightly_smiling_face: This makes sense.

kisvegabor, 2019-10-29:
> This makes sense.

Okay, then leave it with the modified `already_running` name.

commit 8fe92bcd (2019-10-28): lv_task: rename task_handler_mutex to already_running. Fixes #1238
closed 2019-10-29 by kisvegabor