lvgl/lvgl #1241: Placeholder for Textarea should follow align setting for text

JavierReyes945, 2019-10-28:
**Describe the bug**

For a Textarea element, the `lv_ta_set_text_align(ta, LV_LABEL_ALIGN_CENTER)` sets the text layout as centered.
When a Placeholder is placed when the Textarea is empty, the Placeholder gets left aligned. This is counterintuitive, and there is no way to set this Placeholder alignment.

**To Reproduce**

```c
lv_obj_set_size( ta, 600, 150 );
lv_ta_set_text( ta, "" );
lv_ta_set_placeholder_text( ta, "Insert some text here..." );
lv_ta_set_text_align( ta, LV_LABEL_ALIGN_CENTER );
```
The placeholder gets aligned to the left.

**Expected behavior**

The Placeholder should follow the setting for the text to be inserted, or there should be a setting for the alignment of the Placeholder

**Screenshots**

**Additional context**

N.A.

embeddedt, 2019-10-28:
Looks like an oversight to me ([web browsers do it the way you are suggesting](https://jsfiddle.net/38pftoez/2/)).

The fix looks simple - we just need to set the alignment of the placeholder as well in `lv_ta_set_text_align`: https://github.com/littlevgl/lvgl/blob/68c752ab5097433bba30fd08723f0d2fae288f1f/src/lv_objx/lv_ta.c#L741-L766

kisvegabor, 2019-10-29:
I've fixed it in `master`.

closed 2019-10-29 by kisvegabor

JavierReyes945, 2019-10-31:
I have done the update in my project, but seems ineffective. The place holder still gets left justified. Could I ask this fix to be tested on your side?

embeddedt, 2019-10-31:
@JavierReyes945 Does the fix I suggested [here](https://github.com/littlevgl/lvgl/commit/dd100e5e07c8ae18c6e885d97d9c5938049fff44#commitcomment-35748893) work?

JavierReyes945, 2019-10-31:
> @JavierReyes945 Does the fix I suggested [here](https://github.com/littlevgl/lvgl/commit/dd100e5e07c8ae18c6e885d97d9c5938049fff44#commitcomment-35748893) work?

That is exactly the update that I meant. It does not work.

JavierReyes945, 2019-10-31:
Please excuse, I just tested with the PC Simulator project, and the fix provided works as expected. For some reason in my embedded target is not working, I will deep into it, but the fix is now tested. It is my side issue. Thanks for the reply nonetheless.

ngeisler11, 2021-01-23:
@kisvegabor I see that this was fixed awhile back and the PR was completed, but it seems like this bug might be back again?
I am using lvgl 7.10 and I see the same behavior - the placeholder text doesnt seem to align with the parent object.

Are you able to see if you are repro'ing the same?

kisvegabor, 2021-01-25:
@ngeisler11 
It works well for me:
```c

    ta1 = lv_textarea_create(lv_scr_act(), NULL);
    lv_obj_set_size(ta1, 200, 100);
    lv_obj_align(ta1, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_textarea_set_text(ta1, "");    /*Set an initial text*/
    lv_obj_set_event_cb(ta1, event_handler);
    lv_textarea_set_placeholder_text(ta1, "Hello");
    lv_textarea_set_text_align(ta1, LV_LABEL_ALIGN_CENTER);
```

commit dd100e5e (2019-10-29): ta: fix place holder alignment. Fixes #1241