lvgl/lvgl #1242: Little Bug: lv_cb_is_inactive returns false if inactive

CURTLab, 2019-10-28:
In the header file lv_cb.h on line 150 
`
/**
 * Get whether the check box is inactive or not.
 * @param cb pointer to a check box object
 * @return true: inactive; false: not inactive
 */
static inline bool lv_cb_is_inactive(const lv_obj_t * cb)
{
    return lv_btn_get_state(cb) == LV_BTN_STATE_INA ? false : true;
}
`

True and false should be swaped in order to return true if the checkbox is inactive.

embeddedt, 2019-10-28:
I agree with the fix.

I'm wondering whether I should apply it to `master` or `dev-6.1`. If I apply it to `master`, it'll break the existing inverted behavior that someone might be expecting. Is that worth it?

CURTLab, 2019-10-28:
I see a further problem, since the implementation is coupled with the button state. If lv_cb_set_inactive is called and the check box is unchecked (LV_BTN_STATE_REL), lv_cb_is_checked() will return true. Maybe a tri-state enum would help (like in Qt).

kisvegabor, 2019-10-28:
> I see a further problem, since the implementation is coupled with the button state. If lv_cb_set_inactive is called and the check box is unchecked (LV_BTN_STATE_REL), lv_cb_is_checked() will return true. Maybe a tri-state enum would help (like in Qt).

Swapping `true` and `false` will solve it, won't it?

> If I apply it to master, it'll break the existing inverted behavior that someone might be expecting. Is that worth it?

I vote for `dev-6.1` because v6.1 will be [released in a month](https://github.com/littlevgl/lvgl/issues/1132#issuecomment-546976849).

kisvegabor, 2019-10-31:
I've just fixed it in `dev-6.1`.

Thank you for pointing it out.

closed 2019-10-31 by kisvegabor