lvgl/lvgl #1243: [v7.0] General discussion

kisvegabor, 2019-10-28:
## Schedule
TBD

## Main new features

### New drawing
Complete rework of LittlevGL's draw engine to use `mask`s for more advanced and higher qualitly graphical effects. The drawings happen by defining masks (such as rounded rectangle, line, angle). When something is drawn the currently active masks can make some pixels transparent.  The most obvious use of this system is to remove the overflowing content from the rounded edges. It also allows drawing perfect circles.

- [x] arc anti-aliasing
- [x] triangle/polygon anti-aliasing
- [x] better shadows (`style.body.shadow.offset/spread`)
- [x] gradient stop position and horizontal gradient (`style.body.main/grad_color_stop`)
- [x] more accurate line positions
- [x] `LV_BLEND_MODE_NORMAL/ADDITIVE/SUBTRACTIVE`  blending modes (`style.xxx.blend_mode = `)
- [x] `style->body.corner_mask` to crop the content on the rounded corners

### Object updates
- [x] Add `lv_objmask` object type. Masks can be added to it. The children will be masked accordingly. 
- [x] Canvas: drawing to true color alpha and alpha only canvas
- [ ] Canvas: add effects (blur, zoom etc)
- [x] Fix scroll propagation on page-like objects
- [x] Add `LV_DRAG_DIR_ONE`

### Themes
- [ ] Store themes' data dynamically [LINK](https://forum.littlevgl.com/t/de-initing-themes/387/5)
- [ ] Rework and add visually improve themes

### Rework example repos

kisvegabor, 2019-10-31:
Thank you very much for the feedback @embeddedt !

**Button border**
I was a forgotten test style config.

**Switch shadow and knob position**
Fixed.
 

**Slider shadow**
It seems to me there is no shadow in neither `dev-6.1` not `dev-7.0`.

**Button shadow**
I added y offset. It looks better now.

**Arc**

The rotation is not intentional, it's just the natural base position in the new algorithm. We can add decide which base direction is the most convenient.

In addition, it works clock-wise now, which is more human-friendly IMO.

**Tabview indicator**
Fixed

**Inertia **
I'll investigate it.

kisvegabor, 2019-10-31:
I've fixed the *Intertia issue* too.

embeddedt, 2019-10-31:
@kisvegabor Great! I'll try it when I get a chance.

embeddedt, 2019-11-01:
All of what I reported appears to be fixed. Thanks!

One other thing: I noticed that the drop shadow is missing from the tabview in `lv_test_theme_1` on 7.0 (it's there in 6.1).

embeddedt, 2019-11-02:
I also notice that the autorepeat frequency on the keyboard and text area is significantly slower than how I remember it from 6.1. I haven't objectively verified this, however, so don't worry about fixing it yet.

kisvegabor, 2019-11-02:
> One other thing: I noticed that the drop shadow is missing from the tabview in lv_test_theme_1 on 7.0 (it's there in 6.1). It's also missing from the text area on Tab 2.

With which theme? Can you send a screenshot?

> I also notice that the autorepeat frequency on the keyboard and text area is significantly slower than how I remember it from 6.1. I haven't objectively verified this, however, so don't worry about fixing it yet.

I don't remember that I've changed anything related to it. But my memory is also not an objective thing. :smile:

embeddedt, 2019-11-02:
> With which theme?

The material theme. I'm comparing with the material theme from 6.1.

kisvegabor, 2019-11-03:
I've just added `lv_img_set_angle` to set image angle without canvas. The rotation is calculated during the drawing so the huge buffer (as in case of canvas) is not required. The image is alway rotated around its middle point.

Besides eliminating the buffer `lv_img_set_angle` is faster than using `lv_canvas_rotate`.

Some benchmark with a 160x100 image on a 200x200 canvas with STM32F429 Discovery (180MHz):
- TRUE_COLOR
     - canvas v6.1: 54 ms
     - canvas v7.0: 47 ms
     - img: 31-47 ms

- TRUE_COLOR_ALPHA
    - canvas v6.1:  102 ms
    - canvas v7.0: 84 ms
    - img: 33-49 ms
    
So the `lv_canvas_rotate` became 10-20% faster but using `lv_img_set_angle` was 70-300% faster then `v6.1` `lv_canvas_rotate`.

embeddedt, 2019-11-03:
@kisvegabor Neat! What happens to the image object dimensions in the event that, for example, I have a 120x33 image and I rotate it by 90 degrees? Does it crop the image, or does the image object resize automatically?

kisvegabor, 2019-11-04:
> The material theme. I'm comparing with the material theme from 6.1.

I've fixed the shadow issue.

> What happens to the image object dimensions in the event that, for example, I have a 120x33 image and I rotate it by 90 degrees? Does it crop the image, or does the image object resize automatically?

It keeps the original position and size of the image object but adds extra virtual draw area.
I've found that [CSS works](https://www.w3schools.com/cssref/tryit.asp?filename=trycss3_transform) the same way.

kisvegabor, 2019-11-04:
Here is an example to test image rotation:
[image]

```c
    LV_IMG_DECLARE(hand);
    lv_obj_t * img = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(img, &hand);
    lv_obj_align(img, NULL, LV_ALIGN_CENTER, 0, 0);

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_exec_cb(&a, img, lv_img_set_angle);
    lv_anim_set_values(&a, 0, 359);
    lv_anim_set_time(&a, 1000, 0);
    lv_anim_set_repeat(&a, 0);
    lv_anim_create(&a);

```

[hand.zip](https://github.com/littlevgl/lvgl/files/3803011/hand.zip)

kisvegabor, 2019-11-06:
I've added `lv_canvas_blur_hor/ver`.
[image]

seyyah, 2019-11-07:
**Idea**: icon loop wrapping on a area

For the object updates, we met an idea in the [forum](https://forum.littlevgl.com/t/how-to-scroll-a-pic/1095).

[image]

embeddedt, 2019-11-07:
I suppose that wherever we draw images we could add `shift_x` and `shift_y` options to start drawing the image at a given X or Y offset, and wrap around to the other end of the area as necessary.

kisvegabor, 2019-11-08:
If an image object's size id larger than its image source the images will be repeated to fill the space.
 `lv_img_offset` can be used to add an offset to the image source. The only missing thing is the first part(s) will not be drawn, however it should be easy to fix.

[image]

kisvegabor, 2019-11-08:
I've added `lv_img_set_zoom(img, zoom_factor)`. Similarly to rotate it can real-time zoom the images.
The defaul (not zoomed) zoom factor is 256. 128 means half size, 512 means double size.
There are two modes for transformations: anti-aliased and not anti-aliased (Can be changed with `lv_img_set_antialias(img, true/false)`)

The render time on STM32f429 with a 160x100 image:
- when the image is small only few ms 
- when the image is just a little bit zoomed: 40-50 ms
- when the size was doubled: 100-120 ms

[image]
(The lag comes from the gif recorder)

TridentTD, 2019-11-09:
Miss-type

`lv_img_set_antialais(...)`   ---> `lv_img_set_antialias(...)`

https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_objx/lv_img.h#L131

TridentTD, 2019-11-09:
@kisvegabor 
I've just tried,  zoom & rotate for c image source is worked well.  So nice!
Wish support for png decoder source too. It seems still not support ?

kisvegabor, 2019-11-12:
@TridentTD 

> I've just tried, zoom & rotate is worked well. So nice!

Thank you for the feedback! 

> lv_img_set_antialais(...) ---> lv_img_set_antialias(...)

I've fixed it, thanks!

embeddedt, 2019-11-15:
Another regression: each table cell now casts a shadow on its neighbor to the left.

[image]

embeddedt, 2019-11-15:
Now I'm testing with `lv_test_theme_2`.

This slider knob seems just a bit too big. :wink:

[image]

Minor: this seems like a bit of an odd rotation for the line meter to have.

[image]

`lv_test_theme_2` crashes altogether if I choose the Mono theme (on both the PC simulator and STM32, running `dev-7.0`).

embeddedt, 2019-11-15:
> lv_test_theme_2 crashes altogether if I choose the Mono theme

Quick debugging shows that it's caused by `th_styles[0]` being completely empty (meaning that the font pointer inside is `NULL`).

kisvegabor, 2019-11-16:
@embeddedt Thanks for the review.

I believe I've fixed all of them.

embeddedt, 2019-11-17:
@kisvegabor They all appear to be fixed. I noticed just now that Tab 3 is crashing when I use the mono theme with `lv_test_theme_1`.

embeddedt, 2019-11-17:
As a matter of fact, on a device with debug enabled, `lv_test_theme_1` won't load at all with the mono theme. An assertion checking for a valid style seems to fail.

kisvegabor, 2019-11-17:
> As a matter of fact, on a device with debug enabled, lv_test_theme_1 won't load at all with the mono theme. An assertion checking for a valid style seems to fail.

I've fixed it.

kisvegabor, 2019-11-20:
I've added `style->text.underline/strikethrough`.

kisvegabor, 2019-11-20:
`lv_ddlist_set/get_symbol` is also added to replace `lv_ddlist_set/get_draw_arrow`

kisvegabor, 2019-11-21:
@seyyah 
> Idea: icon loop wrapping on a area

I've pushed a fix, now it should work with `lv_img_set_offsetx/y`.

LanFengXiMenChuiXue, 2019-11-28:
HI:
I have two questions:

1.Possible bug:In function lv_tileview_scrl_signal() line401:

```
if(drag_dir & LV_DRAG_DIR_HOR)
        lv_obj_set_y(scrl, -ext->act_id.y * lv_obj_get_height(tileview) + style_bg->body.padding.top);
if(drag_dir & LV_DRAG_DIR_VER)
        lv_obj_set_x(scrl, -ext->act_id.x * lv_obj_get_width(tileview) + style_bg->body.padding.left);
```
This causes the page to be undraggable.
You can do it this way, or you can do it better:

```
if (drag_dir == LV_DRAG_DIR_BOTH){
	if ((ext->drag_right_en && indev->proc.types.pointer.drag_sum.x <= -LV_INDEV_DEF_DRAG_LIMIT) ||
	        (ext->drag_left_en && indev->proc.types.pointer.drag_sum.x >= LV_INDEV_DEF_DRAG_LIMIT))
		lv_obj_set_y(scrl, -ext->act_id.y * lv_obj_get_height(tileview) + style_bg->body.padding.top);
	else if ((ext->drag_bottom_en && indev->proc.types.pointer.drag_sum.y <= -LV_INDEV_DEF_DRAG_LIMIT) ||
		(ext->drag_top_en && indev->proc.types.pointer.drag_sum.y >= LV_INDEV_DEF_DRAG_LIMIT))
		lv_obj_set_x(scrl, -ext->act_id.x * lv_obj_get_width(tileview) + style_bg->body.padding.left);
	}
else if(drag_dir == LV_DRAG_DIR_HOR)
         lv_obj_set_y(scrl, -ext->act_id.y * lv_obj_get_height(tileview) + style_bg->body.padding.top);
else if(drag_dir == LV_DRAG_DIR_VER)
         lv_obj_set_x(scrl, -ext->act_id.x * lv_obj_get_width(tileview) + style_bg->body.padding.left);
```

2.It is recommended to add image rotation center and free Settings,Here's my guide(”zlm“ marked):

```
static lv_res_t lv_img_draw_core(const lv_area_t * coords, const lv_area_t * mask, const void * src,
	const lv_style_t * style, uint16_t angle, lv_point_t * center, uint16_t zoom, bool antialias, lv_opa_t opa_scale)
{

    ...

        if(angle || zoom != LV_IMG_ZOOM_NONE) {
            /*Get the exact area which is required to show the rotated image*/
			lv_coord_t pivot_x = (lv_area_get_width(coords)/2) + coords->x1;
			lv_coord_t pivot_y = (lv_area_get_height(coords)/2) + coords->y1;

			if (center){//zlm add
				pivot_x = center->x + coords->x1;
				pivot_y = center->y + coords->y1;
			}
   ....

		lv_draw_map(coords, &mask_com, cdsc->dec_dsc.img_data, opa, chroma_keyed, alpha_byte, style, angle, center, zoom, antialias);//zlm modify
    }
```
```
static void lv_draw_map(const lv_area_t * map_area, const lv_area_t * clip_area, const uint8_t * map_p, lv_opa_t opa,
	bool chroma_key, bool alpha_byte, const lv_style_t * style, uint16_t angle, lv_point_t * center, uint16_t zoom, bool antialaias)
{

 ...

            trans_dsc.cfg.angle = angle;
            trans_dsc.cfg.zoom = zoom;
            trans_dsc.cfg.src = map_p;
            trans_dsc.cfg.src_w = map_w;
            trans_dsc.cfg.src_h = map_h;
            trans_dsc.cfg.cf = cf;
	    trans_dsc.cfg.pivot_x = map_w/2;
	    trans_dsc.cfg.pivot_y = map_h/2;
	    if (center){//zlm add
			trans_dsc.cfg.pivot_x = center->x;
			trans_dsc.cfg.pivot_y = center->y;
	    }
...
```

```
static lv_res_t lv_img_signal(lv_obj_t * img, lv_signal_t sign, void * param)
{
...

    } else if(sign == LV_SIGNAL_REFR_EXT_DRAW_PAD) {
        /*If the image has angle provide enough room for the rotated corners */
        if(ext->angle || ext->zoom != LV_IMG_ZOOM_NONE) {
            lv_sqrt_res_t ds;
            lv_sqrt(ext->w * ext->w + ext->h * ext->h, &ds);
            ds.i = (ds.i * ext->zoom + 0) >> 8;        /*+10 to be sure anything won't be clipped*/

            lv_coord_t d = (ds.i - LV_MATH_MIN(ext->w, ext->h)) ;//zlm modify, remove "/2"
            img->ext_draw_pad = LV_MATH_MAX(img->ext_draw_pad, d);
        }
    }

    return res;
}
```
```
void lv_img_set_angle(lv_obj_t * img, int16_t angle, lv_coord_t center_x, lv_coord_t center_y)
{
    if(angle < 0 || angle >= 360) angle = angle % 360;

    lv_img_ext_t * ext = lv_obj_get_ext_attr(img);
    if(angle == ext->angle) return;

    lv_obj_invalidate(img);
    ext->angle = angle;
	ext->center.x = center_x;//zlm add center,
	ext->center.y = center_y;
    lv_obj_refresh_ext_draw_pad(img);
    lv_obj_invalidate(img);
}
```
It should have a better way.

embeddedt, 2019-11-28:
@LanFengXiMenChuiXue For `#2`, please send a pull request. It makes it easier to see the changes and also allows you to be credited for the change.

kisvegabor, 2019-11-28:
@LanFengXiMenChuiXue 
Please also send a Pull request for `#1`.

LanFengXiMenChuiXue, 2019-11-29:
@kisvegabor @embeddedt A Pull request has been sent.

LanFengXiMenChuiXue, 2019-11-29:
@kisvegabor @embeddedt 
I've also added a few things that I don't know if they can be submitted to source code.
1. I've added gesture judgment to my code, which is an event, like a click.In some low performance MCU, can not implement drag , gesture will be a good screen cut trigger event;
2.I've added a drop-down menu object that looks like it does on an Android phone（to set the wifi, bluetooth on or off）, and actually the implementation is a rewrite of tileview.

embeddedt, 2019-11-29:
> A Pull request has been sent.

Thanks! @kisvegabor Can you review it?

> I've added a drop-down menu object

Can you send a screenshot?

> I've added gesture judgment to my code, which is an event, like a click.

This would be interesting. Can you send a pull request for this?

kisvegabor, 2019-11-29:
> @kisvegabor Can you review it?

I've just started to review it.

kisvegabor, 2019-11-29:
I've commented on the PR.
And I'm also very curious of the other drop down list and gesture features.

LanFengXiMenChuiXue, 2019-11-30:
@kisvegabor @embeddedt

1.Gestures, like the one in the image below, are judged when the swipe is over and the action is made (in this case, the screen cut).
[image]
It can realize up, down, left and right gesture judgment, and then give it as an event to the object that enables gesture judgment.
2.The drop-down menu looks like this, and this is a screenshot of my phone, which is actually the same as this.
[image]

kisvegabor, 2019-11-30:
@LanFengXiMenChuiXue Both look interesting. Could you send 2 separate PRs for them?

kisvegabor, 2019-11-30:
I've added
- `lv_slider_set_knob_img(slider, &my_img)`
- `lv_sw_set_knob_off_img(slider, &my_img)`
- `lv_sw_set_knob_on_img(slider, &my_img)`
- `lv_gauge_set_needle_img(slider, &my_img, pivot_x, pivot_y)`

LanFengXiMenChuiXue, 2019-11-30:
@kisvegabor Yes, I will send it out next Monday. It may not be perfect enough. I am very glad to get your advice.

amirgon, 2019-11-30:
@kisvegabor, @embeddedt I have a suggestion for a change in `lv_conf.h` (and `lv_conf_template.h`, checker script, etc.)  

I find myself in a need to override some of the `lv_conf.h` macros  in the Makefile.  
Instead of writing a new `lv_conf.h`, I prefer to define the macros in the Makefile and override `lv_conf.h`.
For example, see https://github.com/littlevgl/lv_binding_micropython/pull/58

For this to work, macros we want to override **must to be defined conditionally**.
For example, 
```c
#ifndef LV_FONT_DEFAULT
#define LV_FONT_DEFAULT        &lv_font_roboto_16
#endif  // LV_FONT_DEFAULT
```
This would define `LV_FONT_DEFAULT` only if it wasn't already defined.
When written this way, we can use 
```Makefile
CFLAGS+=-DLV_FONT_DEFAULT=&myFont
```
to set the default font from the Makefile, or even from the command line when building the project.

I'm already using this to set `LV_COLOR_DEPTH` and `LV_COLOR_16_SWAP` macros on lv_micropython, but I found out I need more and more macros I can control from outside.

What do you think about defining **all** macros in `lv_conf.h` conditionally, in order to control the definition of any macro?
(I can do it on my project's `lv_conf.h` of course, but I'm suggesting to change this on `lv_conf_template.h` as well)

embeddedt, 2019-11-30:
The main issue I see with defining macros conditionally in `lv_conf[_template].h` is that it will triple the length of the file, since every option now takes up three lines instead of one. Do you think that's an issue?

kisvegabor, 2019-12-02:
I agree with @embeddedt that it would highly reduce the readability of `lv_conf.h`. It would have been still ok if the majority of users had overwritten the configs from outside but I think it's not the case. 

So IMO it's better to manage it locally. :frowning_face:

puzrin, 2019-12-02:
> (I can do it on my project's `lv_conf.h` of course, but I'm suggesting to change this on `lv_conf_template.h` as well)

I can understand idea, but IMO it will add too much garbage "in advance". If it's not blocker for you, i'd suggest consider such changes only when at least 3+ use cases available.

kisvegabor, 2019-12-02:
I got an idea:
Lets have a macro like:
```c
#define LV_CONF(name, val) \
#ifndef  name   \
#define name value \
#endif 
```

And use it in `lv_conf.h` like:
```c
LV_CONF(LV_MEM_SIZE, 10*1024)
```

This way `lv_conf.h` would be still readable.

What do you think?

amirgon, 2019-12-02:
> What do you think?

Unfortunately, a C macro cannot define another C macro.
C preprocessor just doesn't support that.

kisvegabor, 2019-12-02:
> Unfortunately, a C macro cannot define another C macro.

@amirgon I hoped you know some kind of magic for it. :frowning_face: 

What about having 2 `lv_conf_temp.h`: a normal (like now) and `lv_coof_templ_guard.h`.

embeddedt, 2019-12-02:
Could we generate the second one using a script that wraps each option in an `#ifndef`?

amirgon, 2019-12-02:
> @amirgon I hoped you know some kind of magic for it.

When I needed something like this in the past I usually generated the H file from a configuration file with some script, or used a different more powerful preprocessor (like [m4](https://en.wikipedia.org/wiki/M4_(computer_language))).  

> What about having 2 `lv_conf_temp.h`?
> A normal (like now) and `lv_coof_templ_guard.h`.

I'm not sure it's worth the trouble maintaining two scripts. And it's code duplication, which is not a good practice in most cases.

> Could we generate the second one using a script that wraps each option in an `#ifndef`?

It's possible, but again, I'm not sure it's worth the trouble. It's another step in the build process which adds complexity.

If you think most users wouldn't need this, I can maintain these conditionals on micropython binding's `lv_conf.h` for those macros I think are relevant. It's no big deal, I'm already doing that.  
The reason I brought this up was that I thought maybe others would want this on other contexts, but apparently no one does.

kisvegabor, 2019-12-03:
< If you think most users wouldn't need this, I can maintain these conditionals on micropython binding's lv_conf.h for those macros I think are relevant. It's no big deal, I'm already doing that.
The reason I brought this up was that I thought maybe others would want this on other contexts, but apparently no one does.

Okay. Let's warm up this topic again is more request arrives.

barbiani, 2019-12-05:
> * `lv_gauge_set_needle_img(slider, &my_img, pivot_x, pivot_y)`

Any special image format here? I am getting a thicker line or 'NO DATA' where the image is supposed to be.

kisvegabor, 2019-12-05:
> Any special image format here? 

No, normal C arrays should work. If you send a code snippet and the image you have used I'll test it.

barbiani, 2019-12-05:
> > Any special image format here?
> 
> No, normal C arrays should work. If you send a code snippet and the image you have used I'll test it.

Found the problem while creating the snippet. Setting the needle count seems to override the needle image so I skipped it to make it work perfectly.

//lv_gauge_set_needle_count(gauge1, 1, &needle_colors);
lv_gauge_set_needle_img(gauge1, &arrow, 11, 10);

Thank you!

kisvegabor, 2019-12-05:
> Setting the needle count seems to override the needle image so I skipped it to make it work perfectly.

They should work simultaneously (and it's really working for me). Can you send a code to reproduce?

barbiani, 2019-12-05:
I am experimenting with the visual studio sdl sample.
Lvgl branch dev-7.0.

```
 	extern const lv_img_dsc_t arrow;
	/*Describe the color for the needles*/
	static lv_color_t needle_colors;
	needle_colors = LV_COLOR_BLUE;
	/*Create a gauge*/
	lv_obj_t* gauge1 = lv_gauge_create(lv_scr_act(), NULL);
	//lv_gauge_set_needle_count(gauge1, 1, &needle_colors);
	lv_obj_set_size(gauge1, 250, 250);
	lv_obj_align(gauge1, NULL, LV_ALIGN_CENTER, 0, 50);
	lv_gauge_set_needle_img(gauge1, &arrow, 12, 10);
```

arrow.c
removed values other than for 32bpp color to reduce the post size.

```
#include "lvgl/lvgl.h"

#ifndef LV_ATTRIBUTE_MEM_ALIGN
#define LV_ATTRIBUTE_MEM_ALIGN
#endif

#ifndef LV_ATTRIBUTE_IMG_ARROW
#define LV_ATTRIBUTE_IMG_ARROW
#endif

const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_IMG_ARROW uint8_t arrow_map[] = {
  #if LV_COLOR_DEPTH == 32
	0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x70, 0x6f, 0x70, 0xff, 0x47, 0x47, 0x47, 0xff, 0x35, 0x35, 0x35, 0xff, 0x28, 0x28, 0x28, 0xff, 0x29, 0x28, 0x29, 0xff, 0x2d, 0x2d, 0x2d, 0xff, 0x41, 0x42, 0x41, 0xff, 0x5c, 0x5c, 0x5c, 0xff, 0x7f, 0x7f, 0x7f, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff,
	0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x71, 0x71, 0x71, 0xff, 0x29, 0x29, 0x29, 0xff, 0x02, 0x03, 0x02, 0xff, 0x00, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0xff, 0x01, 0x01, 0x01, 0xff, 0x12, 0x12, 0x12, 0xff, 0x54, 0x54, 0x54, 0xff, 0x7f, 0x7f, 0x7f, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0x [... cut ...] x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff, 0x80, 0x80, 0x80, 0xff,
  #endif
};

const lv_img_dsc_t arrow = {
  .header.always_zero = 0,
  .header.w = 113,
  .header.h = 20,
  .data_size = 2260 * LV_IMG_PX_SIZE_ALPHA_BYTE,
  .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
  .data = arrow_map,
};
```

kisvegabor, 2019-12-05:
Ahh, it's something different. The areas around the needle should be transparent. In your image the first pixel is `0x80, 0x80, 0x80, 0xff,`, 0xff is the opacity. It should be 0x00. Like this image:
[image]

However, when you set colors for the needle lvgl will recolor the images. If all pixels has 0xff opacity then whole image will be recolored.

barbiani, 2019-12-05:
That is alright then... thanks.

I do not have an editor to rotate the hand/arrow without disturbing the opacity like irfanview did.

kisvegabor, 2019-12-05:
> That is alright then... thanks.

Great!

> Irfanview killed the alpha channel... replacing with this provided png fixed it.

I use GIMP as an image editor.  It's easy to use for basic things like image rotation.

LanFengXiMenChuiXue, 2019-12-09:
@kisvegabor  For a more perfect animation, do you want to change the units of angles to 0.1 degrees.(For example, the rotation Angle of the picture).

[176 more replies]