lvgl/lvgl #1244: Inconsistent reporting of lv_page_get_fit_width and lv_page_get_scrl

Boldie, 2019-10-28:
Background: I want to create a TabView which has tabs filled from the left to the right border entirely with content (an image, line and rectorangle in production code). This was not working as expected. There is always a gap on the right border of 10 pixels in my case. After debugging around, I found out the page object from the tab returns inconsistent values for the following calls:
```
" tabHome width: %d", lv_obj_get_width(tabHome)
" tabHome lv_page_get_scrl: %d", lv_obj_get_width(lv_page_get_scrl(tabHome))
" tabHome lv_page_get_fit_width: %d", lv_page_get_fit_width(tabHome)
>>> OUTPUT
I (6578) GUI.MainScreen:  tabHome width: 320
I (6578) GUI.MainScreen:  tabHome lv_page_get_scrl: 310
I (6582) GUI.MainScreen:  tabHome lv_page_get_fit_width: 320
```

In my opinion and from the description (Get that width which can be set to the children to still not cause overflow) of the lv_page_get_fit_width function it shall be equivalent to  lv_obj_get_width(lv_page_get_scrl(tabHome)).

**To Reproduce**

Code snippets to reproduce (just add bug_demo() to the pc simulator works):
```
void bug_demo()
{
  lv_theme_t* th = lv_theme_material_init(120, NULL);
  lv_theme_set_current(th);

  lv_coord_t hres = lv_disp_get_hor_res(NULL);
  lv_coord_t vres = lv_disp_get_ver_res(NULL);

  // prev_scr = lv_disp_get_scr_act(NULL);

  lv_obj_t *m_screen = lv_obj_create(NULL, NULL);
  lv_obj_set_size(m_screen, hres, vres);
  lv_disp_load_scr(m_screen);

  /*Create a Tab view object*/
    lv_obj_t *tabview;
    tabview = lv_tabview_create(m_screen, NULL);
    lv_tabview_set_btns_pos(tabview, LV_TABVIEW_BTNS_POS_BOTTOM);

    /*Add 3 tabs (the tabs are page (lv_page) and can be scrolled*/
    lv_obj_t *tabHome = lv_tabview_add_tab(tabview, "Home");

    printf(" TV width: %d\n", lv_obj_get_width(tabview) );

    // Home page
    lv_page_set_scrl_fit( tabHome, LV_FIT_NONE );
    lv_page_set_sb_mode( tabHome, LV_SB_MODE_ON);

    static lv_style_t tabHomeBG_style;
    lv_style_copy(&tabHomeBG_style, lv_page_get_style(tabHome, LV_PAGE_STYLE_BG));
    tabHomeBG_style.body.padding.inner = 0;
    tabHomeBG_style.body.padding.top = 0;
    tabHomeBG_style.body.padding.right = 0;
    tabHomeBG_style.body.padding.left = 0;
    tabHomeBG_style.body.padding.bottom = 0;
    lv_page_set_style(tabHome, LV_PAGE_STYLE_BG, &tabHomeBG_style);

    static lv_style_t tabHomeSCRL_style;
    lv_style_copy(&tabHomeSCRL_style, lv_page_get_style(tabHome, LV_PAGE_STYLE_SCRL));
    tabHomeSCRL_style.body.padding.inner = 0;
    tabHomeSCRL_style.body.padding.top = 0;
    tabHomeSCRL_style.body.padding.right = 0;
    tabHomeSCRL_style.body.padding.left = 0;
    tabHomeSCRL_style.body.padding.bottom = 0;
    lv_page_set_style(tabHome, LV_PAGE_STYLE_SCRL, &tabHomeSCRL_style);

    static lv_style_t tabHomeSB_style;
    lv_style_copy(&tabHomeSB_style, lv_page_get_style(tabHome, LV_PAGE_STYLE_SB));
    tabHomeSB_style.body.padding.inner = 0;
    tabHomeSB_style.body.padding.top = 0;
    tabHomeSB_style.body.padding.right = 0;
    tabHomeSB_style.body.padding.left = 0;
    tabHomeSB_style.body.padding.bottom = 0;
    lv_page_set_style(tabHome, LV_PAGE_STYLE_SB, &tabHomeSB_style);

    printf(" tabHome width: %d\n", lv_obj_get_width(tabHome) );
    printf(" tabHome lv_page_get_scrl: %d\n", lv_obj_get_width(lv_page_get_scrl(tabHome)) );
    printf(" tabHome lv_page_get_fit_width: %d\n", lv_page_get_fit_width(tabHome) );

    static lv_style_t topBar_style;
    lv_obj_t * topBar;
    lv_style_copy(&topBar_style, &lv_style_plain_color);
    topBar_style.body.main_color = topBar_style.body.grad_color = lv_color_hsv_to_rgb(120, 90, 70);
    topBar_style.body.radius = 0; /* no radius */
    topBar = lv_obj_create(tabHome, NULL);
    lv_obj_set_style(topBar, &topBar_style);
    lv_obj_set_size(topBar, lv_obj_get_width(tabHome), 23);
    lv_obj_align(topBar, tabHome, LV_ALIGN_IN_TOP_RIGHT, 0, 0);
}
```

**Expected behaviour**

lv_obj_get_width(lv_page_get_scrl(tabHome))  ==  lv_page_get_fit_width(tabHome)

Additional expectation: There is no gap of the green bar on the right top side to the end of the screen, but maybe it is intended, but I am open for suggestions how to close it.

**Screenshots**

Screenshot of PC-Simulator

[image]

**Additional context**

N/A

Boldie, 2019-10-28:
Update: I found out my problem with the gap. It was dependent on the scrollbar stuff. -> So this is ok.

But I think, the following equation shall always be true (I read another issue pointing lv_page_get_fit_width is new, and there seems to be a missing assumption for the page type):
lv_obj_get_width(lv_page_get_scrl(tabHome)) == lv_page_get_fit_width(tabHome)

kisvegabor, 2019-10-29:
Can you send a simplified code example to show only the issue?

Boldie, 2019-11-01:
The following code is the most minimal to produce the differing output. I would expect to have the same width with disabled scrollbars for the scroll area and the tab itself if all paddings has been removed.

The interesting thing is: If I comment out at least one of the tow styling blocks, the equation lv_obj_get_width(lv_page_get_scrl(tabHome)) == lv_page_get_fit_width(tabHome) will be true again.

```
void bug_demo() {
  lv_obj_t *m_screen = lv_obj_create(NULL, NULL);
  lv_disp_load_scr(m_screen);

  lv_obj_t *tabview;
  tabview = lv_tabview_create(m_screen, NULL);
  lv_tabview_set_btns_pos(tabview, LV_TABVIEW_BTNS_POS_BOTTOM);

  lv_obj_t *tabHome = lv_tabview_add_tab(tabview, "Home");

  // Home page
  lv_page_set_scrl_fit(tabHome, LV_FIT_NONE);
  lv_page_set_sb_mode(tabHome, LV_SB_MODE_OFF);

  static lv_style_t tabHomeBG_style;
  lv_style_copy(&tabHomeBG_style, lv_page_get_style(tabHome, LV_PAGE_STYLE_BG));
  tabHomeBG_style.body.padding.inner = 0;
  tabHomeBG_style.body.padding.right = 0;
  tabHomeBG_style.body.padding.left = 0;
  lv_page_set_style(tabHome, LV_PAGE_STYLE_BG, &tabHomeBG_style);

  static lv_style_t tabHomeSCRL_style;
  lv_style_copy(&tabHomeSCRL_style,
                lv_page_get_style(tabHome, LV_PAGE_STYLE_SCRL));
  tabHomeSCRL_style.body.padding.inner = 0;
  tabHomeSCRL_style.body.padding.right = 0;
  tabHomeSCRL_style.body.padding.left = 0;
  lv_page_set_style(tabHome, LV_PAGE_STYLE_SCRL, &tabHomeSCRL_style);

  printf(" tabHome width: %d\n", lv_obj_get_width(tabHome));
  printf(" tabHome lv_page_get_scrl: %d\n",
         lv_obj_get_width(lv_page_get_scrl(tabHome)));
  printf(" tabHome lv_page_get_fit_width: %d\n",
         lv_page_get_fit_width(tabHome));
}
```

kisvegabor, 2019-11-02:
If you set `lv_page_set_scrl_fit(tabHome, LV_FIT_NONE);` you won't let LittlevGL to automatically "fit" the object size to the parent's size. So the object's size will leave as it was with the default size. Removing this line solved the issue for me.

kisvegabor, 2019-11-13:
If it's still not working please open for it in the Forum:
https://forum.littlevgl.com/

closed 2019-11-13 by kisvegabor