lvgl/lvgl #1246: Rename repos

kisvegabor, 2019-10-31:
I've started to organize the repos and introduce some naming conventions. As GitHub redirects the querries to the old name to the new renaming should be harmless.

I started to use these prefixes:
- `lv_port_*` for ports to hardware
- `lv_sim_*` for simulator projects
- `lv_lib_*` for examples with external libs
- `lv_binding_*` for binding to other languages

However, I'm struggling with some repos:
- `lv_arduino` (`lv_port` ?)
- `lv_micropython` (leave as it is?)
- `emscripten` (`lv_port` or `lv_sim`?)

What do you think?

@embeddedt @amirgon @pablo2048

amirgon, 2019-10-31:
@kisvegabor Regarding `lv_micropython`, my opinion is to leave it as is, mainly because people may already recognize it by this "brand" name, and renaming it now would be confusing.

Pablo2048, 2019-10-31:
I'm on with @amirgon - same thing with ```lv_arduino```. Renaming this repo can confuse unexperienced Arduino users.

embeddedt, 2019-10-31:
I'm fine with renaming the Emscripten port to `lv_sim_emscripten` (or `lv_port_emscripten`).

The PlatformIO project is a mix between a simulator and a port (if I remember correctly, both STM32 and SDL are supported) so I'm not entirely sure what to do about that one. Probably renaming it to `lv_port_platformio` is better (since that shows that it is a "port").

kisvegabor, 2019-10-31:
Thanks for the feedback!

I agree to leave `lv_micropython` and `lv_arduino`. 

For Emscripten I vote for `lv_sim_emscripten`.

I'm also not sure about platformio. It's more than a simulator but not a port to hardware. Maybe letting it as `lv_platformio` is fine.

What do you think, `port`, `sim`, `lib`, `binding` should be prefixed with `lv_`?

kisvegabor, 2019-10-31:
> IMO renaming will not solve any serious problem and can be postponed, until can not be avoided.
@puzrin 

I'm planning to organize the repos into lists in README, docs, and littlevgl.com and I'd like to make a little bit of standard.

amirgon, 2019-10-31:
> What do you think, `port`, `sim`, `lib`, `binding` should be prefixed with `lv_`?

I would keep the `lv_`. When someone mentions a "sim" or a "port" in a forum or some other media, it's too general. `lv_`  prefix immediately associates this sim/port with lvgl.

puzrin, 2019-10-31:
> I'm planning to organize the repos into lists in README, docs, and littlevgl.com and I'd like to make a little bit of standard.

I'm ok in advance with any rename. I mean, it looks like renaming just for fun.

Boldie, 2019-11-02:
Is "lv_port_* for ports to hardware" a real port to h hardware? A port in my opinion is to exchange some code in the lib to make it working on a special hardware. I think what you mean is an abstraction (for e.g. the esp32 stuff to adapt SPI etc.), adaption or a kind of driver.

embeddedt, 2019-11-02:
A port, by this definition, constitutes a complete platform project that can be compiled and run on a target board, rather than just a set of drivers that need to be added to a new project.

kisvegabor, 2019-11-02:
> A port, by this definition, constitutes a complete platform project that can be compiled and run on a target board, rather than just a set of drivers that need to be added to a new project. Do you think it's a poor use of the word?

@Boldie Do you know a better word for this?

puzrin, 2019-11-03:
As far as i understand, you wish to describe combinations of "IDE + board" with one word :). Not sure this is possible. Also, number of combinations has quadratic grow, tending to clutter repo list.

It looks like those projects are just one-time bootstrap templates. So, it could be enough to keep all in single repo. Name could be something like `lv_bootstrap_templates`.

embeddedt, 2019-11-03:
>  it could be enough to keep all in single repo

kisvegabor, 2019-11-04:
> Also, number of combinations has quadratic grow, tending to clutter repo list.

It's not the goal to have all the combinations just a few examples to help to get started. 

> Name could be something like lv_bootstrap_templates.

`lv_port` is really that bad? Maybe it's not perfect, but it's short everybody knows what that project is about. -> A project which works on a specific platform.

Maybe `lv_hw_...` can be an other option.

puzrin, 2019-11-04:
> `lv_port` is really that bad? Maybe it's not perfect, but it's short everybody knows what that project is about. -> A project which works on a specific platform.

It's only my personal opinion, as user. Those repos are used only once to create new project, and don't need updates tracking. I'd prefer to not see this noise in repos list (place all into single repo).

embeddedt, 2019-11-04:
>  (place all into single repo).

That makes the history get kind of messy. Perhaps creating a new lv_projects organization on GitHub and putting all the projects under that is a better option.

puzrin, 2019-11-04:
> That makes the history get kind of messy.

Google for "mono repository", that's a common practice for big projects.

kisvegabor, 2019-11-04:
As @embeddedt mentioned we started with a mono repo but it was hell. I've googled on it and saw that even Google uses mono repo(s), however in our case the advantages we got are:
- Much easier development and contribution. When I wanted to pull the monorepo and make some tests in repo `A`, I needed to clean up all other test codes and local trash from the totally unrelated `B`, `C`, `D` repos.
- It was a hell of submodules. It took several minutes to clone the repo (clone lvgl and lv_example 10 times)
- As projects are mostly unrelated we have a cleaner issue tracking. Some projects have maintainers from the community. It's much easier for them to deal with only those issues they are interested in. 

> That makes the history get kind of messy. Perhaps creating a new lv_projects organization on GitHub and putting all the projects under that is a better option.

I was also considering it. But what to do with `lv_lib_...` and `lv_sim_...`? We can have > 5 libs in the near future.

IMO the real solution would be if GitHub added some opportunity to organize the repos (directories?) as BitBucket and GitLab do. However, I haven't heard news related to it.

embeddedt, 2019-11-04:
> But what to do with lv_lib_... and lv_sim_...? We can have > 5 libs in the near future.

* Leave libraries under the littlevgl umbrella - that organization can be used for core code.
* Move simulators and ports to the lv_projects umbrella.

amirgon, 2019-11-04:
> * Move simulators and ports to the lv_projects umbrella.

Can you move repos to another organization without losing issues/PRs etc? Do they keep their URL?

embeddedt, 2019-11-04:
> Can you move repos to another organization without losing issues/PRs etc?

I think so. I will test it.

> Do they keep their URL?

Not sure if the old URL redirects or not.

kisvegabor, 2019-11-05:
I still don't know how important are the drawbacks of moving ports and sim to another organization:
- make the organization more fragile
- more difficult to find something if you need to search for 2 places
- potentially making URLs broken (still not sure)

Actually I don't know about organizations that opened a new organization for a group of projects. They rather have 80 repos. Another way would be to use the "Pinned repositories" area more intensively. We can add even 10 repos (popular ports/libs/sims, lv_examples etc) too in the order we want.

embeddedt, 2019-11-08:
> Can you move repos to another organization without losing issues/PRs etc?

Yes. See https://github.com/embeddedt/organization_test_repo (or https://github.com/littlevgl/organization_test_repo; they both redirect to the same repository).

The only important information that would be lost is issue assignment, which is not an issue in our case.

>I don't know about organizations that opened a new organization for a group of projects.

I did a really quick survey based on a few GitHub organizations I know of.

Emscripten has split it up to be more organizable. They have [emscripten-core](https://github.com/emscripten-core) for Emscripten itself and [emscripten-ports](https://github.com/emscripten-ports) for libraries that have been ported to work with it.

That change was made this year; before then the repository was under *kripken*'s (one of the lead developers) account. So I'm assuming (possibly prematurely) it was deliberately done as an attempt to keep things manageable.

[Bootstrap](https://github.com/twbs), on the other hand, has elected to keep 19 repositories (3 of which are old and archived) in the same organization. They have pinned the most important repositories to the top.

[Microsoft](https://github.com/microsoft)'s organization size is quite extreme; they have almost 3,000 repositories under it. Obviously no one is expected to find any specific project by browsing that page. I'd avoid following their example. :slightly_smiling_face:

kisvegabor, 2019-11-08:
> Yes. See https://github.com/embeddedt/organization_test_repo (or https://github.com/littlevgl/organization_test_repo; they both redirect to the same repository).
The only important information that would be lost is issue assignment, which is not an issue in our case.

Good news, thank you for testing it.

Here are some examples I've found:
- [Electron](https://github.com/electron): 71 repos, 6 pinned
- [Docker](https://github.com/docker) 137 repo, 0 pinned
- [Espressif](https://github.com/PX4) 73 repo, 6 pinned
- [Adafruit](https://github.com/adafruit) 1.2 repos, 0 pinned
- [Arduino](https://github.com/arduino) 147 repos, 3 pinned

I tend to believe that we will be fine with pinning 6 repos.

embeddedt, 2019-11-09:
> I tend to believe that we will be fine with pinning 6 repos.

I agree.

kisvegabor, 2019-11-12:
I've pinned 6 repos (this is the maximum) but still can decide a few things:
1. lvgl
2. lv_sim_eclipse_sdl
3. lv_sim_codeblocks_win or lv_sim_visualstudio
4. lv_examples
5. a popular project: esp32? 
6. and another popular project: micropython?

I'd like to add an STM project too but it seems there is no place for it.

What do you think?

embeddedt, 2019-11-12:
Aside from LittlevGL and the examples, I would pin one simulator, MicroPython, and STM32. Inside the README of the pinned simulator we could mention that there are other similar simulator projects if you prefer to use a different IDE. Plus we already have that table on the main README.

kisvegabor, 2019-11-12:
@embeddedt 
I've added only the Eclipse SDL simulator and updated its readme as you suggested.

So the first row:
1. lvgl
2. examples
3. simulator

Ant the second:
4. ESP
5. STM
6. Micropython

I still miss the frame buffer project but it seems that's what we can do.

kisvegabor, 2019-11-19:
I think we have discussed everything on this topic.

Thank you for the ideas!

closed 2019-11-19 by kisvegabor