lvgl/lvgl #1247: Page scroll bar is shown while it shouldn't be

Masi2018, 2019-10-31:
**Describe the bug**

I have created a page and then added a container which includes an image and two label objects. When I set the container layout to LV_LAYOUT_COL_L everything is fine but if I set it to LV_LAYOUT_OFF , while I have added only one container that is smaller that page size,  scrollbar is shown and also container is moved a little up.

**To Reproduce**

lv_obj_t *show_message(lv_obj_t *parent, msgType_t msg_type, char *text, char *usr_name)
{
  static lv_style_t outgoing_style, incoming_style, offline_style;

    lv_obj_t *cont = lv_cont_create(parent, NULL);
    lv_style_copy(&incoming_style, lv_cont_get_style(cont, LV_CONT_STYLE_MAIN));
    lv_style_copy(&outgoing_style, &incoming_style);
    lv_style_copy(&offline_style, &incoming_style);

    incoming_style.body.main_color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);
    incoming_style.body.grad_color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);
    incoming_style.text.color = LV_COLOR_MAKE(0x1a,0x1a,0x1a);

    outgoing_style.body.main_color = LV_COLOR_MAKE(0x04,0x7B,0xC1);
    outgoing_style.body.grad_color = LV_COLOR_MAKE(0x04,0x7B,0xC1);
    outgoing_style.text.color = LV_COLOR_MAKE(0xea,0xea,0xea);

    offline_style.body.border.width = 1;
    offline_style.body.border.color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);
    offline_style.text.color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);

    lv_cont_set_fit2(cont, LV_FIT_NONE , LV_FIT_TIGHT);
    lv_obj_set_width(cont, MSG_CONTENT_WIDTH);
    lv_cont_set_layout(cont, LV_LAYOUT_OFF);

    lv_cont_set_style(cont, LV_CONT_STYLE_MAIN,  msg_type == outgoing ? &outgoing_style : &incoming_style);
    lv_obj_align(cont, NULL, LV_ALIGN_IN_TOP_LEFT, PADDING_BORDER, 0);

    lv_obj_t *user_presence = NULL;

  if (msg_type == incoming)
  {

    vig_presence_status_t presence = offline;

    /* online status*/
    user_presence = lv_img_create(cont, NULL);
    lv_img_set_src(user_presence, presence == online ? &online_img : &offline1_img);
    lv_obj_align(user_presence, NULL, LV_ALIGN_IN_TOP_LEFT, 10, 0);

    /* sender name */
    static lv_style_t sender_name_style;
    lv_obj_t *sender_name = lv_label_create(cont, NULL);
    lv_style_copy(&sender_name_style, lv_label_get_style(sender_name, LV_LABEL_STYLE_MAIN));
    sender_name_style.text.font = &RobotoBold24;

    if (usr_name)
    {
      lv_label_set_text(sender_name, usr_name);
    }

    lv_label_set_long_mode(sender_name, LV_LABEL_LONG_DOT);
    lv_obj_align(sender_name, user_presence, LV_ALIGN_OUT_RIGHT_BOTTOM, 0, 0);

  }

/*Display message content*/
  lv_obj_t *msg_content = NULL;

    msg_content = lv_label_create(cont, NULL);
    lv_label_set_text(msg_content, text);//lv_label_set_static_text
    lv_label_set_long_mode(msg_content, LV_LABEL_LONG_BREAK);
    lv_obj_set_width(msg_content, MSG_CONTENT_WIDTH);
    if (user_presence)
    {
      lv_obj_align(msg_content, user_presence, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 10);
    }

  return cont;
}

void test_page()
{
  lv_obj_t * obj = lv_page_create(lv_scr_act(), NULL);

  lv_page_set_sb_mode(obj, LV_SB_MODE_AUTO);                     /*Show scroll bars is scrolling is possible*/
  lv_obj_set_pos(obj, 0, STATUS_BAR_H);
  lv_obj_set_size(obj, LV_HOR_RES_MAX, MSG_SCREEN_H);
  lv_page_set_style(obj, LV_PAGE_STYLE_SCRL, &lv_style_transp_tight);
  lv_page_set_style(obj, LV_PAGE_STYLE_BG, &lv_style_transp_tight);

  show_message(obj, incoming, "Hello James", "James");

}

**Expected behavior**
I need to set the container layout LV_LAYOUT_OFF to be able to add the objects in my disired location. And when the total size of objects inside the page is smaller than page scrollbar shouldn't be displayed . 

**Screenshots**
When container layout is off :
[image]

While I expect to have the following screen:
[image]

Any help is appreciated.

kisvegabor, 2019-11-02:
@Masi2018 
Please send a minimal code which can be copy-pasted to the simulator to reproduce the issue.
The code example has dependencies such as `msgType_t`,  `online_img`.

Masi2018, 2019-11-04:
@kisvegabor  
Sorry about that, here is the simplified code. I removed the image and instead used a label.

#define LV_HOR_RES_MAX          (480)
#define LV_VER_RES_MAX          (320)

`lv_obj_t *show_message(lv_obj_t *parent, char *text, char *usr_name)
{
  static lv_style_t outgoing_style, incoming_style, offline_style;

    lv_obj_t *cont = lv_cont_create(parent, NULL);
    lv_style_copy(&incoming_style, lv_cont_get_style(cont, LV_CONT_STYLE_MAIN));
    lv_style_copy(&outgoing_style, &incoming_style);
    lv_style_copy(&offline_style, &incoming_style);

    incoming_style.body.main_color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);
    incoming_style.body.grad_color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);
    incoming_style.text.color = LV_COLOR_MAKE(0x1a,0x1a,0x1a);

    outgoing_style.body.main_color = LV_COLOR_MAKE(0x04,0x7B,0xC1);
    outgoing_style.body.grad_color = LV_COLOR_MAKE(0x04,0x7B,0xC1);
    outgoing_style.text.color = LV_COLOR_MAKE(0xea,0xea,0xea);

    offline_style.body.border.width = 1;
    offline_style.body.border.color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);
    offline_style.text.color = LV_COLOR_MAKE(0xcc,0xcc,0xcc);

    lv_cont_set_fit2(cont, LV_FIT_NONE , LV_FIT_TIGHT);
    lv_obj_set_width(cont, 350);
    lv_cont_set_layout(cont, LV_LAYOUT_OFF);
    lv_cont_set_style(cont, LV_CONT_STYLE_MAIN,  &incoming_style);
    lv_obj_align(cont, NULL, LV_ALIGN_IN_TOP_LEFT, PADDING_BORDER, 0);

    lv_obj_t *user_presence = NULL;
    /* online status*/
  //  user_presence = lv_img_create(cont, NULL);
  //  lv_img_set_src(user_presence, &online_img );
      user_presence = lv_label_create(cont, NULL);
      lv_label_set_text(user_presence, ".");
    lv_obj_align(user_presence, NULL, LV_ALIGN_IN_TOP_LEFT, 10, 0);

    /* sender name */
    static lv_style_t sender_name_style;
    lv_obj_t *sender_name = lv_label_create(cont, NULL);

    if (usr_name)
    {
      lv_label_set_text(sender_name, usr_name);
    }

    lv_label_set_long_mode(sender_name, LV_LABEL_LONG_DOT);
    lv_obj_align(sender_name, user_presence, LV_ALIGN_OUT_RIGHT_BOTTOM, 0, 0);

/*Display message content*/
  lv_obj_t *msg_content = NULL;

    msg_content = lv_label_create(cont, NULL);
    lv_label_set_text(msg_content, text);//lv_label_set_static_text
    lv_label_set_long_mode(msg_content, LV_LABEL_LONG_BREAK);
    lv_obj_set_width(msg_content, 350);
    lv_obj_align(msg_content, user_presence, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 10);

  return cont;
}

void test_page()
{
  lv_obj_t * obj = lv_page_create(lv_scr_act(), NULL);

  lv_page_set_sb_mode(obj, LV_SB_MODE_AUTO);                     /*Show scroll bars is scrolling is possible*/
  lv_obj_set_pos(obj, 0, 37);
  lv_obj_set_size(obj, LV_HOR_RES_MAX, 225);
  lv_page_set_style(obj, LV_PAGE_STYLE_SCRL, &lv_style_transp_tight);
  lv_page_set_style(obj, LV_PAGE_STYLE_BG, &lv_style_transp_tight);

  show_message(obj, "Hello James", "James");

}`

kisvegabor, 2019-11-05:
This code showed the scrollbars for me too.
It's because when you create `cont` on  `y=0`  and add children it will grow to involve the children. It will go upwards too, therefore, will cause an overflow.

To solve it:
1. Set `lv_page_set_scrl_fit4(obj, LV_FIT_FILL, LV_FIT_FILL,LV_FIT_FLOOD,LV_FIT_FILL);` for the page. It will prevent the scrollable part to grow automatically upwards
2. Call `lv_obj_align(cont, NULL, LV_ALIGN_IN_TOP_LEFT, PADDING_BORDER, 0);` at the end of `show_message` to align `cont` when it got its final size.

Masi2018, 2019-11-05:
@kisvegabor Thank you very much. Issue has been resolved.

closed 2019-11-05 by Masi2018